springfall2008/batpred #1397: Repeated changing discharge rate on inverter 2

gcoan, 2024-08-27:
**Describe the bug**
Noticed that the charge rate on inverter 2 was changed repeatedly up and down between max and min yesterday evening (6-11pm), but inverter 1 remained constant (on max).

Inverter 2:
[image]

Inverter 1:
[image]

And through this period Predbat was on idle:
[image]

**Expected behavior**
Charge rate only changed when necessary to prevent repeated hammering of the inverter modbus registers.

I suspect this may be due to the cross-charging code kicking in every 5 minutes, and as noted in #829 (raised in March), this doesn't seem to be working properly.  I'm still getting cross charging occurring between the two inverters, it's not fast cross charging, usually only a few hundred watts, but it occurs frequently.

SoC graph for the same period 18:00-23:00 shows inverter 2 (H) SoC slowly rising and inverter 1 (G) SoC dropping:
[image]

It looks like battery pause mode isn't utilised during cross charging prevention. I suggested in #1031 that this might be a more effective way to prevent cross charging.  Tests of turning predbat into read only and setting pause mode to PauseCharging at sunset and then turning it off when charging is needed show that this strategy works well.
It wouldn't prevent the cross charging during the evening when there's still solar generation, but would go a big way to stopping this.

**Predbat version**
8.3.6

**Environment details**
HAOS 2024.8.3
2x Gen 1 hybrids, each with their own battery

**Log file**

gcoan, 2024-08-27:
Logfile:
[appdaemon.log.5.txt](https://github.com/user-attachments/files/16763787/appdaemon.log.5.txt)

Looking at the logfile I can see (as anticipated) lots of effort by Predbat to try to stop the cross charging, e.g.:
```
2024-08-26 18:32:50.588734 INFO pred_bat: Reserve min: 4.0% Battery_min:4.0%
2024-08-26 18:32:51.826329 INFO pred_bat: Inverter 1 Wrote scheduled_discharge_enable to False successfully and got off
2024-08-26 18:32:51.827720 INFO pred_bat: Inverter 1 sensor.predbat_GE_1_scheduled_discharge_enable set to False
2024-08-26 18:32:51.828749 INFO pred_bat: BALANCE: socs [86, 78] reserves [0.38, 0.17] battery_powers [-2134.0, 2022.0] total -112.0 battery_max_rates [2600.0, 2600.0] charge_rates [2496.0, 2496.0] pv_power [174.0, 271.0] load_power [0.0, 2206.0] total 4992.0 discharge_rates [2704.0000000000005, 0.0] total 2704.0000000000005
2024-08-26 18:32:51.829761 INFO pred_bat: BALANCE: out_of_balance True above_reserve [True, True] below_full [True, True] can_power_house [False, True] can_store_pv [True, True] power_enough_discharge [False, True] power_enough_charge [True, False] soc_low [False, True] soc_high [True, False]
2024-08-26 18:32:51.830906 INFO pred_bat: BALANCE: Inverter 1 is cross discharging during charge, attempting to balance it
2024-08-26 18:32:51.839603 INFO pred_bat: BALANCE: Completed this run
2024-08-26 18:32:51.885726 INFO pred_bat: BALANCE: Enabled balance charge False discharge False crosscharge True threshold charge 1.0 discharge 1.0
2024-08-26 18:32:51.887231 INFO pred_bat: --------------- PredBat - update at 2024-08-26 18:30:00+01:00 with clock skew 0 minutes, minutes now 1110
2024-08-26 18:32:51.888709 INFO pred_bat: Inverter 0: Type GE GivEnergy
2024-08-26 18:32:53.198242 INFO pred_bat: Inverter 0 has timed pause support enabled
2024-08-26 18:32:53.200155 INFO pred_bat: Reserve min: 4.0% Battery_min:4.0%
2024-08-26 18:32:54.476037 INFO pred_bat: Inverter 0 Wrote scheduled_discharge_enable to False successfully and got off
2024-08-26 18:32:54.477471 INFO pred_bat: Inverter 0 sensor.predbat_GE_0_scheduled_discharge_enable set to False
2024-08-26 18:32:54.478691 INFO pred_bat: Inverter 1: Type GE GivEnergy
2024-08-26 18:32:55.839951 INFO pred_bat: Inverter 1 has timed pause support enabled
2024-08-26 18:32:55.848237 INFO pred_bat: Warn: Out of range index 1 within item inverter_battery_rate_min value 0
2024-08-26 18:32:55.849484 INFO pred_bat: Reserve min: 4.0% Battery_min:4.0%
2024-08-26 18:32:57.138179 INFO pred_bat: Inverter 1 Wrote scheduled_discharge_enable to False successfully and got off
2024-08-26 18:32:57.139282 INFO pred_bat: Inverter 1 sensor.predbat_GE_1_scheduled_discharge_enable set to False
2024-08-26 18:32:57.140010 INFO pred_bat: BALANCE: socs [86, 78] reserves [0.38, 0.17] battery_powers [-2134.0, 2022.0] total -112.0 battery_max_rates [2600.0, 2600.0] charge_rates [2496.0, 2496.0] pv_power [174.0, 271.0] load_power [0.0, 2206.0] total 4992.0 discharge_rates [2704.0000000000005, 0.0] total 2704.0000000000005
2024-08-26 18:32:57.140783 INFO pred_bat: BALANCE: out_of_balance True above_reserve [True, True] below_full [True, True] can_power_house [False, True] can_store_pv [True, True] power_enough_discharge [False, True] power_enough_charge [True, False] soc_low [False, True] soc_high [True, False]
2024-08-26 18:32:57.141563 INFO pred_bat: BALANCE: Inverter 1 is cross discharging during charge, attempting to balance it
2024-08-26 18:32:57.142356 INFO pred_bat: BALANCE: Completed this run
2024-08-26 18:33:46.447089 INFO pred_bat: BALANCE: Enabled balance charge False discharge False crosscharge True threshold charge 1.0 discharge 1.0
2024-08-26 18:33:46.447863 INFO pred_bat: --------------- PredBat - update at 2024-08-26 18:30:00+01:00 with clock skew 0 minutes, minutes now 1110
2024-08-26 18:33:46.448535 INFO pred_bat: Inverter 0: Type GE GivEnergy
2024-08-26 18:33:47.711135 INFO pred_bat: Inverter 0 has timed pause support enabled
2024-08-26 18:33:47.712643 INFO pred_bat: Reserve min: 4.0% Battery_min:4.0%
2024-08-26 18:33:48.776484 INFO pred_bat: Inverter 0 Wrote scheduled_discharge_enable to False successfully and got off
2024-08-26 18:33:48.777541 INFO pred_bat: Inverter 0 sensor.predbat_GE_0_scheduled_discharge_enable set to False
2024-08-26 18:33:48.778263 INFO pred_bat: Inverter 1: Type GE GivEnergy
2024-08-26 18:33:50.077639 INFO pred_bat: Inverter 1 has timed pause support enabled
2024-08-26 18:33:50.079210 INFO pred_bat: Warn: Out of range index 1 within item inverter_battery_rate_min value 0
2024-08-26 18:33:50.080521 INFO pred_bat: Reserve min: 4.0% Battery_min:4.0%
2024-08-26 18:33:51.202322 INFO pred_bat: Inverter 1 Wrote scheduled_discharge_enable to False successfully and got off
2024-08-26 18:33:51.204163 INFO pred_bat: Inverter 1 sensor.predbat_GE_1_scheduled_discharge_enable set to False
2024-08-26 18:33:51.204972 INFO pred_bat: BALANCE: socs [87, 78] reserves [0.38, 0.17] battery_powers [-277.0, -7.0] total -284.0 battery_max_rates [2600.0, 2600.0] charge_rates [2496.0, 2496.0] pv_power [140.0, 280.0] load_power [0.0, 251.0] total 4992.0 discharge_rates [2704.0000000000005, 0.0] total 2704.0000000000005
2024-08-26 18:33:51.205736 INFO pred_bat: BALANCE: out_of_balance True above_reserve [True, True] below_full [True, True] can_power_house [True, True] can_store_pv [True, True] power_enough_discharge [False, False] power_enough_charge [True, False] soc_low [False, True] soc_high [True, False]
[... cut ...]
2024-08-26 18:34:00.920026 INFO pred_bat: Inverter 1 set discharge rate 2600 via REST successful on retry 0
2024-08-26 18:34:00.923785 INFO pred_bat: BALANCE: Completed this run
2024-08-26 18:34:46.524983 INFO pred_bat: BALANCE: Enabled balance charge False discharge False crosscharge True threshold charge 1.0 discharge 1.0
```

I'll increase the charge and discharge threshold from 1% of SoC to 3% which should reduce Predbat being so enthusiastic, but the other comments about use of pause charge and repeatedly changing the charge rate still I think

gcoan, 2024-08-27:
Update:

Looking at the behaviour tonight, the SoC's of the two batteries have drifted apart with H (inv 2) still climbing as G (inv 1) falls
[image]

Looking at inverter 2 charge rate, I can't see that changing the balance charge threshold set to 3% today is really any different to yesterday when it was set to the default 1%.  Still being repeatedly turned up and down, every minute with 5 minutes? between cycles.

Today 3%:
[image]

Yesterday 1% for comparison:
[image]

springfall2008, 2024-09-22:
I think the balancing code maybe fighting someone with the main run loop, one turning off charge and then other turning it back on hence the 5 minute pulses. In some ways this was kind of expected as the implementation is very basic, but perhaps could be improved

gcoan, 2024-09-22:
Given this behaviour I've turned the balance inverter switch off and quite often I end up leaving Predbat in read only mode overnight so I can set PauseCharge on the batteries to stop them cross charging.

Its not ideal, I'd like to be able to leave Predbat running, so a change to use PauseCharge would be definitely appreciated

gcoan, 2024-11-30:
@springfall2008 r/e recent speaktothegeek YT video

I know it's on your longer term list

gcoan, 2025-12-09:
ability to set pause charge overnight (freeze discharge mode) added in 8.27.5