springfall2008/batpred #1399: Solis - Write to charge_end_minute didn't complete

AKlamaznikovs, 2024-08-28:
**Describe the bug**
When Predbat tries to set charging time, the last step "charge_end_minute" fails to update.

It also appears in the log file that "charge_end_minute" returns the value of "charge_end_hour" after the failure. 

**Expected behavior**
Update the charging time correctly

**Predbat version**

8.3.7 (and previous), will test on 8.4.0

**Environment details**

Solis Inverter with Solax Modbus Integration
HAOS VM installed on a TrueNAS Scale

**apps.yaml - updated to latest Solax entity names**

  charge_discharge_update_button:
    - button.solis_update_charge_discharge_times

  timed_charge_current:
    - number.solis_timed_charge_current
  charge_start_hour:
    - number.solis_timed_charge_start_h
  charge_start_minute:
    - number.solis_timed_charge_start_m
  charge_end_hour:
    - number.solis_timed_charge_end_h
  charge_end_minute:
    - number.solis_timed_charge_end_m

  timed_discharge_current:
    - number.solis_timed_discharge_current

  discharge_start_hour:
    - number.solis_timed_discharge_start_h
  discharge_start_minute:
    - number.solis_timed_discharge_start_m
  discharge_end_hour:
    - number.solis_timed_discharge_end_h
  discharge_end_minute:
    - number.solis_timed_discharge_end_m

  energy_control_switch:
    - select.solis_energy_storage_control_switch

**Log file**
2024-08-28 07:00:04.380570: Configuring charge window now (now 08-28 07:00:00 target set_window_minutes 30 charge start time 08-29 01:30:00
2024-08-28 07:00:04.380662: Adjust idle time, charge 01:30:00-03:30:00 discharge 00:00:00-00:00:00
2024-08-28 07:00:04.380794: Adjust idle time computed idle is 07:00:00-23:59:00
2024-08-28 07:00:08.414809: Inverter 0 Wrote 01:30:00 to charge_start_time, successfully now 01:30:00
2024-08-28 07:00:12.457532: Inverter 0 Wrote 1 to charge_start_hour, successfully now 1
2024-08-28 07:00:12.458273: Inverter 0 No write needed for charge_start_minute: 30 == 30
2024-08-28 07:00:16.501868: Inverter 0 Wrote 03:30:00 to charge_end_time, successfully now 03:30:00
2024-08-28 07:00:20.537067: Inverter 0 Wrote 3 to charge_end_hour, successfully now 3
2024-08-28 07:00:20.574638: Warn: Failed to decode response <Response [500]> from http://supervisor/core/api/services/number/set_value
2024-08-28 07:00:24.596532: Warn: Failed to decode response <Response [500]> from http://supervisor/core/api/services/number/set_value
2024-08-28 07:00:28.635660: Warn: Failed to decode response <Response [500]> from http://supervisor/core/api/services/number/set_value
2024-08-28 07:00:32.657423: Warn: Failed to decode response <Response [500]> from http://supervisor/core/api/services/number/set_value
2024-08-28 07:00:36.678665: Warn: Failed to decode response <Response [500]> from http://supervisor/core/api/services/number/set_value
2024-08-28 07:00:40.702223: Warn: Failed to decode response <Response [500]> from http://supervisor/core/api/services/number/set_value
2024-08-28 07:00:44.714018: Warn: Inverter 0 Trying to write 30 to charge_end_minute didn't complete got 3
2024-08-28 07:00:44.728889: Info: record_status Warn: Inverter 0 write to charge_end_minute failed
2024-08-28 07:00:48.868603: Successfully pressed button button.solis_update_charge_discharge_times on Inverter 0
2024-08-28 07:00:48.868663: Inverter 0 Updated start and end charge window to 01:30:00 - 03:30:00 (old 06:30:00 - 07:00:00)
2024-08-28 07:00:48.868745: Setting Solis Energy Control Switch to 35 Self-Use from 33 Self-Use - No Timed Charge/Discharge for eco True
2024-08-28 07:00:53.067155: Inverter 0 Wrote select.solis_energy_storage_control_switch to Self-Use successfully

marcoaddario, 2024-08-28:
there is a typo in line 1960 in  inverter.py.
it actually is:

`self.write_and_poll_value("charge_end_minute", self.base.get_arg("charge_end_hour", indirect=False, index=self.id), int(new_end[3:5]))`

while it should be 

`self.write_and_poll_value("charge_end_minute", self.base.get_arg("charge_end_minute", indirect=False, index=self.id), int(new_end[3:5]))`

AKlamaznikovs, 2024-08-28:
> there is a typo in line 1960 in inverter.py. it actually is:
>
> `self.write_and_poll_value("charge_end_minute", self.base.get_arg("charge_end_hour", indirect=False, index=self.id), [...]

Thanks, sounds like this might be the issue. I've updated the inverter.py in my installation, will test and revert back tomorrow.

AKlamaznikovs, 2024-08-29:
Hello,

Just wanted to confirm that this change in inverter.py actually worked, see log below. 

Suggest this change is implemented in the "main".

Thanks for your help!

```
Inverter 0 Wrote 07:30:00 to charge_start_time, successfully now 07:30:00
2024-08-29 07:10:21.271041: Inverter 0 Wrote 7 to charge_start_hour, successfully now 7
2024-08-29 07:10:21.271145: Inverter 0 No write needed for charge_start_minute: 30 == 30
2024-08-29 07:10:25.360619: Inverter 0 Wrote 08:00:00 to charge_end_time, successfully now 08:00:00
2024-08-29 07:10:29.400584: Inverter 0 Wrote 8 to charge_end_hour, successfully now 8
2024-08-29 07:10:33.451091: Inverter 0 Wrote 0 to charge_end_minute, successfully now 0
2024-08-29 07:10:37.737307: Successfully pressed button button.solis_update_charge_discharge_times on Inverter 0
2024-08-29 07:10:37.737368: Inverter 0 Updated start and end charge window to 07:30:00 - 08:00:00 (old 01:30:00 - 07:00:00)
```

springfall2008, 2024-08-29:
Good spot I'll push a fix

springfall2008, 2024-08-30:
Now released

closed 2024-08-30 by springfall2008