springfall2008/batpred #1400: Predbat 8.4.0 fails to install

nipar4, 2024-08-28:
**Describe the bug**
With latest update to 8.4.0 with the web interface a new installation fails to install the components and run 
an upgrade of an existing installation works ok
I've tried adding web.py to the config directory manually but it still fails and I cannot see any other new files in the predbat apps directory

**Log file**
Starting Predbat standalone...
starting startup.py...........
Traceback (most recent call last):
  File "/config/hass.py", line 5, in <module>
    import predbat
  File "/config/predbat.py", line 38, in <module>
    from web import WebInterface
  File "/config/web.py", line 6, in <module>
    from config import CONFIG_ITEMS
ModuleNotFoundError: No module named 'config'
Bootstrap Predbat
Startup
Shutdown, sleeping 30 seconds before restarting
Predbat crashed. Restarting in 5 seconds...

nipar4, 2024-08-29:
The only way round this I can see is to copy all the python files across as part of the installation
This was not required before so just trying to identify what additional changes are needed to the docker build or if this is the new normal

springfall2008, 2024-08-29:
Is config.py missing?

springfall2008, 2024-08-30:
How did you update, using the Predbat update option in the menu?

nipar4, 2024-08-30:
Just to be clear this is only for a new installation, for an existing installation and updating via the menu it all works as all files are present

I did some more testing and the installation now needs both config.py and web.py to get through to downloading the installed version of predbat - previously it only needed predbat.py and download.py, which meant all I needed to monitor was two file changes, well one really, and let the program do the rest

EDIT: I hadn't noticed that you changed the installation docs from just predbat.py to *.py

-- Download predbat.py from Predbat repo (https://github.com/springfall2008/batpred/blob/main/apps/predbat/predbat.py) and place it into rootfs
++ Download *.py from Predbat repo (https://github.com/springfall2008/batpred/blob/main/apps/predbat/predbat.py) and place the code into rootfs

I guess the choice is either keep with just the basic predbat files and let the program download the missing files or include all the predbat files in the docker image

As its what the documentation says, I think I'll include all the py files in the image

closed 2024-08-30 by nipar4