springfall2008/batpred #1401: Check Fox ESS Configuration?

letsstartbeinganon, 2024-08-28:
Just thought I'd post this here as I think I've managed to get Batpred working on my Fox ESS inverter, connected via an Elfin EW11 modbus and using Nathan's Fox ESS Modbus tool.

I know many people have requested this type of config file so thought I'd share mine.  It would be great if other people could cast their eyes over this to make sure it is set up optimally?

`pred_bat:
  module: predbat
  class: PredBat
 
  # Sets the prefix for all created entities in HA - only change if you want to run more than once instance
  prefix: predbat
 
  # Timezone to work in
  timezone: Europe/London
 
  # Currency, symbol for main currency second symbol for 1/100s e.g. $ c or £ p or e c
  currency_symbols:
   - '£'
   - 'p'
 
  # Number of threads to use in plan calculation
  # Can be auto for automatic, 0 for off or values 1-N for a fixed number
  threads: auto
 
  # XXX: This is a configuration template, delete this line once you edit your configuration
  #template: True
 
  # Set to auto-match with a GivEnergy serial number, but you can override the serial or the sensor names
  # if it doesn't work or if you have more than one inverter you will need to list both
  #geserial: 're:sensor.givtcp_(.+)_soc_kwh'
  #geserial2: 're:sensor.givtcp2_(.+)_soc_kwh'
 
  # Sets the maximum period of zero load before the gap is filled, default 30 minutes
  # To disable set it to 1440
  load_filter_threshold: 30
 
  #
  # Sensors, more than one can be specified and they will be summed up automatically
  #
  # For two inverters the load today would normally be the master load sensor only (to cover the entire house)
  # If you have three phase and one inverter per phase then you would need three load sensors
  #
  # For pv_today if you have multiple solar inverter inputs then you should include one entry for each inverter
  #
  load_today:
    - sensor.load_energy_today
  import_today:
    - sensor.grid_consumption_energy_today
  export_today:
    - sensor.feed_in_energy_today
  pv_today:
    - sensor.solar_energy_today
 
  # Load forecast can be used to add to the historical load data (heat-pump)
  # To link to Predheat
  # Data must be in the format of 'last_updated' timestamp and 'energy' for incrementing kWh
  #load_forecast:
  #  - predheat.heat_energy$external
  #
  # If you enable ge_cloud_data then the load/import and export data will be fetches from the GE cloud instead of from GivTCP sensors
  # this is usually less efficient and of course prone to internet downtime, but could be useful if you lost your GivTCP data
  # Set the serial to the inverter serial to pull the data from and the key to your API key
  # When this is set load_today, import_today and export_today are not used
  #
  #ge_cloud_data: False
  #ge_cloud_serial: '{geserial}'
  #ge_cloud_key: 'xxxx'
  #
  # Controls/status - must by 1 per inverter
  #
  num_inverters: 1
[... 308 lines cut ...]
  # Gas rates for comparison
  #metric_octopus_gas: 're:(sensor.(octopus_energy_|)gas_[0-9a-z]+_[0-9a-z]+_current_rate)'
 
  # Nordpool market energy rates
  #futurerate_url: 'https://www.nordpoolgroup.com/api/marketdata/page/325?currency=GBP'
  #futurerate_adjust_import: True
  #futurerate_adjust_export: False
  #futurerate_peak_start: "16:00:00"
  #futurerate_peak_end: "19:00:00"
  #futurerate_peak_premium_import: 14
  #futurerate_peak_premium_export: 6.5
 
  # Watch list, a list of sensors to watch for changes and then update the plan if they change
  # This is useful for things like the Octopus Intelligent Slot sensor so that the plan update as soon as you plugin in
  # Only uncomment the items you actually have set up above in apps.yaml, of course you can add your own as well
  # Note those using +[] are lists that are appended to this list, whereas {} items are single items only
  watch_list:
  #  - '{octopus_intelligent_slot}'
  # - '{octopus_ready_time}'
  # - '{octopus_charge_limit}'
    - '{octopus_saving_session}'
  #  - '+[car_charging_planned]'
  #  - '+[car_charging_soc]'
  #  - '{car_charging_now}'
 `

springfall2008, 2024-08-30:
Thanks, I'll push it to the templates area

letsstartbeinganon, 2024-08-30:
> Thanks, I'll push it to the templates area

Thanks Trefor, I've just noticed that in line 258 on your template my account number is available as I wrote it in full rather than using the auto-sensor that is used elsewhere. Any chance you could remove that?

springfall2008, 2024-08-30:
Sorry now done

closed 2024-09-07 by springfall2008