springfall2008/batpred #1402: docker-compose - fail to start after editing apps.yaml

PianSom, 2024-08-29:
I am a HA Core user, who has been successfully using Predbat for months. I run appdeamon in its own docker container, and have predbat as an app inside. I now wish to move to standalone predbat, so that I can take advantage of the web page functionality offered by 8.4

I use the docker-compose, so followed [this](https://github.com/springfall2008/predbat_addon/blob/main/predbat/DOCS.md#alternative-compose-files-outside-container) ("Alternative compose")  instruction set (amended for my own paths obvs). 

The container starts just fine, and offers up the "Please Update apps.yaml" message. When edit the apps.yaml in my mapped /config it does not seem to find my changes, and just repeats the Update message. I have restarted the container, no joy. 

Is apps.yaml not stored in /config?

nipar4, 2024-08-29:
Did you delete "template: True" in line 13?

PianSom, 2024-08-29:
I replaced (first attempt by mv, then by editing and copy-paste) the file in config with my own apps.yaml, which works and does not have template set. 

I also tried replacing the apps.yaml in /addon, which had no effect

nipar4, 2024-08-29:
So is it working now?

assuming you are using the latest tag form docker hub
The /addon directory is just for storage so changing anything in there will have no affect
As part of the start-up the script copies everything from /addon to /config
Once you delete template: True - as in remove it, do not just comment it out, it should continue to boot
I'll update the script at some point to cater for commenting it out

PianSom, 2024-08-29:
Deleting the commented `template: True` line did the trick

Thank you!

closed 2024-08-29 by PianSom