springfall2008/batpred #1406: Implement Charge Freeze for Sofar using Battery Save Mode

dbw1966, 2024-08-30:
While car charging, predbat appears to put a Sofar inverter to charge mode.  This means battery is fully charging irrespective of tomorrow's forecast solar, buying at 8.5p plus losses, to sell at 8p so making a loss on every excess unit purchased.

Predbat log states Discharge Freeze, Charge Freeze and Reserve are disabled.

Sofar has "Battery Save Mode" which appears to match the description of Charge Freeze.

  Sofar2mqtt/set/auto - send value "true" or "battery_save"
  Battery Save mode is a hybrid mode where the battery will charge from excess solar but not discharge

Implementing this would allow Predbat to prevent discharge of batteries when car charging, or in iboost.

springfall2008, 2024-08-30:
Good idea, will take a look once I get time

dbw1966, 2024-08-30:
Thanks Trefor, Predbat does not seem to do any benefit for my case (with an ev) until this is done.  Negative every day on OG tariff as always charges to 100% overnight.

dbw1966, 2024-10-07:
Update - now on Agile and things improved with other fixes, but this charge freeze would be ideal.  Code on sofar2mqtt is as below to give an insight on how it works.

void batterySave()
{
static unsigned long lastRun = 0;

if(checkTimer(&lastRun, BATTERYSAVE_INTERVAL) && BATTERYSAVE)
{
modbusResponse  rs;

//Get grid power
unsigned int p = 0;

if(!readSingleReg(SOFAR_SLAVE_ID, SOFAR_REG_GRIDW, &rs))
p = ((rs.data[0] << 8) | rs.data[1]);
else
Serial.println("modbus error");

Serial.print("Grid power: ");
Serial.println(p);
Serial.print("Battery save mode: ");

// Switch to auto when any power flows to the grid.
// We leave a little wriggle room because once you start charging the
battery,
// gridPower should be floating just above or below zero.
if((p < 65535/2 || p > 65525) && (INVERTER_RUNNINGSTATE != discharging))
{
//exporting to the grid
if(!sendPassiveCmd(SOFAR_SLAVE_ID, SOFAR_FN_AUTO, 0, "bsave_auto"))
Serial.println("auto");
}
else
{
//importing from the grid
if(!sendPassiveCmd(SOFAR_SLAVE_ID, SOFAR_FN_STANDBY, SOFAR_PARAM_STANDBY,
"bsave_standby"))
Serial.println("standby");
}
}
}

dbw1966, 2026-08-27:
> Hi @dbw1966 <https://github.com/dbw1966> — we're going through older
> issues at the moment, closing what we can and fixing what still needs it,
> so apologies this one has sat since 2024.
>
> Is this still a problem for you? The reason for asking is that the generic
> mechanism you were effectively asking for arrived shortly after you raised
> this, in #1579 <https://github.com/springfall2008/batpred/pull/1579> (Nov
> 2024): Predbat now has configurable charge_freeze_service and
> discharge_freeze_service entries in apps.yaml, which are called when it
> wants a passive hold rather than a real charge or discharge — including
> exactly the "don't discharge while the car is charging, or while iBoost is
> running" case you described.
>
> The other half looks like it's already there too.
> templates/sofar_inverter.yaml defines a *"SoFar Battery Saver Mode"*
> switch that publishes battery_save to Sofar2mqtt/set/auto — the command
> you documented above. What appears to be missing is simply the wiring
> between the two: templates/sofar.yaml doesn't set charge_freeze_service
> to point at that switch. If that's all it takes, this may be a
> configuration and documentation fix rather than any new Sofar-specific code.
>
> Two things I can't check from here. I don't have a Sofar to test against,
> so I can't confirm that battery_save behaves closely enough to Predbat's
> charge-freeze contract (charge from surplus PV, but don't discharge). And
> you mentioned back in October 2024 that things had improved on Agile with
> [... 7 lines cut ...]
> the template and docs for the next person.
>
> Tagging as an autoclose candidate for now — if we don't hear back this
> will be closed, but please do comment if it's still live and we'll pick it
> back up.

Hi Rik
Thanks for the note.

I have not configured any services.  I believe Predbat got smart enough to
either set a very low power charge or discharge in place of
freeze_charge....

I have not configured any services.  I have updated config.yaml to define
the switches as here.

    - name: "SoFar Auto Saver Mode"
      command_topic: "Sofar2mqtt/set/auto"
      state_topic: "Sofar2mqtt/set/auto"
      qos: 1
      payload_on: "true"
      payload_off: "battery_save"
      retain: true
    - name: "SoFar Battery Saver Mode"
      command_topic: "Sofar2mqtt/set/auto"
      state_topic: "Sofar2mqtt/set/auto"
      qos: 1
      payload_on: "battery_save"
      retain: true

Anyhow, the code for the battery save is below so you can see for yourself,
this is running on the sofar2mqtt device.  If power is going to the grid
and the battery is not discharging (so it must be coming from the PV),
switch to auto, otherwise switch to standby (so not discharging).  If you
can indicate how and where to define the service, I'll give it a go.
Ideally you'd also advise what test you want performed and how.

battery_save is a hybrid auto mode that will charge from excess solar but
not discharge.

void batterySave()
{
  static unsigned long  lastRun = 0;

  if(checkTimer(&lastRun, BATTERYSAVE_INTERVAL) && BATTERYSAVE)
  {
    modbusResponse  rs;

    //Get grid power
    unsigned int  p = 0;

    if(!readSingleReg(SOFAR_SLAVE_ID, SOFAR_REG_GRIDW, &rs))
      p = ((rs.data[0] << 8) | rs.data[1]);
    else
      Serial.println("modbus error");

    Serial.print("Grid power: ");
    Serial.println(p);
    Serial.print("Battery save mode: ");

    // Switch to auto when any power flows to the grid.
    // We leave a little wriggle room because once you start charging the
battery,
    // gridPower should be floating just above or below zero.
    if((p < 65535/2 || p > 65525) && (INVERTER_RUNNINGSTATE != discharging))
    {
      //exporting to the grid
      if(!sendPassiveCmd(SOFAR_SLAVE_ID, SOFAR_FN_AUTO, 0, "bsave_auto"))
        Serial.println("auto");
    }
    else
    {
      //importing from the grid
      if(!sendPassiveCmd(SOFAR_SLAVE_ID, SOFAR_FN_STANDBY,
SOFAR_PARAM_STANDBY, "bsave_standby"))
        Serial.println("standby");
    }
  }
}

Cheers
Dave

dbw1966, 2026-09-13:
Hi Rik
On further investigation, Battery Save is not a good implementation, but
setting charge or dischage to 1W should work (it could be 0W but for my
laziness in changing C++ code in the sofar2mqtt adaptor).

Therefore I've added 2 switches to config.yaml
    - name: "SoFar Charge Freeze 1W"
      command_topic: "Sofar2mqtt/set/charge"
      state_topic: "Sofar2mqtt/set/charge"
      qos: 1
      payload_on: "1"
      retain: true

    - name: "SoFar Discharge Freeze 1W"
      command_topic: "Sofar2mqtt/set/discharge"
      state_topic: "Sofar2mqtt/set/discharge"
      qos: 1
      payload_on: "1"
      retain: true

And the following to Apps.yaml
# Enable freeze controls in Predbat
support_charge_freeze: true
support_discharge_freeze: true

# Service triggers to engage 1W freezes (cab be 0W if messageValue > 0
changed to >= 0)
charge_freeze_service:
  - service: switch.turn_on
    entity_id: switch.sofar_charge_freeze_1w

discharge_freeze_service:
  - service: switch.turn_on
    entity_id: switch.sofar_discharge_freeze_1w

# Service triggers to revert inverter to normal operation when freeze slots
end
charge_stop_service:
  - service: switch.turn_on
    entity_id: switch.sofar_auto_mode

discharge_stop_service:
  - service: switch.turn_on
    entity_id: switch.sofar_auto_mode

But I am seeing the following in the logs, is there something hard coded?
Not sure how to test if it's working.
78576 2026-09-13 12:14:40.788113: Note: Inverter does not support reserve -
disabling reserve functions
78575 2026-09-13 12:14:40.788103: Note: Inverter does not support charge
freeze - disabled
78574 2026-09-13 12:14:40.787869: Note: Inverter does not support discharge
freeze - disabled

springfall2008, 2026-09-13:
Automated first-pass triage (a bot will re-check this; a maintainer reviews before any action) — picking up the latest question from @dbw1966 about the three "Inverter does not support ..." log lines.

**Classification:** enhancement (the original 2024 ask), with the live question now being a configuration one — the freeze machinery exists, it's just gated off for Sofar by default.

**What I checked (against main, v9.0.2-28):**

Yes, something is hard-coded, but it's deliberately overridable. Predbat keeps a per-inverter-type capability table (`INVERTER_DEF` in `apps/predbat/config.py:1775`), and the Sofar entries (`SF` at config.py:1972, `SFMB` at config.py:2000) ship with `support_charge_freeze: False`, `support_discharge_freeze: False` and `has_reserve_soc: False`. At inverter creation Predbat copies those into `inv_support_charge_freeze` / `inv_support_discharge_freeze` (`apps/predbat/inverter.py:415-416`), and when either is False `execute.py` force-disables the feature and prints exactly the lines you saw (`apps/predbat/execute.py:965-978`):

- `charge freeze - disabled` → `set_charge_freeze = False`
- `discharge freeze - disabled` → `set_export_freeze = False`
- `reserve - disabling reserve functions` → reserve functions off

That's why nothing you configured fired: with `set_charge_freeze` forced off, the plan never requests a freeze, so `charge_freeze_service` / `discharge_freeze_service` are never called (`apps/predbat/inverter.py:3089` / `3136` are the only call sites).

**Why your apps.yaml lines had no effect:** `support_charge_freeze` / `support_discharge_freeze` are not recognised as top-level Predbat settings — there is no config item with those names (the switches Predbat reads are `set_charge_freeze`, `set_charge_freeze_only`, `set_export_freeze`, etc., `apps/predbat/fetch.py:3118-3128`). The capability flags live only in the per-inverter table and are overridden through the `inverter:` block in `apps.yaml` (`apps/predbat/inverter.py:376-386` merges user keys into `INVERTER_DEF` before the flags are read). This is documented under "I want to add an unsupported inverter" in the docs, and other templates do exactly this — e.g. `templates/luxpower.yaml` and `templates/fronius.yaml` set `support_charge_freeze` inside an `inverter:` block.

**To enable freeze on your Sofar, replace the top-level `support_charge_freeze` / `support_discharge_freeze` lines with:**

```yaml
  inverter_type: "SF"
  inverter:
    support_charge_freeze: true
    support_discharge_freeze: true
```

(keep `has_reserve_soc` unset — the reserve line in your log is correct for Sofar and shouldn't be overridden). After a restart the two "does not support" lines should disappear and Predbat should start calling your `charge_freeze_service` / `discharge_freeze_service` switches when the plan wants a hold; your 1W switch approach looks sound against the service contract.

**Residual enhancement:** Rik's suggestion stands — `templates/sofar.yaml` could ship the `inverter:` override plus commented `charge_freeze_service` / `discharge_freeze_service` examples pointing at the Battery Saver / 1W switches, so the next Sofar user doesn't hit this. The user-level wiring itself needs no new code. One open question for anyone who can test on hardware: whether `battery_save` on sofar2mqtt matches Predbat's charge-freeze contract well enough (charge from surplus PV, no discharge), or whether the 1W switch route is the better default to document.

Possibly relevant: #1579 (the generic freeze mechanism this predates).

No debug yaml or full log was attached, so I verified from code and the quoted log lines only — if the flags are enabled and the services still don't fire, a `predbat_debug.yaml` would settle it.