springfall2008/batpred #1407: energy_today_generation getting erroneous number, Sofar2Mqtt

dbw1966, 2024-08-30:
The energy_today_generation sensor is getting occasional numbers (once or twice a day) which are less than the previous one.  This causes issues with the Energy dashboard and probably other areas.  Is there some filtering that could be done?  Is there a record of all the data from the MQTT where it can be pinponted?

See attached screenshot of sensor.
[image]
[image]

dbw1966, 2024-08-30:
I have noted HA log issues at the same time
2024-08-29 18:37:15.405 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'solarPVAmps' when rendering '{{ value_json.solarPVAmps }}'
2024-08-29 19:37:16.917 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'batterySOC' when rendering '{{ value_json.batterySOC }}'

and 

2024-08-29 12:52:17.040 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'solarPVAmps' when rendering '{{ value_json.solarPVAmps }}'
[HA Log DW 240830 1008.txt](https://github.com/user-attachments/files/16814622/HA.Log.DW.240830.1008.txt)

springfall2008, 2024-08-30:
I'm not sure this is a Predbat sensor is it, where does it come from?

dbw1966, 2024-08-30:
From the Sofar.yaml taken from this site, examples below.  Is something is wrong with a particular batch of readings would this cause the problem?  I dont see any symptoms except for with energy_today_generation, and that could be because a "believable" number is taken, whereas for these it isn't, what do you think?

    - name: "solar_current"  
      state_topic: "Sofar2mqtt/state"
      unit_of_measurement: "A"
      device_class: current
      state_class: measurement
      value_template: "{{ value_json.solarPVAmps }}"

    - name: "inverter_battery_SOC"  
      state_topic: "Sofar2mqtt/state"
      unit_of_measurement: "%"
      device_class: battery
      state_class: measurement
      value_template: "{{ value_json.batterySOC }}"

    - name: "energy_today_generation"  
      state_topic: "Sofar2mqtt/state"
      unit_of_measurement: "kWh"
      device_class: energy
      state_class: total_increasing
      value_template: "{{ value_json.today_generation /100 }}"

springfall2008, 2024-08-30:
I don't think its an issue for Predbat, this relates to the Sofar integration. You could put some sort of filter on it maybe?

dbw1966, 2024-08-31:
Someone on FB pointed to an error in the code for sofar2mqtt. The last lines read

bool checkCRC(uint8_t frame[], byte frameSize)
{
unsigned int calculated_crc, received_crc;

received_crc = ((frame[frameSize-2] << 8) | frame[frameSize-1]);
calcCRC(frame, frameSize);
calculated_crc = ((frame[frameSize-2] << 8) | frame[frameSize-1]);
return (received_crc = calculated_crc);
}

And they should have read
bool checkCRC(uint8_t frame[], byte frameSize)
{
unsigned int calculated_crc, received_crc;

received_crc = ((frame[frameSize-2] << 8) | frame[frameSize-1]);
calcCRC(frame, frameSize);
calculated_crc = ((frame[frameSize-2] << 8) | frame[frameSize-1]);
return (received_crc == calculated_crc);
}

== is the comparitor that should have been used.

I will see if this fixes it.

springfall2008, 2025-11-26:
Stale ticket, please re-open if there is still an issue

closed 2025-11-26 by springfall2008