springfall2008/batpred #1414: Inverter using grid rather than battery during peak period

johnleeshore, 2024-08-31:
**Describe the bug**
I'm on Octopus Flux: Inverter is using grid to top up solar during peak hours of 4 to 7pm

**Expected behavior**
Expecting inverter to use battery rather than grid for topping up solar for house load during peak hours

**Predbat version**
8.4.1

**Environment details**
- GIV-HY3.6
- 9.5kWh Battery
- Standard HAOS installer
- Predbat 1.1.23

**Screenshots**
This is what the plan was:
[image]
This is what happened:
[image]
These are my settings:
[Predbat Web Interface.pdf](https://github.com/user-attachments/files/16825641/Predbat.Web.Interface.pdf)
**Log file**
[predbat.log](https://github.com/user-attachments/files/16825676/predbat.log)

springfall2008, 2024-09-01:
Sorry what do you mean "Grid to top up solar"?

johnleeshore, 2024-09-01:
Sorry I meant the inverter is using the grid for extra house load during the peak time rather than using the battery which I would have expected so that it wasn't using peak electricity from the grid.

springfall2008, 2024-09-01:
Where do you see it using grid, the plan only shows battery in Eco and exporting battery?

johnleeshore, 2024-09-01:
Yes the plan shows Eco and exporting battery but what happened (see graph below) was that the battery wasn't used and the inverter imported from the grid only. I would have expected any excess load to have come from the battery but the battery charge was maintained.
[image]

johnleeshore, 2024-09-01:
[image]

johnleeshore, 2024-09-01:
Although today it does seem to have worked as I thought it would:
[image]
I did uncomment
  export_limit:
    - 3600
in apps.yaml this morning. But I can't see this making a difference

johnleeshore, 2024-09-02:
I was just checking apps.yaml on web ui and not that the scheduled discharge enable entity points to a predbat sensor which is different to the other entities. Is this relevant?
[image]

johnleeshore, 2024-09-03:
Found out the issue. The Discharge 1 Setting in the Remote Control area was somehow set at 100% despite the slot times being 00.00. Changed it to 4% which has sorted the issue today.

closed 2024-09-03 by johnleeshore