springfall2008/batpred #1420: SunSynk Inverter Start/stop charge and discharge services not recognised

tmh88, 2024-09-01:
**Describe the bug**
SunSynk Inverter Start/stop charge and discharge services not recognised

Using the web UI to check the apps.yaml file is shows that the Start/stop charge and discharge services are not recognised.

[image]

apps.yaml

[image]

With the SunSynk inverter you change the mode with the following entity.

[image]

[image]

**Expected behavior**
When calling the service I would expect it to change the mode (value) in the entity

**Predbat version**

v8.4.4

**Environment details**

- SunSynk inverter
- Standard HAOS installer

savethemanual, 2024-11-24:
I've got the same issue, but I can't even see the work mode entity.
Does this get pulled in with the SunSynk Add-on?
I only have select.sunsynk_inverter_prog[1-6]_mode.
Any thoughts @tmh88 ?

Predbat: v8.7.2
Core: 2024.11.3
Supervisor: 2024.11.4
Operating System: 13.2
Sunsynk/Deye Inverter Add-on (multi) Current version: 0.7.5

closed 2025-02-10 by tmh88

springfall2008, 2025-02-11:
It's probably just an issue with the WebUI?