springfall2008/batpred #1422: New web interface v8.4.4, appdaemon relic warning

SwiftRR, 2024-09-02:
**Describe the bug**
A clear and concise description of what the bug is.
The log in the new web interface shows the following warning:

`21000 | 2024-09-02 09:17:45.903582: Warn: unable to find /addon_configs/6adb4f0d_predbat/appdaemon.yaml skipping checks as Predbat maybe running outside of AppDaemon`

**Expected behavior**
As v8.4.x doesn't run in appdaemon, the warning is superfluous and shouldn't be there.

**Predbat version**

v8.4.4

**Environment details**

- Inverter and battery setup: GE AC3.0 to two 8.2 kWh batteries
- Standard HAOS installer or Docker: HAOS on raspberry pi
- Anything else? None

gcoan, 2024-09-02:
This isn't a superfluous message as Predbat can still run under Appdaemon (and in fact I am running the latest release like that).

I have modified the message in a recent release to make it clearer that this is a warning that can be ignored if predbat isn't running under appdaemon. But if running in appdaemon then its valid as its part of the startup sanity checks predbat does

SwiftRR, 2024-09-02:
So does v8.4.x still run under appdaemon, but without the web UI? 
..... or have I just misinterpreted?

Rob

gcoan, 2024-09-02:
Yes 8.4.4 is running fine for me under appdaemon, it's just the web interface that isn't available.

I personally plan to move off appdaemon to the predbat add-on as that's the direction of travel, however at the moment I also run Octoblock under appdaemon which calculates the cheapest periods to run the washing machine overnight.  I've got complicated (messy) date/time manipulation logic to port over to the Octopus Integration target rate sensors and then I can shut appdaemon down.

gcoan, 2024-09-07:
Added explainer about this warning to the install doc

closed 2024-09-07 by gcoan