springfall2008/batpred #1424: Option to end iBoost early and replan, e.g. when hot water cylinder recovered

nbullus, 2024-09-02:
**Is your feature request related to a problem? Please describe.**
The **predbat_iboost_max_energy** entity controls the maximum amount of energy to plan for when suitably priced charge slots are available. On some days the maximum amount of energy may not actually be required, e.g. cylinder already 90% heated before iBoost triggers. Therefore iBoost will charge the remaining 10% and then stay on for the full duration. 

It can be more cost effective to end iBoost early in this situation by reducing the **predbat_iboost_max_energy** to the amount of energy actually used. This will then cause a re-plan which will drop the remaining charge slots it originally planned for iBoost charging and could result in a better more cost effective plan.

**Describe the solution you'd like**

Predbat to detect when iBoost power has stopped, e.g. immersion thermostat turned off, automatically adjust the max energy to what has been used, and a re-plan will then occur and remove the remaining scheduled iBoost. It is unclear whether Predbat can detect this or whether we would need to supply one of the following in the yaml:

1. Supply a power sensor for your iBoost device. Predbat can then detect when power drops to zero when iBoost is currently active.
2. Supply a binary sensor informing Predbat to end iBoost early for today.

Not everyone may want this feature so it would need to be controlled by a new switch to decide whether to end iBoost early. Also we would need to revert the iBoost max energy back at the start of day by using two entities instead of one:

- predbat_iboost_max_energy - maximum daily iBoost energy required
- predbat_iboost_target_energy - iBoost energy to target today

User would set the predbat_iboost_max_energy as they do now which would be used to set predbat_iboost_target_energy at 00:00. iBoost would then plan off predbat_iboost_target_energy. When instructed to finish early by the aforementioned binary/power sensor, Predbat would set predbat_iboost_target_energy to zero and trigger iBoost to end for the day. Entity predbat_iboost_target_energy would then be reset back to predbat_iboost_max_energy at the start of the next day.

**Describe alternatives you've considered**
I have amended my own iBoost automation to include this feature along with a number of other features making it quite a big and cumbersome automation now:

1. Turn off my gas hot water schedule when iBoost is planned to avoid conflicts and prevent the cylinder being heated by gas before iBoost is planned to start.
2. Restore gas hot water schedule once iBoost has completed or finished early.
3. Trigger hot water boost using gas if iBoost completes and the daily target has not been reached. This ensures the cylinder is topped up using gas.

**Additional context**

Following screenshot shows the control aspects of my own automation:

[image]

immersion_daily_target = my own helper holding the daily target, e.g. 9kWh
iBoost target = predbat_iboost_max_energy
End iBoost when cylinder recovered - my own helper for turning this feature on/off as required

My automation sets predbat_iboost_max_energy to zero when predbat_iboost_active = True, my switch is currently 'On' and my immersion power sensor drops to zero. At 00:00 I reset predbat_iboost_max_energy back to value of immersion_daily_target .

springfall2008, 2024-09-07:
Could you not just have an automation to set iboost_today to iboost_max?

nbullus, 2024-09-07:
That might be simpler to do if it triggers a replan to remove the remaining iboost slots for the remainder of the day. I'll give it a try. Thanks.

nbullus, 2024-09-10:
Tested this out today. Although iboost_today is now set to iboost_max_energy, iboost_today gets reset back to its previous value when it next re-plans and so locks out the future iboost slots again. Anything we can change here to stop it thinking it needs more iBoost?

dbw1966, 2024-09-14:
I had been thinking along similar lines, with a plan to use 3x  PT1000 temp sensors on the Eddi on the tank to estimate the energy required to heat it, and also to indicate when the tank is fully heated and have all future iboost sessions cancelled.  It may be some time until I get around to it but the capability you ask for would be interesting to have.

nbullus, 2024-09-14:
Chnages automation to set iboost_max_energy to iboost_today once cylinder is heated, and this seems to work and forces the replan to take out the remaining scheduled iboost slots. At midnight the automation has to reset iboost_max_energy back to the original daily target value. This will do until something better and more automated is considered.

nbullus, 2024-09-30:
Although updating iboost_max_energy is a way to trick predbat into cancelling the remaining iboost slots for the remainder of the day, it causes  a problem for the iboost forecast for tomorrow as it now thinks you want less energy. 

I think something else is needed to both preserve iboost_max_energy for future planning but also force Predbat to cancel remaining slots for today as this can be more cost effective.

closed 2026-08-28 by chalfontchubby