springfall2008/batpred #1427: Support for Dual AIO

arfond, 2024-09-02:
Here is the current REST output from a dual AIO gateway. I'm using the serial number for the Gateway (GW) in predbat. All the sensors that predbat needs exist, just some of them are not properly populated at the moment, like Battery Capacity and charge / discharge rate. These sensors are available in the individual AIO devices, so I was thinking I could refer PredBat to add those two up. To explain - with a dual AIO there are three invertors in the setup of GivTCP - the gateway and the individual AIOs.

So, I was thinking in the absence of this data in the REST output of the Gateway (for now) - I'd create it in apps.yaml. Something like:

battery_capacity_kwh: (if this is the thing PredBat is trying to read from the REST at startup
    - sensor.givtcp_CH2330G294_battery_capacity_kwh + sensor.givtcp2_CH2414G514_battery_capacity_kwh

Gateway REST output

{ "Control": { "Active_Power_Rate": 100, "Battery_Calibration": "Off", "Battery_Charge_Rate": 0, "Battery_Charge_Rate_AC": 70, "Battery_Discharge_Rate": 0, "Battery_Discharge_Rate_AC": 75, "Battery_Power_Cutoff": 0, "Battery_Power_Reserve": 4, "Battery_pause_mode": "Disabled", "Charge_Target_SOC_1": 100, "Charge_Target_SOC_10": 100, "Charge_Target_SOC_2": 100, "Charge_Target_SOC_3": 100, "Charge_Target_SOC_4": 100, "Charge_Target_SOC_5": 100, "Charge_Target_SOC_6": 100, "Charge_Target_SOC_7": 100, "Charge_Target_SOC_8": 100, "Charge_Target_SOC_9": 100, "Discharge_Target_SOC_1": 4, "Discharge_Target_SOC_10": 0, "Discharge_Target_SOC_2": 0, "Discharge_Target_SOC_3": 0, "Discharge_Target_SOC_4": 0, "Discharge_Target_SOC_5": 0, "Discharge_Target_SOC_6": 0, "Discharge_Target_SOC_7": 0, "Discharge_Target_SOC_8": 0, "Discharge_Target_SOC_9": 0, "Eco_Mode": "disable", "Enable_Charge_Schedule": "disable", "Enable_Discharge_Schedule": "disable", "Force_Charge": "Normal", "Force_Charge_Num": 0, "Force_Export": "Normal", "Force_Export_Num": 0, "Mode": "Eco (Paused)", "Reboot_Addon": "disable", "Reboot_Invertor": "disable", "Sync_Time": "disable", "Target_SOC": 100, "Temp_Pause_Charge": "Running", "Temp_Pause_Charge_Num": 43, "Temp_Pause_Discharge": "Normal", "Temp_Pause_Discharge_Num": 0 }, "Energy": { "Today": { "Battery_Charge_Energy_Today_kWh": 30.7, "Battery_Discharge_Energy_Today_kWh": 28.4, "Export_Energy_Today_kWh": 24.9, "Import_Energy_Today_kWh": 22.6, "Load_Energy_Today_kWh": 12.3, "PV_Energy_Today_kWh": 20.7, "Parallel_Total_Charge_Energy_Today_kWh": 32.3, "Parallel_Total_Discharge_Energy_Today_kWh": 25.7 }, "Total": { "Battery_Charge_Energy_Total_kWh": 3.04, "Battery_Discharge_Energy_Total_kWh": 2.97, "Export_Energy_Total_kWh": 4.1, "Import_Energy_Total_kWh": 3.58, "Load_Energy_Total_kWh": 3.08, "PV_Energy_Total_kWh": 3.37, "Parallel_Total_Charge_Energy_Total_kWh": 3.57, "Parallel_Total_Discharge_Energy_Total_kWh": 2.76 } }, "GW2330G276": { "Battery_Capacity_kWh": 0.0, "Export_Limit": 11000, "Gateway_Mode": "On grid", "Gateway_Software_Version": "GAAA0010", "Gateway_State": "Static", "Invertor_Serial_Number": "GW2330G276", "Invertor_Time": "2024-09-02T21:31:11+01:00", "Invertor_Type": "Gen 1 - Gateway", "Parallel_Total_AIO_Number": 2, "Parallel_Total_AIO_Online_Number": 2 }, "Inverters": { "AIO_1": { "AC_Charge_Energy_Today_kWh": 15.4, "AC_Charge_Energy_Total_kWh": 3.0, "AC_Discharge_Energy_Today_kWh": 14.3, "AC_Discharge_Energy_Total_kWh": 2.92, "AIO_1_Serial_Number": "CH2330G294", "Invertor_Power": -59, "SOC": 4 }, "AIO_2": { "AC_Charge_Energy_Today_kWh": 15.3, "AC_Charge_Energy_Total_kWh": 0.05, "AC_Discharge_Energy_Today_kWh": 14.1, "AC_Discharge_Energy_Total_kWh": 0.05, "AIO_2_Serial_Number": "CH2414G514", "Invertor_Power": -76, "SOC": 4 } }, "Meter_Details": { "Meter_ID1": { "Export_Energy_kWh": 0.0, "Frequency": 5.0, "Import_Energy_kWh": 0.0, "Phase_1_Current": 12.6, "Phase_1_Power": 3038, "Phase_1_Power_Factor": 0.0, "Phase_1_Voltage": 246.6, "Phase_2_Current": 0.0, "Phase_2_Power": 0, "Phase_2_Power_Factor": 0.0, "Phase_2_Voltage": 0.0, "Phase_3_Current": 0.0, "Phase_3_Power": 0, "Phase_3_Power_Factor": 0.0, "Phase_3_Voltage": 0.0 } }, "Power": { "Power": { "Battery_Power": 0, "Grid_Current": 12.6, "Grid_Power": -2987, "Grid_Relay_Voltage": 243.3, "Grid_Voltage": 243.7, "Inverter_Relay_Voltage": 243.3, "Invertor_Power": -136, "Liberty_Power": -223, "Load_Current": 13.0, "Load_Power": 3084, "Load_Voltage": 244.0, "PV_Current": 1.3, "PV_Power": 320, "SOC": 4, "SOC_kWh": 0.0 } }, "Stats": { "GivTCP_Version": "2.4.739beta", "Last_Updated_Time": "2024-09-02T20:33:04.701754+00:00", "Time_Since_Last_Update": 31.727202, "status": "online" }, "Timeslots": { "Battery_pause_end_time_slot": "00:00:00", "Battery_pause_start_time_slot": "00:00:00", "Charge_end_time_slot_1": "15:52:00", "Charge_end_time_slot_10": "00:00:00", "Charge_end_time_slot_2": "00:00:00", "Charge_end_time_slot_3": "00:00:00", "Charge_end_time_slot_4": "00:00:00", "Charge_end_time_slot_5": "00:00:00", "Charge_end_time_slot_6": "00:00:00", "Charge_end_time_slot_7": "00:00:00", "Charge_end_time_slot_8": "00:00:00", "Charge_end_time_slot_9": "00:00:00", "Charge_start_time_slot_1": "12:52:00", "Charge_start_time_slot_10": "00:00:00", "Charge_start_time_slot_2": "00:00:00", "Charge_start_time_slot_3": "00:00:00", "Charge_start_time_slot_4": "00:00:00", "Charge_start_time_slot_5": "00:00:00", "Charge_start_time_slot_6": "00:00:00", "Charge_start_time_slot_7": "00:00:00", "Charge_start_time_slot_8": "00:00:00", "Charge_start_time_slot_9": "00:00:00", "Discharge_end_time_slot_1": "19:00:00", "Discharge_end_time_slot_10": "00:00:00", "Discharge_end_time_slot_2": "00:00:00", "Discharge_end_time_slot_3": "00:00:00", "Discharge_end_time_slot_4": "00:00:00", "Discharge_end_time_slot_5": "00:00:00", "Discharge_end_time_slot_6": "00:00:00", "Discharge_end_time_slot_7": "00:00:00", "Discharge_end_time_slot_8": "00:00:00", "Discharge_end_time_slot_9": "00:00:00", "Discharge_start_time_slot_1": "16:00:00", "Discharge_start_time_slot_10": "00:00:00", "Discharge_start_time_slot_2": "00:00:00", "Discharge_start_time_slot_3": "00:00:00", "Discharge_start_time_slot_4": "00:00:00", "Discharge_start_time_slot_5": "00:00:00", "Discharge_start_time_slot_6": "00:00:00", "Discharge_start_time_slot_7": "00:00:00", "Discharge_start_time_slot_8": "00:00:00", "Discharge_start_time_slot_9": "00:00:00" }, "raw": { "batteries": {}, "invertor": { "active_power_rate": 100, "aio1_serial_number": "", "aio1_serial_number_new": "CH2330G294", "aio1_soc": 4, "aio2_serial_number": "CH23", "aio2_serial_number_new": "CH2414G514", "aio2_soc": 4, "aio3_serial_number": "94CH24", "aio3_serial_number_new": "", "aio3_soc": 0, "aio_state": 0, "arm_firmware_version": 0, "battery_charge_limit": 0, "battery_charge_limit_ac": 70, "battery_discharge_limit": 0, "battery_discharge_limit_ac": 75, "battery_discharge_min_power_reserve": 0, "battery_errors": [], "battery_firmware_version": 612, "battery_high_voltage_protection_limit": 0.0, "battery_low_force_charge_time": 0, "battery_low_voltage_protection_limit": 0.0, "battery_nominal_capacity": 0.0, "battery_pause_mode": 0, "battery_pause_slot_1": "TimeSlot(start=datetime.time(0, 0), end=datetime.time(0, 0))", "battery_pause_slot_1_end": 0, "battery_pause_slot_1_start": 0, "battery_percent": 0, "battery_soc_reserve": 4, "battery_type": 0, "battery_voltage_adjust": 0.0, "bms_firmware_version": 0, "certification_type": 12, "charge_slot_1": "TimeSlot(start=datetime.time(12, 52), end=datetime.time(15, 52))", "charge_slot_10": "TimeSlot(start=datetime.time(0, 0), end=datetime.time(0, 0))", "charge_slot_10_end": 0, "charge_slot_10_start": 0, "charge_slot_1_end": 1552, "charge_slot_1_start": 1252, "charge_slot_2": "TimeSlot(start=datetime.time(0, 0), end=datetime.time(0, 0))", "charge_slot_2_end": 0, "charge_slot_2_start": 0, "charge_slot_3": "TimeSlot(start=datetime.time(0, 0), end=datetime.time(0, 0))", "charge_slot_3_end": 0, "charge_slot_3_start": 0, "charge_slot_4": "TimeSlot(start=datetime.time(0, 0), end=datetime.time(0, 0))", "charge_slot_4_end": 0, "charge_slot_4_start": 0, "charge_slot_5": "TimeSlot(start=datetime.time(0, 0), end=datetime.time(0, 0))", "charge_slot_5_end": 0, "charge_slot_5_start": 0, "charge_slot_6": "TimeSlot(start=datetime.time(0, 0), end=datetime.time(0, 0))", "charge_slot_6_end": 0, "charge_slot_6_start": 0, "charge_slot_7": "TimeSlot(start=datetime.time(0, 0), end=datetime.time(0, 0))", "charge_slot_7_end": 0, "charge_slot_7_start": 0, "charge_slot_8": "TimeSlot(start=datetime.time(0, 0), end=datetime.time(0,  [... cut ...] .0, "i_phase_1": 12.6, "i_phase_2": 0.0, "i_phase_3": 0.0, "i_total": 0.0, "p_active_phase_1": 3038, "p_active_phase_2": 0, "p_active_phase_3": 0, "p_active_total": 0, "p_apparent_phase_1": 0, "p_apparent_phase_2": 0, "p_apparent_phase_3": 0, "p_apparent_total": 0, "p_reactive_phase_1": 0, "p_reactive_phase_2": 0, "p_reactive_phase_3": 0, "p_reactive_total": 0, "pf_phase_1": 0.0, "pf_phase_2": 0.0, "pf_phase_3": 0.0, "pf_total": 0.0, "v_phase_1": 246.6, "v_phase_2": 0.0, "v_phase_3": 0.0 } } } }

springfall2008, 2024-09-02:
Interesting, so we have something similar with EMS so it maybe feasible to apply that approach. 

What is controlled via the Gateway and what is controlled via the individual inverters?

gcoan, 2024-09-02:
Are you running the latest Beta version of GivTCP?  It has enhanced support for multiple AIO's including I think deriving the correct battery SoC.

Have a look at the GivTCP Github issues, there's one that's been dealing with getting it working for AIO's

arfond, 2024-09-02:
I'm running the latest Dev 2.4.745

arfond, 2024-09-02:
Yep, PredBat has started!

[image]

arfond, 2024-09-03:
> Interesting, so we have something similar with EMS so it maybe feasible to apply that approach. 
> 
> What is controlled via the Gateway and what is controlled via the individual inverters?

Nothing via the individual AIOs (strict instructions from Paul not to mess with the individual AIOs). All done via the GW, which I have checked has all the sensors listed in the apps.yaml that a single one has. So, I'm thinking simply pointing PredBat to the gateway might work once a couple of sensors are ironed out (battery capacity now sorted, last remaining are battery power and battery charge/discharge rate). I'm hoping those will be sorted before end of the week so I can try PredBat out on the gateway.

Here is my current error, as mentioned this is currently missing.

[image]

arfond, 2024-09-03:
Interesting exchange with Mark just then, as I noticed my gateway was no longer providing data from GivTCP since last night. I worked out is was PredBat running on the background - when I stopped it (running in monitor mode) the gateway data started flowing fine. I think this is due to the missing charge / discharge sensor - but strange that one add-on not running correctly can disrupt another like that.

[image]

[image]

arfond, 2024-09-04:
@springfall2008 - I think it's up and running. The only intervention I've made from the default is to point battery_power to a sensor I've created. That data isn't available at the gateway yet, Paul suggested it's not coming anytime soon, so I created a sensor in HA that combines the individual AIO battery sensors.

Before I activate for the 1st time can I ask how the export limit works. I've added my DNO limit in apps.yaml. This is also commissioned in the gateway, but rather poor the gateway shuts off for a minute if export goes above that.... I ask as I noticed as soon as I ran PredBat it changed my export rate to 100% or 12kW. Once I enable I expect PredBat will see an opportunity to export... Will it limit discharge to the export limit (taking into account predicted load and PV)?

[image]

I'll report back once it's running. I know fairly quickly if it's good or not.

Also, if PredBat switch is off in control does it mean it doesn't action any of the plan but I can see the potential outcome of a plan?

gcoan, 2024-09-04:
> Before I activate for the 1st time can I ask how the export limit works. I've added my DNO limit in apps.yaml. This is also commissioned in the gateway, but rather poor the gateway shuts off for a [...]

My understanding (per the documentation https://springfall2008.github.io/batpred/apps-yaml/#inverter-control-configurations) is that the export limit in apps.yaml is purely used by Predbat to model the max export you can achieve, and take that into account in the forward plan/cost projection.

Predbat relies on the inverter to actually manage any export limit set by the DNO as this has to be done in realtime based upon house load, solar generation,etc.

You could reduce the max discharge rate of the inverter in apps.yaml to avoid the inverter triggering the export limit.

arfond, 2024-09-04:
Thank you. I've tried doing that via adding a number in apps.yaml but I get an exception 

[image]

Is there somewhere else I should do this. The scaling factor in PredBat I think will do what you said above - won't impact the inverter.

> My understanding (per the documentation https://springfall2008.github.io/batpred/apps-yaml/#inverter-control-configurations) is that the export limit in apps.yaml is purely used by Predbat to model [...]

gcoan, 2024-09-04:
you've configured it wrongly

charge_rate and discharge_rate in apps.yaml should point to the inverter sensors that control the charge and discharge rate

you need to set inverter_limit_discharge per the documentation to your discharge limit.  

Bear in mind that this just limits the battery discharge rate so if you have solar generation at the same time, it'll still exceed your export limit - which is why export limit has to be implemented in the inverter.  This is a workaround

arfond, 2024-09-04:
Thank you! That's well understood regarding the workaround. I've reduced max discharge by 3kW for now which is my max solar output during the flux export window.

I can confirm that dual AIO is working perfectly with PredBat. I've had a week without being able to use it since the 2nd AIO install - I've missed it looking after things.

Thanks again.

As noted above - the only workaround is creating a battery power sensor.

gcoan, 2024-09-04:
Excellent news

Can you p[rovide a brief summary of what's needed, what software version, config, workaround template helpers, how to configure apps.yaml etc and I can add it to the documentation.

Great work from @britkat1980

arfond, 2024-09-05:
> Excellent news
> 
> Can you p[rovide a brief summary of what's needed, what software version, config, workaround template helpers, how to configure apps.yaml etc and I can add it to the documentation.
> 
> Great work from @britkat1980 

Yes, very happy to. I'll do that once V3 is out. It's possible the missing battery_power sensor data will be populated (that would be an approximation if @britkat1980 does add it). If it's added it's a simple point to the gateway, if not a little extra guidance needed on GivTCP setup and a new sensor and apps.yaml.

Thank you all again for all your excellent work.

arfond, 2024-09-07:
> Excellent news
> 
> Can you p[rovide a brief summary of what's needed, what software version, config, workaround template helpers, how to configure apps.yaml etc and I can add it to the documentation.
> 
> Great work from @britkat1980 

As an update. Latest V3 beta and dev now includes the missing battery power sensor, therefore there are now no workarounds. The only thing I note in the documentation dual dual / parallel AIO users is to only use the gateway sensors, not the individual AIO ones. It's been working perfectly since my last post, no issues at all.

gcoan, 2024-09-07:
Brilliant thanks, I'll add that point in about using the gateway

gcoan, 2024-09-07:
I've added a reference to this to the latest documentation.  More signposting and detail of the givtcp & mqtt install is needed, but that's already on my to-do list

gcoan, 2024-09-21:
documentation updated

closed 2024-09-21 by gcoan