springfall2008/batpred #1429: Longstanding issue with IOG + Predbat around midnight

PianSom, 2024-09-03:
I am on Intelligent Octopus Go, which means that my low rate starts at 11.30pm. For a long time (many months, possible forever) Predbat seems not keen to acknowledge this, and does some discharging between 11.30pm and midnight.

This is especially noticeable today, a low solar generation combined with a high usage day. The plan below shows the battery being emptied by around 10pm, and then Predbat sceduling a charge and discharge cycle in that half hour, before reverting to more expected, rational behaviour. 

I am at a loss to work out what to change to address this. Any suggestions?

[image]

mpartington, 2024-09-04:
Not sure I see an issue, it's maximizing profit with the 23.30 slot (buy 7p, sell 15p excluding losses). You can't discharge before 23.30 as your battery is empty and still end up with a 97% full battery at the end of the cheap period.  Probably can't make 100% as the charge rate tails of as 100% approaches.  Presumably you hit 100% later in the plan, or have enough to get through to next charge.

If you don't like this behaviour (discharge in a charge slot), I think it's this switch that needs to be turned off

switch.predbat_calculate_discharge_oncharge

PianSom, 2024-09-04:
My point is that odd behaviour is limited to the 23.30-00.00 slot. It does not consistently display it in other slots, whereas it does show it consistently then.

Today's plan is a more normal one. I would expect Predbat to start discharge earlier than it is planning to, and then start charging at 11.30pm. But it never plans that. (Doing so would increase profit, since it would have 30 mins extra of low cost import.)  Actually, in practise, when the time arrives it always (almost always?) does a charge then quick discharge cycle in that slot before settling down.

(I have wondered whether this effect is something to do with the fact that standing charge is applied just after midnight, but haven't seen anything to support that.)

[image]

mpartington, 2024-09-04:
Got you, yes that second screenshot does show an oddity. The first one seemed normal.
Unfortunately I'm still waiting for IOG compatibility 😔, so not something I see.  Sorry no real advice, but agree with your summary.

PianSom, 2024-09-05:
I am now pretty convinced that this is some unexpected (by me, at least) behaviour to do with Combine charge slots and Combine discharge slots - both of which I have enabled

springfall2008, 2024-09-07:
I'd suggest turning off combine charge/combine discharge for IOG

PianSom, 2024-09-07:
Thanks Trefor 

I’ve found that switching them on gives a slightly better economic  result, and - I think more importantly? - gives a plan which is “kinder” on the battery. 

“On” results in a plan which has (broadly) just one charge-discharge-charge overnight. Whereas “off” cycles  pretty much every half hour through the night. This can’t be good for the battery chemistry every day?

I will follow your suggestion for a few days, see how I get on.

springfall2008, 2024-09-09:
You could probably turning off 'calculate discharge during charge'?

PianSom, 2024-09-09:
So turning off the "Combine charge/discharge slots" options solves the 23:30-00:00 issues, but results in an overnight repeating charge/discharge cycles at a high SoC - like this:
[image]

A recent video commentary [here](https://www.youtube.com/watch?v=w1zKfIQUQ-s) highlighted research showing that the worst charging regimen modelled for LFP cells is exactly this - short cycling at a high SoC! So I think I will leave the options on, and live with the odd half hour behaviour.

Thanks for the suggestion anyway, Trefor

PianSom, 2024-09-09:
Oops - crossed over. Thanks for the additional suggestion. Let me try that.

sammort, 2024-09-13:
Personally I think one longer discharge is better, especially for battery health. Would it be possible to adjust the modelling so if a long period exists (like on off-peak tarriffs such as IOG) then charging and discharging is combined in that particular scenario?

I also think that with tariffs like IOG, with long periods of off-peak pricing, where Predbat is planning to charge/discharge/charge the battery, it shouldn't be intending to take the battery to 100%, because of the slower charging rates at higher SOC levels. Wouldn't it be optimal to take in to account the `battery_charge_power_curve` and `battery_discharge_power_curve` and not charge/discharge in those SOCs which do not provide 100% of power @springfall2008?

PianSom, 2024-09-18:
I am finding that - as suggested by another IOG user with the same issue - manually setting a discharge period 23:30-00:00 solves this problem effectively, and results in a more rationale charging schedule. I will put in place a HA automation to do this every day going forward.

closed 2025-01-01 by springfall2008