springfall2008/batpred #1431: Huawei Inverter Grid Charge Rate

UKHarj, 2024-09-05:
In the apps.yaml I have specified the charge/discharge rate as 3 kwh. However this is to force the plan to ensure it calculates this charging rate in the plan. My battery supports DC charging at 5 kwh and discharging at 5 kwh. Note the 3 kwh AC charging is apparently the limit allowed in the UK and is set up by my installer. I've confirmed this in other Huawei forums, so in summary AC charging is limited to 3kwh.

The 3 overrides the DC charging and when the sun is shining limits charging of the battery to 3 kwh,. 

I would like the config to be amended so that grid charging is set at 3 kwh and the DC charging/charging is left at 5 kwh. 

[image]

gcoan, 2024-09-05:
> Note the 3 kwh AC charging is apparently the limit allowed in the UK and is set up by my installer

Not quite sure I follow what the issue with Predbat is,  but on the point above about 3kWh AC charging limit,  there isn't any such UK charging limit to my knowledge.  I have two 2.6kWh inverters and can charge both batteries at 5.2kWh import,  and if the oven, washing machine, etc is on, I've seen 12kWh being imported before now.

It may be that your DNO fuse is quite small and that's what the installer is referring to?  My fuse is 100A, and if yours is substantially smaller then you can get the DNO to replace it.

But 3kWh sounds way too low.  If you put the oven on and the kettle at the same time you will draw more than 3kWh from the grid so I really don't know what your installer is referring to.

There are grid export limits and if you didn't get a G99 approval then you will be limited to 3.6kWh of export, but that's not related to import.

UKHarj, 2024-09-05:
Thanks for coming back to me.

I have a 100A fuse so it's not an issue with my electrics. The 3kwh limit
is applied by Huawei. It's a 10kwh battery and at 3kwh charge limit, it's
take 3.3hrs to charge. I can change the battery from sunlight at 5 kWh and
discharge at 5 kWh. However I'm the yaml I need to set the 3 kWh rate, when
I set this it overrides my DC rates and reduces them to 3 kWh.

gcoan, 2024-09-05:
Can you share your apps.yaml config to see if there's anything in the way you've set these limits?

UKHarj, 2024-09-05:
# ------------------------------------------------------------------
# Huawei Solar Config example - https://github.com/wlcrs/huawei_solar
# ------------------------------------------------------------------
---
pred_bat:
  module: predbat
  class: PredBat

  # Sets the prefix for all created entities in HA - only change if you want to run more than once instance
  prefix: predbat

  # Timezone to work in
  timezone: Europe/London

  # If you are using Predbat outside of HA then set the HA URL and Key (long lived access token here)
  #ha_url: 'http://homeassistant.local:8123'
  #ha_key: 'xxx'

  # Currency, symbol for main currency second symbol for 1/100s e.g. $ c or £ p or e c
  currency_symbols:
   - '£'
   - 'p'

  # Number of threads to use in plan calculation
  # Can be auto for automatic, 0 for off or values 1-N for a fixed number
  threads: 4

  # Sensors, more than one can be specified and they will be summed up automatically
  #
  # For two inverters the load today would normally be the master load sensor only (to cover the entire house)
  # If you have three phase and one inverter per phase then you would need three load sensors
  #
  # For pv_today if you have multiple solar inverter inputs then you should include one entry for each inverter
  #
  load_today:
    - sensor.consumption_today
  import_today:
    - sensor.import_today
  export_today:
    - sensor.export_today
  pv_today:
    - sensor.solar_energy_riemann

  # Load forecast can be used to add to the historical load data (heat-pump)
  # To link to Predheat
  # Data must be in the format of 'last_updated' timestamp and 'energy' for incrementing kWh
  #load_forecast:
  #  - predheat.heat_energy$external
  #
  num_inverters: 1
  inverter_type: "HU"
  #
  # Run balance inverters every N seconds (0=disabled) - only for multi-inverter
  balance_inverters_seconds: 0
  #
  # If not using REST then instead set the Control here (one for each inverter)
  # - you can delete this section if using REST
  charge_rate:
    - number.batteries_maximum_charging_power
  discharge_rate:
    - number.batteries_maximum_discharging_power
  battery_power:
    - sensor.batteries_charge_discharge_power
  pv_power:
    - sensor.inverter_input_power
  load_power:
    - sensor.power_meter_active_power
  soc_percent:
    - sensor.batteries_state_of_capacity
  #soc_kw:
[... 227 lines cut ...]
  # Battery scaling makes the battery smaller (e.g. 0.9) or bigger than its reported
  # If you have an 80% DoD battery that falsely reports it's kwh then set it to 0.8 to report the real figures
  battery_scaling: 1.0

  # Can be used to scale import and export data, used for workarounds
  import_export_scaling: 1.0

  # Export triggers:
  # For each trigger give a name, the minutes of export needed and the energy required in that time
  # Multiple triggers can be set at once so in total you could use too much energy if all run
  # Creates an entity called 'binary_sensor.predbat_export_trigger_<name>' which will be turned On when the condition is valid
  # connect this to your automation to start whatever you want to trigger
  export_triggers:
     - name: 'large'
       minutes: 60
       energy: 1.0
     - name: 'small'
       minutes: 15
       energy: 0.25

  # If you have a sensor that gives the energy consumed by your solar diverter then add it here
  # this will make the predictions more accurate. It should be an incrementing sensor, it can reset at midnight or not
  # It's assumed to be in Kwh but scaling can be applied if need be
  #iboost_energy_today: 'sensor.tasmota_energy_today'
  #iboost_energy_scaling: 1.0 `

gcoan, 2024-09-05:
Its very hard to follow the file pasted in like that as github has assumed some of the line with # are headings rather than comments.

Can you either edit the comment above and put ``` on a line before and after the apps.yaml which means it will be treated as text and be much more readable,  or upload the file as an attachment to github (probably need to rename it to a .txt or .log extension first)

UKHarj, 2024-09-05:
[apps.yaml.txt](https://github.com/user-attachments/files/16899187/apps.yaml.txt)

UKHarj, 2024-09-05:
if it helps here is a screenshot from the Huawei software. The battery has setting for charge and discharge through to 5,000 kwh, however from AC the chare is limited to 3,000. Ideally the apps.yaml could have two entries, one for AC charge and one for DC. That way the plan would know what the AC charge will run at 3,000 kwh and the DC could charge at 5,000 and also discharge to loads at 5,000 maximum. 

[image]

nargotik, 2024-12-23:
Any update on this ?

I have maximum power from grid of 5kw and I want to set huawei max charge of 1kw to let 4kw left to other things ....

gcoan, 2025-11-15:
At the moment there isn't a facility in predbat to specify different grid AC and inverter DC limits so predbat could plan and model what you are asking for.

Would be useful though @springfall2008 ?

xxxxminingxxxx, 2025-11-16:
@ward0 

For now i start the charging via a script that charges the battery at 5kW (without the script it is 10kW). Predbat continuously adjusts the plan for this difference, but it is not ideal.

  #charge_start_service: huawei_solar/forcible_charge_soc
  **charge_start_service: script.batterij_laden_vanaf_grid**
  charge_stop_service: huawei_solar/stop_forcible_charge
  discharge_start_service: huawei_solar/forcible_discharge_soc
  discharge_stop_service: huawei_solar/stop_forcible_charge

Script:

batterij_laden_vanaf_grid:
  alias: Batterij laden vanaf grid
  sequence:
  - data:
      target_soc: 100
      power: '5000'
      device_id: 06e8a263ce238d92c851efec9ed6e4d9
    action: huawei_solar.forcible_charge_soc

ward0, 2025-11-16:
> [@ward0](https://github.com/ward0)
>
> For now i start the charging via a script that charges the battery at 5kW (without the script it is 10kW). Predbat continuously adjusts the plan for this [...]

@xxxxminingxxxx 
Thank you for your time and help.
I created the script, afterwards i changed the apps.yaml:

41288 | 2025-11-16 21:26:06.612790: Warn: Validation of apps.yaml found configuration item 'discharge_stop_service' element huawei_solar/stop_forcible_charge is not an dict
-- | --
41287 | 2025-11-16 21:26:06.612739: Warn: Validation of apps.yaml found configuration item 'discharge_start_service' element huawei_solar/forcible_discharge_soc is not an dict
41286 | 2025-11-16 21:26:06.612686: Warn: Validation of apps.yaml found configuration item 'charge_stop_service' element huawei_solar/stop_forcible_charge is not an dict
41285 | 2025-11-16 21:26:06.612604: Warn: Validation of apps.yaml found configuration item 'charge_start_service' element script.batterij_laden_vanaf_grid is not an dict

[image]

[image]

[image]

gcoan, 2025-11-16:
@ward0 these validation errors on the charge/discharge start/stop service can be ignored

They've already been logged on #2670 and I've been trying, without success so far, to find a format that predbat likes.  Its only  a UI validation warning.

Setting a script shouldn't cause a warning either 😠

ward0, 2025-11-16:
@xxxxminingxxxx 

44299 | 2025-11-16 22:30:21.297922: Warn: Service call huawei_solar/stop_forcible_charge data {'device_id': ''} failed with response None

Do you also get this error when the planned Predbat charge needs to stop?

Do i also need a script for this?

Update:

I forgot to uncomment my device_id: xxx value.