springfall2008/batpred #1432: Upgrade to Predbat Add-on Causes unexpected discharge behaviour

rknight20772, 2024-09-05:
**Describe the bug**
Prior to installing the Predbat Add-on, I was using the Appdeamon version. When controlling Charge and Discharge, the battery would stay charged throughout the day, and then usually there would be a discharge slots starting around 10pm, that discharged the battery to 4% by 1130pm, ready to charge again during the intelligent slots from 1130pm-530am.
Since updating to the Add-on, the discharge slots are now happening throughout the day, without enough battery left now to get to the 1130pm slot. This happened yesterday too.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Predbat version**

8.4.4

**Environment details**

GivEnergy 5KW hybrid inverter with 9.5Kwh battery
HA is on a VM 

**Screenshots**
If applicable, add screenshots to help explain your problem. The most useful ones can be your battery chart, the Predbat HTML plan and your current settings in HA.

**Log file**
```
2024-09-05 12:00:18.113139: Info: record_status Freeze discharging target 61%-99%
2024-09-05 12:00:35.315550: Info: record_status Idle
2024-09-05 12:05:31.519447: Info: record_status Freeze discharging target 61%-99%
2024-09-05 12:10:07.011992: Info: record_status Freeze discharging target 61%-99%
2024-09-05 12:15:26.274687: Info: record_status Idle
2024-09-05 12:20:52.881302: Info: record_status Discharging target 61%-49%
2024-09-05 12:26:07.819633: Info: record_status Idle
2024-09-05 12:30:26.934166: Info: record_status Freeze discharging target 57%-99%
2024-09-05 12:35:16.679930: Info: record_status Freeze discharging target 57%-99%
2024-09-05 12:40:05.141706: Info: record_status Freeze discharging target 57%-99%
2024-09-05 12:45:17.074651: Info: record_status Freeze discharging target 57%-99%
2024-09-05 12:50:05.582945: Info: record_status Freeze discharging target 57%-99%
2024-09-05 12:55:58.028756: Info: record_status Discharging target 57%-48%
2024-09-05 13:01:09.511184: Info: record_status Freeze discharging target 54%-99%
2024-09-05 13:05:29.549134: Info: record_status Idle
2024-09-05 13:10:07.001922: Info: record_status Idle
2024-09-05 13:15:18.403592: Info: record_status Idle
2024-09-05 13:20:06.851875: Info: record_status Idle
2024-09-05 13:25:16.350880: Info: record_status Idle
2024-09-05 13:30:15.294361: Info: record_status Freeze discharging target 53%-99%
2024-09-05 13:35:16.541784: Info: record_status Freeze discharging target 53%-99%
2024-09-05 13:40:06.966588: Info: record_status Freeze discharging target 53%-99%
2024-09-05 13:45:18.496703: Info: record_status Freeze discharging target 53%-99%
2024-09-05 13:50:05.883815: Info: record_status Freeze discharging target 53%-99%
2024-09-05 13:55:17.083469: Info: record_status Freeze discharging target 53%-99%
2024-09-05 14:00:05.783671: Info: record_status Freeze discharging target 53%-99%
2024-09-05 14:05:16.117155: Info: record_status Freeze discharging target 53%-99%
2024-09-05 14:10:05.517857: Info: record_status Freeze discharging target 53%-99%
2024-09-05 14:15:16.708938: Info: record_status Freeze discharging target 53%-99%
2024-09-05 14:20:06.319304: Info: record_status Freeze discharging target 53%-99%
2024-09-05 14:25:26.431159: Info: record_status Idle
2024-09-05 14:30:15.151284: Info: record_status Freeze discharging target 53%-99%
2024-09-05 14:35:16.567282: Info: record_status Freeze discharging target 53%-99%
2024-09-05 14:40:07.051314: Info: record_status Freeze discharging target 53%-99%
2024-09-05 14:45:17.402689: Info: record_status Freeze discharging target 53%-99%
2024-09-05 14:50:06.015304: Info: record_status Freeze discharging target 53%-99%
2024-09-05 14:55:16.835634: Info: record_status Freeze discharging target 52%-99%
2024-09-05 15:00:06.446994: Info: record_status Freeze discharging target 52%-99%
2024-09-05 15:05:16.680964: Info: record_status Freeze discharging target 52%-99%
2024-09-05 15:10:05.318591: Info: record_status Freeze discharging target 52%-99%
2024-09-05 15:15:17.609316: Info: record_status Freeze discharging target 52%-99%
2024-09-05 15:20:05.068044: Info: record_status Freeze discharging target 52%-99%
2024-09-05 15:25:16.413202: Info: record_status Freeze discharging target 52%-99%
2024-09-05 15:30:06.033414: Info: record_status Freeze discharging target 52%-99%
2024-09-05 15:35:15.840938: Info: record_status Freeze discharging target 52%-99%
2024-09-05 15:40:05.542859: Info: record_status Freeze discharging target 52%-99%
2024-09-05 15:45:31.576884: Info: record_status Idle
2024-09-05 15:50:07.804727: Info: record_status Idle
2024-09-05 15:55:17.288116: Info: record_status Idle
2024-09-05 16:00:05.761565: Info: record_status Idle
2024-09-05 16:05:15.528082: Info: record_status Idle
2024-09-05 16:10:07.067275: Info: record_status Idle
2024-09-05 16:15:17.196522: Info: record_status Idle
2024-09-05 16:20:05.579573: Info: record_status Idle
2024-09-05 16:25:55.557923: Info: record_status Discharging target 52%-43%
Trigger callback for select select_option
2024-09-05 16:29:56.457210: Info: record_status Idle
2024-09-05 16:30:06.622688: Info: record_status Idle
Trigger callback for select select_option
2024-09-05 16:30:43.773610: Info: record_status Idle
2024-09-05 16:35:05.453424: Info: record_status Idle
```

gcoan, 2024-09-05:
The predbat code is exactly the same whether it runs in appdaemon or the predbat add-on.

If you are getting different results then it's likely a problem with either your apps.yaml or your predbat configuration.

Assume you just copied your appdaemon apps.yaml across over the top of the template apps.yaml supplied with the predbat addon?

Can you try shutting down the predbat add-on and starting appdaemon up, 8.4.4 works just fine in appdaemon, you just don't have the new web interface.
Then running under appdaemon, see what the results are.

Then it will probably be a matter of trying to find what configuration is different between two environments

rknight20772, 2024-09-05:
Hey
Yes, I just followed the instructions on upgrading to the add-on. apps.yaml's contents from the old version was just pasted into the new version. Just checked to make sure id selected everything to replace and I have. I don't have the appdeamon version as I removed it as per the instructions.

Is there any setting to force discharge at the end of the day, rather than throughout the day that Ive not toggled on (or have turned off that I shouldn't)?

Thanks for the quick response
Rich

gcoan, 2024-09-05:
No there isn't a single setting for 'force end of day discharge', predbat should work this out automatically based on your import rates, solar generation and forecast load demand.

You can manually override specific slot behaviour using select.predbat_manual_discharge but that will become tiresome quite quickly.

If you don't have the appdaemon, no issue. Do you know what version of predbat you were running in appdaemon?  It could be that there's a problem with 8.4.4 that didn't appear in whatever you were on before so one option could be to downgrade (use select.predbat_update) to an older version and see if the plan improves.

Can you share your current predbat plan?

rknight20772, 2024-09-05:
I was using 8.4.2 previously. I still have the logs from the previous few days for the previous version if that would help. They are obviously fairly long.

Current plan below.

`<head></head>
Plan starts: 2024-09-05 18:35 last updated: 18:35:00 version: v8.4.4
--
best_soc_min 0.0 best_soc_max 0.0 best_soc_keep 0.8 carbon_metric 0 metric_self_sufficiency 0.0 metric_battery_value_scaling 1.0
Time | Import p | Export p | State |   | Limit % | PV kWh | Load kWh | Car kWh | SOC % | Cost | Total
Thu 18:35 | 22.28 | 15.00 | ↘ |   | 0.03 | 0.34 |   | 45↘ | → | £1.18
Thu 19:00 | 22.28 | 15.00 | ↘ |   | 0.01 | 0.29 |   | 42↘ | → | £1.18
Thu 19:30 | 22.28 | 15.00 | ↘ |   |   | 0.22 |   | 38↘ | → | £1.18
Thu 20:00 | 22.28 | 15.00 | ↘ |   |   | 0.24 |   | 36↘ | → | £1.18
Thu 20:30 | 22.28 | 15.00 | ↘ |   |   | 0.28 |   | 33↘ | → | £1.18
Thu 21:00 | 22.28 | 15.00 | ↘ |   |   | 0.29 |   | 30↘ | → | £1.18
Thu 21:30 | 22.28 | 15.00 | ↘ |   |   | 0.29 |   | 27↘ | → | £1.18
Thu 22:00 | 22.28 | 15.00 | ↘ |   |   | 0.3 |   | 24↘ | → | £1.18
Thu 22:30 | 22.28 | 15.00 | ↘ |   |   | 0.25 |   | 20↘ | → | £1.18
Thu 23:00 | 22.28 | 15.00 | ↘ | Dis↘ | 4 |   | 0.3 |   | 17↘ | -7 p ↘ | £1.18
Thu 23:30 | 7.00 | 15.00 | Chrg↗ | 100 |   | 0.29 |   | 8↗ | +14 p ↗ | £1.11
Fri 00:00 | 7.00 | 15.00 |   | 0.51 |   | 26↗ | +81 p ↗ | £1.25
Fri 00:30 | 7.00 | 15.00 |   | 0.21 |   | 44↗ | +14 p ↗ | £2.07
Fri 01:00 | 7.00 | 15.00 |   | 0.19 |   | 61↗ | +13 p ↗ | £2.21
Fri 01:30 | 7.00 | 15.00 |   | 0.17 |   | 79↗ | +13 p ↗ | £2.35
Fri 02:00 | 7.00 | 15.00 |   | 0.15 |   | 96↗ | +4 p ↗ | £2.48
Fri 02:30 | 7.00 | 15.00 |   | 0.16 |   | 100→ | +1 p ↗ | £2.52
Fri 03:00 | 7.00 | 15.00 |   | 0.2 |   | 100→ | +1 p ↗ | £2.53
Fri 03:30 | 7.00 | 15.00 |   | 0.2 |   | 100→ | +1 p ↗ | £2.55
Fri 04:00 | 7.00 | 15.00 |   | 0.2 |   | 100→ | +1 p ↗ | £2.56
Fri 04:30 | 7.00 | 15.00 |   | 0.15 |   | 100→ | +1 p ↗ | £2.58
Fri 05:00 | 7.00 | 15.00 |   | 0.16 |   | 100→ | +1 p ↗ | £2.59
Fri 05:30 | 22.28 | 15.00 | ↘ |   |   | 0.18 |   | 100↘ | → | £2.60
Fri 06:00 | 22.28 | 15.00 | ↘ |   |   | 0.18 |   | 98↘ | → | £2.60
Fri 06:30 | 22.28 | 15.00 | ↘ |   | 0.02 | 0.18 |   | 96↘ | → | £2.60
Fri 07:00 | 22.28 | 15.00 | ↘ |   | 0.05 | 0.29 |   | 94↘ | → | £2.60
Fri 07:30 | 22.28 | 15.00 | ↘ |   | 0.09 | 0.33 |   | 91↘ | → | £2.60
Fri 08:00 | 22.28 | 15.00 | ↘ |   | 0.12☀ | 0.58 |   | 89↘ | → | £2.60
Fri 08:30 | 22.28 | 15.00 | ↘ |   | 0.13☀ | 0.38 |   | 83↘ | → | £2.60
Fri 09:00 | 22.28 | 15.00 | ↘ |   | 0.17☀ | 0.39 |   | 80↘ | → | £2.60
Fri 09:30 | 22.28 | 15.00 | ↗ |   | 0.33☀ | 0.22 |   | 78↗ | → | £2.60
Fri 10:00 | 22.28 | 15.00 | ↗ |   | 0.51☀ | 0.27 |   | 79↗ | → | £2.60
Fri 10:30 | 22.28 | 15.00 | ↗ |   | 0.69☀ | 0.25 |   | 81↗ | → | £2.60
Fri 11:00 | 22.28 | 15.00 | ↗ |   | 0.88☀ | 0.31 |   | 85↗ | → | £2.60
Fri 11:30 | 22.28 | 15.00 | ↗ |   | 1.04☀ | 0.28 |   | 91↗ | -1 p ↘ | £2.60
Fri 12:00 | 22.28 | 15.00 | ↗ |   | 1.18☀ | 0.39 |   | 98↗ | -7 p ↘ | £2.58
Fri 12:30 | 22.28 | 15.00 | → |   | 1.29☀ | 0.3 |   | 100→ | -14 p ↘ | £2.51
Fri 13:00 | 22.28 | 15.00 | → |   | 1.38☀ | 0.23 |   | 100→ | -16 p ↘ | £2.36
Fri 13:30 | 22.28 | 15.00 | → |   | 1.46☀ | 0.22 |   | 100→ | -17 p ↘ | £2.20
Fri 14:00 | 22.28 | 15.00 | → |   | 1.49☀ | 0.24 |   | 100→ | -17 p ↘ | £2.03
Fri 14:30 | 22.28 | 15.00 | → |   | 1.5☀ | 0.19 |   | 100→ | -18 p ↘ | £1.85
Fri 15:00 | 22.28 | 15.00 | → |   | 1.48☀ | 0.21 |   | 100→ | -17 p ↘ | £1.67
Fri 15:30 | 22.28 | 15.00 | → |   | 1.41☀ | 0.14 |   | 100→ | -18 p ↘ | £1.50
Fri 16:00 | 22.28 | 15.00 | → |   | 1.31☀ | 0.2 |   | 100→ | -15 p ↘ | £1.31
Fri 16:30 | 22.28 | 15.00 | → |   | 1.19☀ | 0.19 |   | 100→ | -14 p ↘ | £1.15
Fri 17:00 | 22.28 | 15.00 | → |   | 1.02☀ | 0.2 |   | 100→ | -11 p ↘ | £1.01
Fri 17:30 | 22.28 | 15.00 | → |   | 0.77☀ | 0.24 |   | 100→ | -7 p ↘ | £0.90
Fri 18:00 | 22.28 | 15.00 | → |   | 0.52☀ | 0.28 |   | 100→ | -3 p ↘ | £0.82
Fri 18:30 | 22.28 | 15.00 | FrzDis→ | 99 | 0.22☀ | 0.19 |   | 100→ | -0 p ↘ | £0.79
Fri 19:00 | 22.28 | 15.00 | ↘ |   | 0.03 | 0.24 |   | 100↘ | → | £0.78
Fri 19:30 | 22.28 | 15.00 | ↘ |   |   | 0.22 |   | 97↘ | → | £0.78
Fri 20:00 | 22.28 | 15.00 | → |   |   | 0.0 |   | 95→ | → | £0.78
Fri 20:30 | 22.28 | 15.00 | → |   |   | 0.0 |   | 95→ | → | £0.78
Fri 21:00 | 22.28 | 15.00 | ↗ | Dis↘ | 4 |   | 0.0 |   | 95↘ | -8 p ↘ | £0.78
Fri 21:30 | 22.28 | 15.00 | Dis↘ | 4 |   | 0.0 |   | 88↘ | -25 p ↘ | £0.69
Fri 22:00 | 22.28 | 15.00 |   | 0.0 |   | 69↘ | -25 p ↘ | £0.44
Fri 22:30 | 22.28 | 15.00 |   | 0.0 |   | 49↘ | -25 p ↘ | £0.18
Fri 23:00 | 22.28 | 15.00 |   | 0.0 |   | 30↘ | -25 p ↘ | -£0.08

`

rknight20772, 2024-09-05:
Im happy with the plan as it is now, but earlier in the day it was planned to discharge throughout the day. It's only due to me setting to 'control charge only' that the earlier discharges were removed. Historically, this hasn't happened, with the only discharges being later on in the day. This is important as the history for the house isn't always reliable. Some days with excess solar and battery, the washing machine and dryer gets used a lot (we've not been able to do that today due to the earlier discharges and lack of solar)
Ideally a setting that was 'leave discharge til after 8pm' would be great (which is how the plan was working out previously)

gcoan, 2024-09-05:
If your household load varies a lot there's a couple of strategies you could adopt:

- increase the number of days of history that predbat looks at, that way it has a better idea of what the load could be
- increase best soc keep, the target soc level for predbat to keep in the battery
- increase load scaling and turn on calculate in-day adjustment - this will ensure predbat takes account of a higher load today and not just base the prediction off historic data

I understand what you mean by a 'keep battery to 8pm' type of switch but different people will have different desires as to when they keep the battery level.  If predbat is working properly it should keep the battery until it's sure that you won't need it.  If 8.4.2 was improved over 8.4.4 then next time it gives a 'bad' plan, do try downgrading and compare the results

gcoan, 2024-09-05:
Just looking at the release history between 8.4.2 and 8.4.4, the changes have been to the new web interface, not the core predbat optimisation engine, so strange that you are getting different results
https://github.com/springfall2008/batpred/releases

rknight20772, 2024-09-05:
Thanks for the update. I'll give some of the suggestions a try.

It may just be a coincidence. I'll probably monitor for a day or so and see what's happening. Maybe today was just a day for a daytime discharge.

closed 2025-01-01 by springfall2008