springfall2008/batpred #1433: Battery has started discharging while charging the car after migrating to using the add on.

WillCalderwood, 2024-09-05:
**Describe the bug**
Predbat is discharging the battery while charging the car.

**Expected behavior**
The battery should not discharge when the car is charging.  This was working correctly, but I then migrated to using the Add On from using AppDeamon and now the battery discharges while charging the car. I thought I copied all the settings across, but maybe I missed something, although I can't see what.

**Predbat version**
8.4.4 with add on

**Environment details**
GivEnergy battery and inverter.

**Screenshots**
`2024-09-05 19:55:00.886285: Octopus Intelligent slot at 09-06 00:30:00-09-06 05:30:00 assumed price 7.0 amount 35.0 kWh location None source smart-charge`

[image]

[image]

**Log file**
Attached.
[predbat-log.txt](https://github.com/user-attachments/files/16899178/predbat-log.txt)

mpartington, 2024-09-06:
Did you copy across the apps.yaml from your AppDaemon instance to the new addon location?  

I'm not 100% sure (this is the case on my system, but I don't use Octopus intelligent Charging), however:

There seems to be a potential issue with your configuration and Predbat doesn't know you have a car, therefore it is not planning the charging slots.

Your log says:
2024-09-05 18:10:00.388279: Configuration: forecast_hours 48 num_cars 0 debug enable is False calculate_plan_every 10

I think Num_cars should be greater than 0 if correctly configured .  I'd go back and carefully check apps.yaml for the car part

mpartington, 2024-09-06:
[image]

WillCalderwood, 2024-09-06:
I understanding was that cars only needed to be set if PredBat was modelling the charging, which it doesn't do with Intelligent. I'll try setting to 1 anyway and see what happens.

mpartington, 2024-09-06:
It will need to know when the car is plugged in, in order to prevent discharge of the battery during the charge slots. Otherwise it will just fill /discharge as much as possible to maximise export

gcoan, 2024-09-06:
> It will need to know when the car is plugged in, in order to prevent discharge of the battery during the charge slots. Otherwise it will just fill /discharge as much as possible during the fixed cheap rate slot to maximise export - which is the behaviour seen on your plan

I'm pretty positive num cars has to be set regardless of the charging approach for the reasons you say

WillCalderwood, 2024-09-06:
> It will need to know when the car is plugged in, in order to prevent discharge of the battery during the charge slots. Otherwise it will just fill /discharge as much as possible during the fixed cheap rate slot to maximise export - which is the behaviour seen on your plan

It does know the car is plugged in and can see the charge slots as it's in the log file.
`2024-09-05 19:55:00.886285: Octopus Intelligent slot at 09-06 00:30:00-09-06 05:30:00 assumed price 7.0 amount 35.0 kWh location None source smart-charge`

mpartington, 2024-09-06:
Possibly, although I see in Trefors video, he has car_charging_planned
status set from his wall box, so I suspect a local 'car connected' may
still be required.

Interesting, as hopefully my charger will be integrated by year end, so
I'll need to do similar.

mpartington, 2024-09-06:
You are right, just checked apps.yaml and it it should know it is connected. You do need to set num_cars to 1 though. That's probably the issue

WillCalderwood, 2024-09-07:
I think this worked last night. I'll test one more time with an emptier battery so there are more charging slots to ensure there's no overlap.

WillCalderwood, 2024-09-12:
Just plugged in again and I can confirm that's resolved the issue,

As I'm sure I won't be the last person to do this, can I suggest a warning in the log file if intelligent is turned on with cars=0? It's probably also worth settings setting cars to 1 if it equals zero and intelligent is turned on.

gcoan, 2024-09-12:
I'll put something in the documentation to make it clearer that num cars must be set. 

Good idea about the warning message

closed 2024-09-21 by gcoan