springfall2008/batpred #1435: Updating Predbat - getting error "HAInterface object has no attribute 'get_slug'"

jameslavery, 2024-09-06:
Having been away from home for some time, returned to find my Raspberry Pi had shut down and not restarted (maybe power cut).

On restarting, saw that Predbat was way behind on versions - V8.0.0(?)

Tried upgrading the Predbat version using the version dropdown (on the expert mode control?), and am getting the above error.

Any solution? Reinstall Predbat from scratch? Or is it a more general HA problem?

gcoan, 2024-09-06:
Predbat was split into multiple files in a version before v8 and there was a specific version you had to migrate to to manage that process.  It may well be that you missed that out.

Easiest way to recover is to just copy ALL of the .py files from predbat on github and restart appdaemon or the predbat addon, whichever you are using

jameslavery, 2024-09-06:
> Predbat was split into multiple files in a version before v8 and there was a specific version you had to migrate to to manage that process. It may well be that you missed that out.
>
> Easiest way to [...]

Thanks - that makes sense, I'll do the manual copy.