springfall2008/batpred #1440: Solis charge times issue

FarmingMartin, 2024-09-08:
**Describe the bug**
Predbat is not setting the charge times correctly I.e setting charge start time to 7.00 and charge end time to 6.30 but writing it at 5.20 causing charging to begin immediately (5.20).
When charge time has ended leaves charge times in place in inverter.
Predbat status shows idle when still charging as predbat charge plan has removed charge windows.

**Expected behavior**
Charge times should be set in above example start 5.30 end 6.30.
When charge time has ended set charge times to start 00.00 end 00.00

[image]
[image]

**Predbat version**
8.4.6

- Inverter and battery setup
- solis RAI-3k-48ES-5G
- Pylontec US5000

springfall2008, 2024-09-08:
I've added a fix to 'main' for testing that will hopefully help with this.

FarmingMartin, 2024-09-08:
That's great when will I be available for general use?

springfall2008, 2024-09-09:
I made a release yesterday, but it seems testing to see if it fixes the issue, can you try?

FarmingMartin, 2024-09-09:
Hi yes mine did update and it seems to have fixed the problem but I will monitor it over the week and update here.

FarmingMartin, 2024-09-20:
Hi it seems after the latest update the charging window is not canceling again. Predbat set a charge between 15.00 and 15.30 which programs my inverter to charge. But when 15.30 came charging is continuing. On closer inspection there is another charge time programed into the inverter settings but not charge scheduled in predbat. This will now allow charging at peak rate on Octopus Agile
Please can you check to see if there is another bug in the latest update effecting my Solis inverter using SolaX modbus.
I have added screenshots/photos fir referance.
[image]
[image]
[image]

FarmingMartin, 2024-09-21:
Further update. I have now noticed that batpred has now assigned a charging window from 22.00 until 7.30 at 0A. This has the effect of now turning off any battery use to power the house resulting in grid use only, although batpred predicts battery charge rate dropping during this time
I have again added some screen shots of my inverter charge menu screen, batpred charging schedule window. Agile electrical pricing and battery charge/discharge graphs.
[image]
[image]
[image]
[image]
Please can you check to see if this is a bug or how batpred is supposed to work.

springfall2008, 2024-09-22:
This seems odd, if the charge window is not active then why is it having an impact, what should be set?

FarmingMartin, 2024-09-22:
I'm not sure which comment you are referring to but with Solis inverters if you set a charge amps of 0 with in a charge time window, during that time the inverter will not charge or discharge the battery.
So from my screen shots you can see the time is 22.30 and the charge time slot is 22.00 to 7.30 at 0 amps, so during that time all household power will come from the grid and the battery will not be charging or discharging to meet house load. At this time, the Agile price is around 19p, so I would expect predbat to be using battery power for house load. If I am miss understanding predbats  usage calculations then please let me know.
I hope this sheds some light on what I am trying to bring to your attention as a possible bug.

springfall2008, 2024-09-22:
This seems wrong, as the plan doesn't show a charge and hence its suppose to by supplying the house load.

Can you capture a logfile for this case?

springfall2008, 2024-09-22:
Think I may have found an issue, can you re-test on 'main' please?

FarmingMartin, 2024-09-22:
Testing now i will report back once I have any info.

springfall2008, 2024-09-23:
Hmm its still not resetting, will look. Do you have logfile?

FarmingMartin, 2024-09-23:
For your information incase you are unaware, with Solis inverters to cancel any charge/discharge times etc we set the start and end times to 00.00.
Hooe this helps

FarmingMartin, 2024-09-25:
I have been monitoring over the last couple of days and have found it looks like charging and hold charge is working as it should. 
The issue seems to be when neither of these are in use and predbat is at idle. It seems to assign a random charge window to the inverter. Last night, it was assigned 22.00 to 23.59 at 40a. At 23.00, which then turned on charging.
If you could get it to assign a start and stop time of 00.00, then this should solve the issue.
Hope this helps.

FarmingMartin, 2024-09-27:
Any update on a possible fix?
Predbat has no charging schedule showing but Solis inverter menu has charge window assigned for 1.30 to 2.30 at 41A.

[image]
[image]