springfall2008/batpred #1448: Freeze charging and freeze discharging confusion

SwiftRR, 2024-09-09:
**Describe the bug**
After being on Predbat for 8 months, I am still confused by Freeze discharging and Freeze charging.
I read the documentation repeatedly, a useful snapshot is in
https://springfall2008.github.io/batpred/what-does-predbat-do/#predbat-status
[image]
[image]

Also: https://springfall2008.github.io/batpred/customisation/#inverter-control-options

Then I look at my plan .........

This is Freeze **discharge** but the plan seems to want to **charge**.
_To me, this is what I would expect for Freeze charging. SOC is increasing using PV_
[image]

And this is Freeze **charge** but the plan seems to want to **discharge**.
_To me, this is what I would expect for Freeze discharging._ SOC is decreasing using grid.
[image]

Aren't these the wrong way round?
Looking back at other forum comments, I am not the only one confused by these terms.

**Expected behavior**
Freeze charging and freeze discharging plan to match documentation.
_It may do so but I just don't understand!_ 

**Predbat version**
v8.4.6

**Environment details**
- GE AC3.0 to two 8.2 kWh batteries
- HAOS 12.3 (Core 2024.6.4) on Raspberry pi 4 (external SSD).
- Tariff: Octopus Agile importing and Octopus 15p fixed export. 

**Screenshots**
See above from Predbat plan.

**Log file**

Help! Is this a bug?

Rob

gcoan, 2024-09-09:
1. The first is I suggest operating per the documentation but the freeze discharge description is however missing the words "to the target (limit) SoC level"
Battery is in discharge mode
Charging **above the target level** is disabled
Excess solar **above the target level** will be exported

What is I think confusing the situation is that the target level is 99% and so the battery is still going to charge to that level before it freeze discharges

2. Freeze charging example seems more muddled as to what predbat is doing.
Grid or solar covers house load. In your example there is no solar so per the description SoC should hold, grid should be taking the load and costs going up.
Costs are going up but also SoC is dropping. The target limit 80% is the same as the initial SoC so that doesn't explain it either.

Personally I agree these extra modes of freeze discharge and freeze charge are confusing, the titles are open to interpretation as to how they work and the description has been iterated several times to try to explain the inner workings of Predbat.  With a blank sheet of paper maybe we could have come up with different titles, but what they are I don't know. Swapping the titles round is going to cause a different set of confusions so I don't advocate that.

At the end of the day though its pence either way so I just leave Predbat to get on with it.

#2 does look like strange behaviour with SoC dropping and costs going up.

gcoan, 2024-09-09:
Its definitely worth also looking at what predbat is actually instructing the battery to do, regardless of what the plan suggests it might do

At the moment my plan is showing freeze discharging for most of the day:
[image]

With a limit of 99% both batteries are well below that limit

But on the power flows I can see that all the solar is being exported there isn't anything going into the batteries:
[image]

The 28W or so "charging" is just a nominal amount because the Battery Management System doesn't properly stop charging ever.  The battery pause mode is set to PauseCharging so I'm happy its operating to specification

SwiftRR, 2024-09-09:
I think this is why I generally ignore the freeze options and just let predbat get on with things.
If I try to understand what the freeze options are actually doing, I think my brain freezes over as well in sympathy.

...... so no bug which is good 😊.

I would probably be happier with no freeze options at all! I can understand the rest.

Rob

SwiftRR, 2024-09-09:
In https://springfall2008.github.io/batpred/what-does-predbat-do/#predbat-status, is the problem the phrase 'the current SoC level' in both freeze charging and freeze discharging definitions.

From your reply, should the definitions be **to the target level%** instead of **the current SOC level**?

Rob

gcoan, 2024-09-09:
> From your reply, should the definitions be to the target level% instead of the current SOC level?

In my own personal example today of a freeze discharge, I'd say no as the target level was 99% and my batteries were well below this and were not (incorrectly) charging.

But your example of freeze charge it did appear that this might be the case

Nilogax, 2024-09-13:
I'm another one who finds these statuses a little tricky to interpret. AIUI Freeze Charging is effectively "Maintain Minimum SOC x%",  so if solar is insufficient to meet house load it'll draw from the battery down to the specified SOC and then draw from the grid. Likewise Freeze Discharge is "Maintain Maximum SOC x%" so if there's excess generation above that SOC it goes to the grid rather than the battery.  Is that about right?

gcoan, 2024-09-13:
> I'm another one who finds these statuses a little tricky to interpret. AIUI Freeze Charging is effectively "Maintain Minimum SOC x%",  
Correct

> so if solar is insufficient to meet house load it'll draw from the battery down to the specified SOC and then draw from the grid. 
Correct.
Also if there is more solar than house load the excess will charge the battery.
"Freeze at minimum soc but allows battery charging"

> Likewise Freeze Discharge is "Maintain Maximum SOC x%" so if there's excess generation above that SOC it goes to the grid rather than the battery.  Is that about right? 
Correct 
"freeze at maximum soc but allows battery discharging"

closed 2025-01-01 by springfall2008