springfall2008/batpred #1449: Prevent unnecessary storage of Predbat attributes

gcoan, 2024-09-09:
**Is your feature request related to a problem? Please describe.**
Currently some of the entity sensors predbat writes have lots of attributes (e.g. the html plan) and these can cause warnings in the HA log from the recorder service.

There is a workaround of extra config in configuration.yaml in the FAQ's to stop these warnings happening, but this requires manual setup by the end-user and has to be extended whenever new attributes exceed the HA limits

**Describe the solution you'd like**
There is a new HA method of stopping entity attributes being recorded, requiring explicit config in code for the attributes that do need to be retained.

See https://github.com/BJReplay/ha-solcast-solar/issues/160#issuecomment-2337823035 

Will result in a cleaner log and better Predbat product as a result

**Describe alternatives you've considered**
Continue with current manual end-user configuration

**Additional context**
Current workaround in the FAQ's that will be able to be removed once this enhancement is added https://springfall2008.github.io/batpred/faq/#predbat-is-causing-warning-messages-about-exceed-maximum-size-in-the-home-assistant-core-log

springfall2008, 2024-09-15:
I commented back on the linked thread, but the solution there is for an Integration, I don't know if Add Ons which use the REST API can access unrecorded_attributes

gcoan, 2024-09-15:
Yes I saw your reply @springfall2008 

Thought it was worth raising the FR to bring to your attention and see if it is possible to use it in an add-on, or if there is an analogous solution for add-on's.

If not, oh well

closed 2026-08-20 by gcoan