springfall2008/batpred #1450: Discharge start hours not being updated

simondrake, 2024-09-09:
**Describe the bug**
Note: Apologies for the brevity (on mobile). 

I'm finding that the discharge start hours isn't being set by predbat. Charging start/end is fine, and discharging end is fine. 

I've double checked the entities are correct in `app.yaml`

**Expected behavior**
Predbat updates the discharge start to the correct values 

**Predbat version**

1.2.0

**Environment details**

- Inverter and battery setup: Solis with four 5 kWh batteries. 
- Standard HAOS installer or Docker: standard 
- Anything else?: everything else seems to be working fine. 

**Log file**

Can't go back far enough on mobile, but the last time it happened the logs said it was changing the discharge period to 00:00 - 00:00 but it stayed at 16:00 - 00:00 which resulted in the batteries discharging for longer than they should have. 

I've included the earliest logs I can access on mobile. 

```
15728	2024-09-09 20:35:05.246782: Best discharge window [ 09-09 22:15:00 - 09-09 22:30:00 @ 15.51p 27%, 09-09 23:00:00 - 09-09 23:30:00 @ 15.51p 21%, 10-09 01:45:00 - 10-09 02:00:00 @ 15.51p 21%, 10-09 06:00:00 - 10-09 10:00:00 @ 15.51p 24%, 10-09 10:05:00 - 10-09 10:30:00 @ 15.51p 21%, 10-09 12:00:00 - 10-09 12:30:00 @ 15.51p 43%, 10-09 13:15:00 - 10-09 13:30:00 @ 15.51p 53%, 10-09 16:00:00 - 10-09 19:00:00 @ 23.63p 48%, 10-09 19:35:00 - 10-09 20:30:00 @ 15.51p 29%, 10-09 20:40:00 - 10-09 21:00:00 @ 15.51p 23% ]
15727	2024-09-09 20:35:05.246672: Best charge window [ 09-09 21:30:00 - 09-09 22:00:00 @ 14.33p 35%, 10-09 01:00:00 - 10-09 01:30:00 @ 5.51p 23%, 10-09 01:30:00 - 10-09 01:45:00 @ 3.53p 29%, 10-09 02:00:00 - 10-09 06:00:00 @ 2.4p 92%, 10-09 10:00:00 - 10-09 10:05:00 @ 2.2p 31%, 10-09 10:30:00 - 10-09 12:00:00 @ 1.43p 53%, 10-09 12:30:00 - 10-09 13:15:00 @ 0.7p 62%, 10-09 13:30:00 - 10-09 16:00:00 @ 0.42p 100%, 10-09 22:30:00 - 10-09 23:00:00 @ 9.04p 23%, 11-09 01:30:00 - 11-09 18:30:00 @ 3.53p 100%, 11-09 19:00:00 - 11-09 20:30:00 @ 20.73p 100% ]
15726	2024-09-09 20:35:05.246533: Plan was last updated on 2024-09-09 20:30:00+01:00 and is now 5.0 minutes old
15725	2024-09-09 20:35:05.246497: Base discharge window [ 09-09 00:00:00 - 09-09 00:00:00 @ 0p 100.0%, 10-09 00:00:00 - 10-09 00:00:00 @ 0p 100.0% ]
15724	2024-09-09 20:35:05.246422: Base charge window [ 09-09 22:30:00 - 09-09 23:00:00 @ 0p 38%, 10-09 22:30:00 - 10-09 23:00:00 @ 0p 38% ]
15723	2024-09-09 20:35:05.206640: Found 1 inverters totals: min reserve 4.2 current reserve 4.2 soc_max 20.0 soc 8.0 charge rate 3.5999999999999996 kW discharge rate 3.5999999999999996 kW battery_rate_min 100.0 w ac limit 3.6 export limit 3.6 kW loss charge 3 % loss discharge 3 % inverter loss 4 %
15722	2024-09-09 20:35:05.206605: Note: Inverter does not support reserve - disabling reserve functions
15721	2024-09-09 20:35:05.206599: Note: Inverter does not support charge freeze - disabled
15720	2024-09-09 20:35:05.206593: Note: Inverter does not support discharge freeze - disabled
```

matttheotter, 2024-09-10:
#1440 related?

springfall2008, 2024-09-14:
This is kind of as expected, if the discharge start is in the past then it won't change it if the desired behaviour is present. Do you see a functional issue?

simondrake, 2024-09-14:
I've noticed some odd behaviour with how predbat used 00:00 as the time. For example, a couple of days ago, it was supposed to charge between 04:00 - 06:00 (free electricity) but it set the charge window to 04:00 - 00:00 resulting in it not charging at all. 

Thanks for replying.

springfall2008, 2024-09-15:
What inverter, this should work for GE but maybe not for another brand?

gcoan, 2024-09-15:
> What inverter, this should work for GE but maybe not for another brand?

in the description it says its a Solis inverter

simondrake, 2024-09-15:
Yeah - Solis. I'm basically using the template from this repo, I can post the whole config tomorrow if it helps. 

It's weird because other times it sets it and works fine, just sometimes it tries to use 00:00 in the start or end and it messes it up.

closed 2025-01-01 by springfall2008