springfall2008/batpred #1451: Web interface not working

tractor137, 2024-09-09:
**Describe the bug**
The web app works from within the predbat add-on but not on via homeassistant.local:5052

**Expected behavior**
A clear and concise description of what you expected to happen.

**Predbat version**
8.4.6

**Environment details**
5kw inv 9.5kwh battery
predbat add-on

**Screenshots**
If applicable, add screenshots to help explain your problem. The most useful ones can be your battery chart, the Predbat HTML plan and your current settings in HA.

[image]

[image]

[image]

**Log file**
[log.txt](https://github.com/user-attachments/files/16936618/log.txt)

matttheotter, 2024-09-10:
@tractor137 I assumed it was moved to ingress so is only visible from within HASS now, @springfall2008 is that the case?

tractor137, 2024-09-10:
I was just going off the guidance in the documentation regarding output data. Interestingly it worked at v8.4.0 (which was the first version I had used after installing the predbat addon) but then as soon as it updated (I think to 8.4.2) it didn’t work and hasn’t since

matttheotter, 2024-09-10:
The only thing I remember is a note in [v8.4.4](https://github.com/springfall2008/batpred/releases/tag/v8.4.4):

> Note you may have to uninstall and install again the add on if the web interface doesn't work, you don't need to edit your apps.yaml or move any files around they will be safe.

tractor137, 2024-09-10:
Yes I saw that too and this was what I did just before posting here

springfall2008, 2024-09-14:
This is as expected, I'm updating the documentation. The direct access to 5052 is a security risk as there is no login.

closed 2025-01-01 by springfall2008