springfall2008/batpred #1452: Warn: Inverter 0: inverter definition is not a dictionary

Dave-Parry, 2024-09-10:
I'm using predbat standalone v8.4.6
in the web gui and the predbat logs, I'm getting this error message twice every 5 minutes.
12014 | 2024-09-10 15:15:03.656443: Warn: Inverter 0: inverter definition is not a dictionary
I'm not sure when it started, but logs show it was doing it all yesterday (09-09-2024)
Everything seems to be working but there's obviously something not right.
Not sure what it means or how to cure it.

matttheotter, 2024-09-10:
Hi Dave,

Grab the latest apps.yaml, a fix was pushed to it a week ago.

Or you can amend the line:[image]

Matt

Dave-Parry, 2024-09-10:
Thanks Matt, that seems to have cured it.

closed 2024-09-10 by Dave-Parry