springfall2008/batpred #1453: Charging from grid when plan suggests should be discharging

raldred, 2024-09-11:
**Describe the bug**
I am trying to get predbat working with my LuxPower inverter, I realise it's not supported, maybe, this is why.
However...

The calculated plan suggests that, currently the battery should be charging, it's set the limit to 58%, and enabled charge, howver, the battery's current SOC is 64%.

Having AC charging enabled on my inverter, even if the battery is above the target SOC of 58%, it will continue to fulfil load from the grid.

I'd like to understand why the plan is enabling charge when if you look at the plan, it's expecting battery SOC to fall during the "Charge" period, which would suggest it's expecting the battery SOC to be dropping because it's servicing the load?

So why is it still trying to charge it when it's well above target SOC?

**Expected behavior**
I would expect the plan to only go into charge status if the SOC is below target SOC?

**Predbat version**

v8.4.6

**Environment details**

- Lux Power Hybrid 7kW
- Hanchu 9.4 kWh battery
- HA 2024.9.0
- HAOS 12.4

<details>
<summary>apps.yaml</summary>

```yaml
# ------------------------------------------------------------------
# This is an example configuration, please modify it
# ------------------------------------------------------------------
---
pred_bat:
  module: predbat
  class: PredBat

  # Sets the prefix for all created entities in HA - only change if you want to run more than once instance
  prefix: predbat

  # Timezone to work in
  timezone: Europe/London

  # Currency, symbol for main currency second symbol for 1/100s e.g. $ c or £ p or e c
  currency_symbols:
   - '£'
   - 'p'

  # Number of threads to use in plan calculation
  # Can be auto for automatic, 0 for off or values 1-N for a fixed number
  threads: auto

  # If you are using Predbat outside of HA then set the HA URL and Key (long lived access token here)
  #ha_url: 'http://homeassistant.local:8123'
  #ha_key: 'xxx'

  # Set to auto-match with a GivEnergy serial number, but you can override the serial or the sensor names
  # if it doesn't work or if you have more than one inverter you will need to list both
  # geserial: 're:sensor.givtcp_(.+)_soc_kwh'
  # geserial2: 're:sensor.givtcp2_(.+)_soc_kwh'

  # Sets the maximum period of zero load before the gap is filled, default 30 minutes
  # To disable set it to 1440
  load_filter_threshold: 30

  #
  # Sensors, more than one can be specified and they will be summed up automatically
  #
  # For two inverters the load today would normally be the master load sensor only (to cover the entire house)
[... 522 lines cut ...]
  "manual_charge":"off",
  "manual_discharge":"off",
  "manual_idle":"off",
  "manual_freeze_charge":"off",
  "manual_freeze_discharge":"off",
  "auto_update":false,
  "load_filter_modal":true,
  "iboost_enable":false,
  "carbon_enable":false,
  "holiday_days_left":0.0,
  "forecast_plan_hours":24.0,
  "plan_debug":false
}
```

</details>

**Screenshots**
If applicable, add screenshots to help explain your problem. The most useful ones can be your battery chart, the Predbat HTML plan and your current settings in HA.

[image]

**Log file**

[predbat.log](https://github.com/user-attachments/files/16970954/predbat.log)

dbw1966, 2024-09-15:
Somewhat counter intuitively, charge 58% when the soc is higher means allow to discharge to that level and then float at that point and take any futher energy needs from the grid.  The only time you will see discharge is when the battery is force discharging, that is, its told to out more than the house load and feed in to the grid.

couzensmark, 2024-09-23:
Hello Raldred,

Could you share your config please as I am attempting to use predbat with Luxpower inverter, however I have so far been unable to get the config to work at all.

Thanks

Mark

raldred, 2024-09-23:
@dbw1966 Yes, but when it's doing this, batpred turns on the charge service.
When charge is enabled, my inverter will hold charge at whatever the current SOC if the target SOC is below the current SOC, rather than allowing it to discharge to the target level.

closed 2025-01-01 by springfall2008