springfall2008/batpred #1454: Discharge not being set on Sofar

dbw1966, 2024-09-14:
**Describe the Bug**

During discharge slots Predbat is publishing Set/Auto on the Mqtt. 

This means discharge does not get set.

I have tried an automation, but of course its gets overwritten.

**Expected behavior**
That the inverter goes to discharge mode when discharge is planned.

**Predbat version**
8.4.6

**Environment details**

- Sofar HYD-3600-ES
- Standard HAOS installer

**Screenshots**
If applicable, add screenshots to help explain your problem. The most useful ones can be your battery chart, the Predbat HTML plan and your current settings in HA.

[image]
[image]
[image]

**Log file**
2024-09-14 07:35:35.997407: Best charging limit socs [3.09, 2.91, 10.24, 10.24, 6.24, 8.02, 3.24, 7.64, 8.24, 10.24, 10.24, 10.24, 10.24] export [49, 23, 20, 20, 20, 20, 83, 40, 30, 20, 20, 82] gives import battery 53.59 house 2.7 export 18.72 metric 327.32 metric10 481.61
2024-09-14 07:35:36.167525: Inverter 0: Type SF Sofar HYD
2024-09-14 07:35:36.167596: Warn: Inverter 0: Using inverter type SF - not all features are available
2024-09-14 07:35:36.167757: Reserve min: 20.0% Battery_min:4.0%
2024-09-14 07:35:36.167813: Inverter 0 with soc_max 10.24 kWh nominal_capacity 10.24 kWh battery rate raw 2800.0 w charge rate 2.8 kW discharge rate 2.8 kW battery_rate_min 0.0 w ac limit 3.68 kW export limit 3.68 kW reserve 4.0 % current_reserve 4.0 %
2024-09-14 07:35:36.167929: Inverter 0 scheduled_charge_enable off scheduled_discharge_enable on
2024-09-14 07:35:36.168035: Inverter 0 SOC: 6.76kW 66% Current charge rate 2800W Current discharge rate 2800W Current power -70.0W Current voltage 52.0V
2024-09-14 07:35:36.168219: Inverter 0 Wrote scheduled_charge_enable to False successfully and got off
2024-09-14 07:35:36.168236: Inverter 0 scheduled_charge_enable set to False
2024-09-14 07:35:36.168286: Inverter 0 scheduled charge enable is False
2024-09-14 07:35:36.168297: Inverter 0 charge windows currently []
2024-09-14 07:35:36.168310: Inverter 0 Charge settings: timed charged is disabled, power 2.8 kW
2024-09-14 07:35:36.168372: Inverter 0 Wrote scheduled_discharge_enable to True successfully and got on
2024-09-14 07:35:36.168383: Inverter 0 sensor.predbat_SF_0_scheduled_discharge_enable set to True
2024-09-14 07:35:36.168407: Inverter 0 scheduled discharge enable is True
2024-09-14 07:35:36.168433: Inverter 0 discharge windows currently [{'start': 420, 'end': 541, 'average': 0}, {'start': 1860, 'end': 1981, 'average': 0}]
2024-09-14 07:35:36.168449: Note: Inverter does not support discharge freeze - disabled
2024-09-14 07:35:36.168459: Note: Inverter does not support charge freeze - disabled
2024-09-14 07:35:36.168468: Note: Inverter does not support reserve - disabling reserve functions
2024-09-14 07:35:36.168514: Found 1 inverters totals: min reserve 0.41 current reserve 0.41 soc_max 10.24 soc 6.76 charge rate 2.8000000000000003 kW discharge rate 2.8000000000000003 kW battery_rate_min 0.0 w ac limit 3.68 export limit 3.68 kW loss charge 6 % loss discharge 6 % inverter loss 4 %
2024-09-14 07:35:36.227846: Base charge    window [  ]
2024-09-14 07:35:36.228023: Base discharge window [ 14-09 07:00:00 - 14-09 09:01:00 @ 0p 0.0%, 15-09 07:00:00 - 15-09 09:01:00 @ 0p 0.0% ]
2024-09-14 07:35:36.228090: Charge window will be: 2024-09-14 10:30:00+01:00 - 2024-09-14 10:40:00+01:00 - current soc 66 target 30
2024-09-14 07:35:36.228121: Disabled charge window while waiting for schedule (now 09-14 07:35:00 target set_window_minutes 30 charge start time 09-14 10:30:00)
2024-09-14 07:35:36.228170: Adjust idle time, charge 00:00:00-00:00:00 discharge 07:00:00-09:01:00
2024-09-14 07:35:36.228302: Adjust idle time computed idle is 09:01:00-23:59:00
2024-09-14 07:35:36.228345: Include original discharge start 09-14 07:00:00 with our start which is 09-14 07:00:00 (charge start 09-14 10:30:00 end 09-14 10:40:00)
2024-09-14 07:35:36.228373: Next discharge window will be: 2024-09-14 07:00:00+01:00 - 2024-09-14 08:01:00+01:00 at reserve 49
2024-09-14 07:35:36.228394: Discharging now - current SOC 6.76 and target 5.02
2024-09-14 07:35:36.228464: Inverter 0 Adjust force discharge to True, change times from 07:00:00 - 09:01:00 to 07:00:00 - 08:01:00
2024-09-14 07:35:36.228477: Adjust idle time, charge 00:00:00-00:00:00 discharge 07:00:00-08:01:00
2024-09-14 07:35:36.228556: Adjust idle time computed idle is 08:01:00-23:59:00
2024-09-14 07:35:36.228569: Inverter 0 Set new end time to 08:01:00 was 09:01:00
2024-09-14 07:35:40.256920: Inverter 0 Wrote 08:01:00 to discharge_end_time, successfully now 08:01:00
2024-09-14 07:35:40.257006: Sleeping (workaround) as start/end of discharge window was just adjusted
2024-09-14 07:36:10.619235: Inverter 0 Current charge limit is 20% and new target is 49%
2024-09-14 07:36:14.649489: Inverter 0 Wrote 49 to charge_limit, successfully now 49
2024-09-14 07:36:14.955034: iBoost model today updated to 0.0
2024-09-14 07:36:15.008640: Car charging Manual SOC current is 10.0 next is 10.0
2024-09-14 07:36:15.008718: Completed run status Discharging
2024-09-14 07:36:15.025773: Info: record_status Discharging target 66%-49%
2024-09-14 07:36:15.042552: Saved current settings to /addon_configs/6adb4f0d_predbat/predbat_config.json

springfall2008, 2024-09-15:
So I think discharge was never implemented for MQTT, I'm trying to add it

dbw1966, 2024-09-15:
Thanks trefor, I'll test.

dbw1966, 2024-09-15:
Do you need me to review - not sure I'm qualified.

springfall2008, 2024-09-15:
It's on main now if you want to test?

springfall2008, 2024-09-15:
> Do you need me to review - not sure I'm qualified.

No that's okay testing is fine

dbw1966, 2024-09-27:
Trefor, this fix, which we tested, no longer seems to work.

chalfontchubby, 2026-08-20:
Closing - the fix in #1458 was reported as not working, but this hasn't been re-reported in a long while. Please reopen if it's still a problem.

closed 2026-08-20 by chalfontchubby