springfall2008/batpred #1455: During iboost, the Full parameter is not being set.

dbw1966, 2024-09-14:
**Describe the bug**
During iboost, the full parameter is not being set.

**Expected behavior**
That when iboost is set, both iboost becomes true and the Full attribute becomes true.

**Predbat version**
8.4.6

**Environment details**

- Sofar HYD-3600-ES with GTX-5000 batteries and an Eddi
- Standard HAOS installer

**Screenshots**
[image]
[image]
[image]

**Log file**
I'll capture a log and add next time

dbw1966, 2024-09-14:
Recent Info
Plan
[image]

Iboost active before plan kicks in 
[image]

iboost active after plan kicks in - has switched "solar" not "full"
[image]

Log on 11.00 update
2024-09-14 11:00:00.240013: --------------- PredBat - update at 2024-09-14 11:00:00+01:00 with clock skew 0 minutes, minutes now 660
2024-09-14 11:00:00.240560: Saved current settings to /addon_configs/6adb4f0d_predbat/predbat_config.json
2024-09-14 11:00:00.256013: Using cached GITHub data for https://api.github.com/repos/springfall2008/batpred/releases age 85.0 minutes
2024-09-14 11:00:00.256102: Predbat /config/predbat.py version v8.4.6 currently running, latest version is v8.4.6 latest beta v8.4.6
2024-09-14 11:00:00.256130: Autoupdate: Currently up to date
2024-09-14 11:00:00.288520: Configuration: forecast_hours 48 num_cars 1 debug enable is False calculate_plan_every 10
2024-09-14 11:00:00.288864: Cars 1 charging from battery False planned [True], charging_now [False] smart [True], max_price [33.0], plan_time ['17:00:00'], battery size [9.361], limit [9.361], rate [3.47], exclusive [False]
2024-09-14 11:00:00.288914: Predbat mode is set to Control charge & discharge
2024-09-14 11:00:00.432456: iBoost energy today from sensor reads 0.0 kWh
2024-09-14 11:00:01.803721: Found 11520 load_today datapoints going back 7 days
2024-09-14 11:00:03.238749: Current data so far today: load 12.16 kWh import 12.31 kWh export 0.69 kWh pv 2.75 kWh
2024-09-14 11:00:03.238855: Fetching futurerate data from None
2024-09-14 11:00:03.258531: Gas rates min 5.58 max 5.58 average 5.58
2024-09-14 11:00:03.258624: Cars: SOC kWh: [10.0] Charge limit [9.361] plan time ['17:00:00'] battery size [9.361]
2024-09-14 11:00:03.265953: Standing charge is set to 47.8485 p
2024-09-14 11:00:03.275787: Adding rate import: {'date': '2024-09-10', 'start': '13:00:00', 'end': '14:00:00', 'rate': 0, 'load_scaling': 1.0} => 09-14 13:00:00 to 09-14 14:00:00 @ 0.0 date 2024-09-10 00:00:00 increment False
2024-09-14 11:00:03.606691: Rate min forward looking: now -0.68 at end of forecast -0.68
2024-09-14 11:00:03.606929: Import rates min -0.68 max 35.7 average 14.72
2024-09-14 11:00:03.614354: Export rates min 15.0 max 15.0 average 15.0
2024-09-14 11:00:03.614703: Rate thresholds (for charge/discharge) are import 35.2p (0.0) export 15.0p (0.0)
2024-09-14 11:00:03.617739: High export rate found rates in range 15.0 to 15.0
2024-09-14 11:00:03.620512: Low Import rate found rates in range -0.68 to 32.91
2024-09-14 11:00:03.620650: Car 0 plan charging from 10.0 to 9.361 with slots [{'start': 660, 'end': 690, 'average': 2.2}, {'start': 690, 'end': 720, 'average': 0.88}, {'start': 720, 'end': 750, 'average': 0.22}, {'start': 750, 'end': 780, 'average': 0.0}, {'start': 780, 'end': 810, 'average': 0.22}, {'start': 810, 'end': 840, 'average': -0.68}, {'start': 840, 'end': 870, 'average': -0.44}, {'start': 870, 'end': 900, 'average': 0.23}, {'start': 900, 'end': 930, 'average': 1.36}, {'start': 930, 'end': 960, 'average': 2.2}, {'start': 960, 'end': 990, 'average': 22.43}, {'start': 990, 'end': 1020, 'average': 25.12}, {'start': 1020, 'end': 1050, 'average': 28.42}, {'start': 1050, 'end': 1080, 'average': 32.91}, {'start': 1140, 'end': 1170, 'average': 20.42}, {'start': 1170, 'end': 1200, 'average': 20.19}, {'start': 1200, 'end': 1230, 'average': 20.22}, {'start': 1230, 'end': 1260, 'average': 19.27}, {'start': 1260, 'end': 1290, 'average': 19.4}, {'start': 1290, 'end': 1320, 'average': 16.4}, {'start': 1320, 'end': 1350, 'average': 18.34}, {'start': 1350, 'end': 1380, 'average': 15.88}, {'start': 1380, 'end': 1410, 'average': 15.88}, {'start': 1410, 'end': 1440, 'average': 15.88}, {'start': 1440, 'end': 1470, 'average': 19.05}, {'start': 1470, 'end': 1500, 'average': 18.91}, {'start': 1500, 'end': 1530, 'average': 17.66}, {'start': 1530, 'end': 1560, 'average': 17.98}, {'start': 1560, 'end': 1590, 'average': 16.86}, {'start': 1590, 'end': 1620, 'average': 14.95}, {'start': 1620, 'end': 1650, 'average': 12.35}, {'start': 1650, 'end': 1680, 'average': 11.2}, {'start': 1680, 'end': 1710, 'average': 12.13}, {'start': 1710, 'end': 1740, 'average': 11.25}, {'start': 1740, 'end': 1770, 'average': 11.02}, {'start': 1770, 'end': 1800, 'average': 12.99}, {'start': 1800, 'end': 1830, 'average': 17.18}, {'start': 1830, 'end': 1860, 'average': 19.3}, {'start': 1860, 'end': 1890, 'average': 18.44}, {'start': 1890, 'end': 1920, 'average': 20.73}, {'start': 1920, 'end': 1950, 'average': 17.6}, {'start': 1950, 'end': 1980, 'average': 18.5}, {'start': 1980, 'end': 2010, 'average': 17.17}, {'start': 2010, 'end': 2040, 'average': 13.98}, {'start': 2040, 'end': 2070, 'average': 11.02}, {'start': 2070, 'end': 2100, 'average': 7.94}, {'start': 2100, 'end': 2130, 'average': 2.2}, {'start': 2130, 'end': 2160, 'average': 0.88}, {'start': 2160, 'end': 2190, 'average': 0.22}, {'start': 2190, 'end': 2220, 'average': 0.0}, {'start': 2220, 'end': 2250, 'average': 0.22}, {'start': 2250, 'end': 2280, 'average': -0.68}, {'start': 2280, 'end': 2310, 'average': -0.44}, {'start': 2310, 'end': 2340, 'average': 0.23}, {'start': 2340, 'end': 2370, 'average': 1.36}, {'start': 2370, 'end': 2400, 'average': 2.2}, {'start': 2400, 'end': 2430, 'average': 22.43}, {'start': 2430, 'end': 2460, 'average': 25.12}, {'start': 2460, 'end': 2490, 'average': 28.42}, {'start': 2490, 'end': 2520, 'average': 32.91}, {'start': 2580, 'end': 2610, 'average': 20.42}, {'start': 2610, 'end': 2640, 'average': 20.19}, {'start': 2640, 'end': 2670, 'av [... cut ...]
2024-09-14 11:00:05.590137: Inverter 0 Current Target SOC is 48%, already at target
2024-09-14 11:00:05.629040: Will not recompute the plan, it is 5.0 minutes old and max age is 10 minutes
2024-09-14 11:00:05.629123: iBoost model today updated to 0.0
2024-09-14 11:00:05.684130: Car charging Manual SOC current is 10.0 next is 10.0
2024-09-14 11:00:05.684212: Completed run status Discharging
2024-09-14 11:00:05.699837: Info: record_status Discharging target 72%-48%
2024-09-14 11:00:05.715490: Saved current settings to /addon_configs/6adb4f0d_predbat/predbat_config.json

springfall2008, 2024-09-15:
You have iboost solar enabled, and its iboosting on the solar during that force discharge slot hence 'full' is not set?

dbw1966, 2024-09-15:
I would expect iboost to still set the full flag rather than the solar flag.   As we are saying "go over and above just scraping off surplus generation"  and dump as much power as you can into the hot water.    This is what the plan is indicating, full power dump, not scrape off the top.  If you will not be changing the code let me know and I'll unset the flag.

springfall2008, 2024-09-22:
Hmm actually you are right it says 3.7 kWh there, let me look again

dbw1966, 2024-09-24:
Thankyou

springfall2008, 2024-09-29:
Pushed a fix to main, effectively iboost solar was active but the extra energy now included the export which was a mistake

dbw1966, 2024-09-30:
Thanks trefor, can you describe the behavior now
Iboost active, solar true, full false
Iboost active, solar false, full true
Iboost active, solar true, full true
Iboost active, solar false, full false

closed 2026-08-28 by chalfontchubby