springfall2008/batpred #1457: Provide (API) mechanism to externally inform Predbat of 'special rate' pricing events

gcoan, 2024-09-14:
**Is your feature request related to a problem? Please describe.**
Predbat has the ability to do price over-rides by entering the date, start time, end time and load multiplier in apps.yaml.

With Octopus Power up events and Free electricity events, these happen often and I want a simple way to be able to inform Predbat of these without having to manually edit apps.yaml each time.

**Describe alternatives you've considered**
I have a solution for this that works perfectly with predbat running in appdaemon, but I've just discovered that it doesn't (and probably can never) work with the predbat add-on.

My (appdaemon) solution is as follows:
- set of helper entities for date, start and end time
- a shell script that appends the correctly formatted date, start and end time to apps.yaml
- a shell_command in configuration.yaml that invokes the above shell script with the helper entity states being passed as parameters

This works perfectly with predbat running in appdaemon as it can write to my /config/appdaemon/apps/batpred/apps/apps.yaml file

However when I modify the script to write to /addon_configs/6adb4f0d_predbat/apps.yaml it fails as the shell command runs in the homeassistant container context and it doesn't have access to /addon_configs folder tree.

I can't identify any way round this as homeassistant can't directly access the add-on file system structure.  Hence this FR

**Describe the solution you'd like**
A mechanism from within Predbat that I can invoke to instruct it to take in an import price over-ride - would need to be able to pass the start date, time, end time and load multiplier.  A service API if possible would be great, or some sort of configuration in apps.yaml to point to the external over-ride data.

e.g. in apps.yaml, something like this:
user_rates_import_override:
    -  date: input_datetime.octopus_powerup_date
       start: input_datetime.octopus_powerup_start_time
       end: input_datetime.octopus_powerup_end_time
       rate: 0
       load_scaling: input_number.octopus_powerup_load_scaling

Can define these helper entities in HA, configure apps.yaml to point to them, and then have a 'trigger' toggle in predbat that reads from these helper entities and writes the override to apps.yaml.

The reason I suggest writing the override to apps.yaml and not just using these helpers directly is we are on occasion getting multiple events, e.g. this weekend there was a powerup event and a free electricity event on Saturday and then another powerup event on Sunday.  The ability to queue and instruct predbat of these one by one rather than having to wait for one to finish before the next can be instructed would be useful 
 PI A clear and concise description of what you want to happen.

dbw1966, 2024-09-15:
Powerups are already taken care of in Predbat with negative prices in the plan.  I believe the "free for everything over your normal usage" periods are not flagged in the octopus api, so can you comment where you are getting them from?

gcoan, 2024-09-15:
> Powerups are already taken care of in Predbat with negative prices in the plan.  I believe the "free for everything over your normal usage" periods are not flagged in the octopus api, so can you comment where you are getting them from?

No, the problem is that powerups are NOT automatically taken care of in Predbat, I think you might be thinking of Agile plunge pricing when the published agile rate goes negative.

Powerups are where the electricity is zero price for a set period no matter how much you use and so is slightly different from the "free for everything over your normal usage".

Neither of these price rate overrides are made available to Predbat via the Octopus API's or BottlecapDave's Octopus Integration.  The only way to get predbat to plan for these events are to put a manual price rate override in apps.yaml so I'm seeking a mechanism to do this more easily than having to manually edit the file each time.

I'm getting the date and times of these activities via https://github.com/8none1/octopus_powerups which gets them into HA. With the appdaemon add-on I could then write these directly to apps.yaml, but for the predbat addon it runs in its own protected space and HA doesn't have access to write to the predbat addon apps.yaml

springfall2008, 2024-09-15:
What about having a power up selector like the force charge one that sets the zero rate?

gcoan, 2024-09-15:
Yes that could work as well and would be reasonably easy to "drive" from a triggering automation to repeatedly select, pause, select, etc the time periods.

Only potential limitation is the time window length on those drop-down selectors, I think 16, 18 hours?  Would need to be a bit longer, maybe not the whole weekend (for a Friday notification) but would want to have the ability to set 24-30 hours ahead

mpartington, 2024-09-16:
Sounds a good idea to add an override for zero rates.  I can also see a benefit for setting more in advance, my only concern would be the amount of scrolling (on a phone screen).for say 30 hrs (that's 60.slots to pick through). I think the current list of times is long enough to navigate.  However power ups and free energy is likely to be less frequent, so maybe there is a compromise/different solution for that

I tend to also additionally set the other override options (for other reasons) using the select:select service from external automations..Not sure if duplicate times e.g. two 01:00:00 in a list would break this

springfall2008, 2024-09-16:
The reason the list is shorter is not only the size of the list but also to avoid having to include the day in the list, it just auto-clears when the time wraps. I could however accept a day when its used as a service it won't matter its not on the list in this case.

mpartington, 2024-09-16:
I generally use the select.select service to ensure something predictable happens everyday. For example, set my afternoon cheap slots to always charge - in case I put.a random high load on in the period. Adding a day requirement I guess would make this more complicated to automate

gcoan, 2024-09-16:
If it's possible for there to be a shorter dropdown list for use via the UI and the ability to optionally pass date (or day) in when select is called via an API call (for longer term setup's) then this seems the best of both worlds

springfall2008, 2024-10-06:
Okay I have an idea, let me give it a try...

springfall2008, 2024-10-06:
I'm testing out an idea, not yet documented as I don't know if I want to keep it as-is yet. Please can you try an update to main.

There is a new selector called 'select.predbat_manual_api' which works a bit like the manual override ones but this can have new values added using the select API in Home Assistant. The only function the selector itself serves is to store override commands, you can clear from the selector but you have to set them using a service call.

The idea is that it overrides settings in apps.yaml, well only ones that have been enabled for that purpose.

Below is an example of setting a rate override, you can clear all overrides by calling 'off' or this specific one only by calling the same thing again but in square brackets [] 

For the rates you can use 'rates_export_override' or 'rates_import_override' with all the same options as apps.yaml but in a URL type format

```
rates_export_override?start=17:00:00&end=19:00:00&rate=0], +rates_export_override start=17:00:00&end=19:00:00&rate=0
```
If you override a single value item with something like:

```
inverter_limit=4000
```

Only the settings below plus rates_import_override and rates_export_override are currently enabled but its possible to add other things.

If the setting is indexed e.g. per inverter then you have to add 'index' if it applies to just one e.g.

```
inverter_limit?value=4000&index=0
```

```py
# These settings can be overridden by the API
CONFIG_API_OVERRIDE = {
    "inverter_limit" : True,
    "export_limit" : True,
    "inverter_battery_rate_min" : True,
    "inverter_reserve_max" : True,
    "battery_rate_max" : True,
    "car_charging_soc" : True,
    "car_charging_limit": True,
    "car_charging_battery_size": True,
    "days_previous": True,
    "days_previous_weight": True,
    "battery_scaling": True,
    "forecast_hours": True,
    "import_export_scaling": True
}
```

[image]

gcoan, 2024-10-06:
Mm, very interesting solution, far more extensible and general purpose than I had considered.

I like it.  I'll give it a go

gcoan, 2024-10-06:
I've installed the main version and can see the new select control entity.

trying it for the first time with a select value of 

select.predbat_manual_api = rates_import_override?date=07/10/24&start=17:00:00&end=19:00:00&rate=0

Not intentional but I mis-formatted the date, Predbat trapped this and logged it in the logfile and set predbat.status:

  Warn: Bad date 07/10/24 provided in energy rates

Able to clear the bad manual API control from the dropdown OK

Tried a second time with

rates_import_override?date=2024-10-07&start=17:00:00&end=19:00:00&rate=0

This was accepted and I could see in the logfile the existing rate overrindes in apps.yaml and the manual one I input:

```
2024-10-06 21:08:43.299847 INFO pred_bat: Adding rate rates_import_override: {'date': '2024-09-26', 'start': '09:00:00', 'end': '11:00:00', 'rate': 0, 'load_scaling': 1.0} => 10-06 09:00:00 to 10-06 11:00:00 @ 0.0 date 2024-09-26 00:00:00 increment False
2024-10-06 21:08:43.301874 INFO pred_bat: Adding rate rates_import_override: {'date': '2024-10-01', 'start': '04:00:00', 'end': '07:00:00', 'rate': 0, 'load_scaling': 1.0} => 10-06 04:00:00 to 10-06 07:00:00 @ 0.0 date 2024-10-01 00:00:00 increment False
2024-10-06 21:08:43.303812 INFO pred_bat: Adding rate rates_import_override: {'date': '2024-10-03', 'start': '11:00:00', 'end': '14:00:00', 'rate': 0, 'load_scaling': 1.0} => 10-06 11:00:00 to 10-06 14:00:00 @ 0.0 date 2024-10-03 00:00:00 increment False
2024-10-06 21:08:43.305990 INFO pred_bat: Adding rate rates_import_override: {'index': None, 'date': '2024-10-07', 'start': '17:00:00', 'end': '19:00:00', 'rate': '0'} => 10-06 17:00:00 to 10-06 19:00:00 @ 0.0 date 2024-10-07 00:00:00 increment False
```

(index being different for the manual one)

And it worked !

[image]

r/e the list of controls you can override in apps.yaml:

inverter_limit and export_limit I would think unlikely to ever override as these are predbat emulating what's already programmed in the inverter

inverter_battery_rate_min, inverter_reserve_max and battery_rate_max likewise I think unlikely to ever change these once set

however inverter_limit_charge and inverter_limit_discharge I think would be useful to add as this would give dynamic control in HA as to what rate predbat instructs the inverter to operate at

car_charging entries, useful for people with more than one car

days_previous & weight, yes can see this would be useful

battery_scaling, can't see the need for this, its a set once for your battery and its done

forecast_hours, yes could occasionally be useful to change

import_export_scaling, another one I think is a set once and forget

My only other questions are about how/how long these overrides persist, do they survive a restart of Predbat (I assume they will), a restart of HA? 
Some guidance needed in the documentation about not using these for long term changes to the predbat config as its very easy to zap the overrides from the UI, best to only use for temporary changes

gcoan, 2024-10-06:
So now I've broken Predbat and its not proving easy to get it back ...

Experimenting with setting the manual_api programatically from a dashboard, I added a new button that was intended to construct the import rate override from some existing input_datetime helpers and pass all that to Predbat.

My dashboard card and button code:

```
type: entities
entities:
  - entity: input_datetime.octopus_power_up_date
  - entity: input_datetime.octopus_power_up_start_time
  - entity: input_datetime.octopus_power_up_end_time
  - type: button
    name: Send Power Up event to Predbat
    icon: mdi:script-text-play-outline
    action_name: Execute
    tap_action:
      action: call-service
      service: select.select_option
      target:
        entity_id:
          - select.predbat_manual_api
      data:
        option: >-
          rates_import_override?date={{
          states('input_datetime.octopus_power_up_date')| as_timestamp |
          timestamp_custom('%Y-%m-%d') }}2024-10-07&start={{
          states('input_datetime.octopus_power_up_start_time') }}&end={{
          states('input_datetime.octopus_power_up_end_time') }}&rate=0
```

Unfortunately this passes the entire string, not processed as YAML through as a select option to Predbat:

2024-10-06 23:00:05.201827 INFO pred_bat: select_event: manual_api, select.predbat_manual_api = rates_import_override?date={{ states('input_datetime.octopus_power_up_date')| as_timestamp | timestamp_custom('%Y-%m-%d') }}2024-10-07&start={{ states('input_datetime.octopus_power_up_start_time') }}&end={{ states('input_datetime.octopus_power_up_end_time') }}&rate=0

Which then crashes:

```
2024-10-06 23:08:08.187857 INFO pred_bat: Adding rate rates_import_override: {'date': '2024-10-03', 'start': '11:00:00', 'end': '14:00:00', 'rate': 0, 'load_scaling': 1.0} => 10-06 11:00:00 to 10-06 14:00:00 @ 0.0 date 2024-10-03 00:00:00 increment False
2024-10-06 23:08:08.189870 INFO pred_bat: Warn: can not resolve start value { states('input_datetime.octopus_power_up_start_time') }
2024-10-06 23:08:08.248603 INFO pred_bat: Info: record_status Warn: can not resolve start value { states('input_datetime.octopus_power_up_start_time') }
2024-10-06 23:08:08.250606 INFO pred_bat: Warn: can not resolve end value { states('input_datetime.octopus_power_up_end_time') }
2024-10-06 23:08:08.302948 INFO pred_bat: Info: record_status Warn: can not resolve end value { states('input_datetime.octopus_power_up_end_time') }
2024-10-06 23:08:08.305717 INFO pred_bat: Warn: can not resolve date value { states('input_datetime.octopus_power_up_date')| as_timestamp | timestamp_custom('%Y-%m-%d') }2024-10-07
2024-10-06 23:08:08.347884 INFO pred_bat: Info: record_status Warn: can not resolve date value { states('input_datetime.octopus_power_up_date')| as_timestamp | timestamp_custom('%Y-%m-%d') }2024-10-07
2024-10-06 23:08:08.349897 INFO pred_bat: Error: Exception raised strptime() argument 1 must be str, not None
2024-10-06 23:08:08.355139 INFO pred_bat: Error: Traceback (most recent call last):
  File "/homeassistant/appdaemon/apps/batpred/predbat.py", line 11378, in update_time_loop
    self.update_pred(scheduled=False)
  File "/homeassistant/appdaemon/apps/batpred/predbat.py", line 10177, in update_pred
    self.fetch_sensor_data()
  File "/homeassistant/appdaemon/apps/batpred/predbat.py", line 9440, in fetch_sensor_data
    self.rate_import = self.basic_rates(self.get_arg("rates_import_override", [], indirect=False), "rates_import_override", self.rate_import, self.rate_import_replicated)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/homeassistant/appdaemon/apps/batpred/predbat.py", line 3159, in basic_rates
    date = datetime.strptime(date_str, "%Y-%m-%d")
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: strptime() argument 1 must be str, not None
2024-10-06 23:08:08.389981 INFO pred_bat: Info: record_status Error: Exception raised strptime() argument 1 must be str, not None
```

Tried manually selecting [Off] option in select.predbat_manual_api but because of the crash the webhook isn't working

Tried restarting Appdaemon add-on, Predbat crashes same error

Tried calling the select.select action with the valid rates_import_override option string, but it just appends the valid option to the select and crashes with the invalid option:

```
2024-10-06 23:18:26.814560 INFO pred_bat: Basic rate API override items for rates_import_override are [{'index': None, 'date': "{{ states('input_datetime.octopus_power_up_date')| as_timestamp | timestamp_custom('%Y-%m-%d') }}2024-10-07", 'start': "{{ states('input_datetime.octopus_power_up_start_time') }}", 'end': "{{ states('input_datetime.octopus_power_up_end_time') }}", 'rate': '0'}, {'index': None, 'date': '2024-10-07', 'start': '17:00:00', 'end': '19:00:00', 'rate': '0'}]
```
[... 43 lines cut ...]
2024-10-06 23:23:48.155331 INFO pred_bat: Adding rate rates_import_override: {'date': '2024-09-24', 'start': '15:00:00', 'end': '17:00:00', 'rate': 0, 'load_scaling': 1.0} => 10-06 15:00:00 to 10-06 17:00:00 @ 0.0 date 2024-09-24 00:00:00 increment False
2024-10-06 23:23:48.159540 INFO pred_bat: Adding rate rates_import_override: {'date': '2024-09-26', 'start': '09:00:00', 'end': '11:00:00', 'rate': 0, 'load_scaling': 1.0} => 10-06 09:00:00 to 10-06 11:00:00 @ 0.0 date 2024-09-26 00:00:00 increment False
[... cut ...]
Love the idea of this, its very powerful and once I work out how to integrate it to my existing dashboard, it'll be great BUT need to think more about the error trapping and how to 'get out' of an invalid option such as I created just now.
Am a bit fearful of what will happen when I swap back to main as the select still has all those invalid values in it and I don't know how to clear it
[image]

(I am hoping that writing a script to do the formatting and call the select.select will work, this article suggests it will https://community.home-assistant.io/t/yaml-help-using-value-from-helper-to-select-option-in-list/506007/2)

springfall2008, 2024-10-07:
Okay, this is a bug in the rates code it should have caught the error, I've fixed this on 'main' so when you update it shouldn't crash.

I've now added documentation here: https://github.com/springfall2008/batpred/blob/main/docs/manual-api.md

Let me know if it works now.

gcoan, 2024-10-07:
Progress, and after a few false starts, it now works !

I upgraded to the latest main version and good news is that the bad date in the select is trapped now and Predbat didn't crash:
[image]

But in my testing I had some dummy values in the manual API dropdown which Predbat then executed and decided to do a force charge right in the peak period when import rates were spiking at 50p !
Had to keep repeatedly turning off the inverter charge schedules before I eventually managed to kill the manual API with an 'off' option.
I found selecting off from the dropdown didn't work in this instance, instead I had to set the value to off via a select.select action in developer tools.

And perfect timing, we have a power-up event tomorrow.

I moved the embedded button code into a script that I could execute from the dashboard, and it didn't work.  Predbat didn't crash but it ignored my import rate.

Found the problem was that in the script I was setting the option to "import_rates....... rate=0" - i.e. a set of double quotes around it which was what was being ignored by Predbat.

Removed these and it works.  See 20:06 onwards and the extra import rate being added at 20:07:00
[appdaemon.log](https://github.com/user-attachments/files/17283444/appdaemon.log)

Here's my script as a sample if you'd like to add it to the documentation:

```
alias: Send Powerup event to Predbat manual API
sequence:
  - action: select.select_option
    data:
      option: >-
        rates_import_override?date={{          
        states('input_datetime.octopus_power_up_date')| as_timestamp |          
        timestamp_custom('%Y-%m-%d') }}&start={{          
        states('input_datetime.octopus_power_up_start_time') }}&end={{          
        states('input_datetime.octopus_power_up_end_time') }}&rate=0
    target:
      entity_id: select.predbat_manual_api
mode: single
description: ""
```

The template code has to be in a script, it doesn't work directly in a dashboard.

Documentation looks good.  Maybe just put the list of supported overrides in alphabetical order as its a bit random at the moment.

Thanks a lot, this looks very useful, gives me what I wanted and should be extensible for lots of other niche automation use cases (like dynamic setting of the charge/discharge rate).

gcoan, 2024-10-16:
The new API is working perfectly, am able to automate Octopus power up events in Predbat from my dashboard.

Brilliant addition, thank you

closed 2024-10-16 by gcoan