springfall2008/batpred #1459: Chart inconsistencies with WebUI in v8.4.8

SwiftRR, 2024-09-16:
**Describe the bug**
Chart display on web portal doesn't fit vertically (on Safari iMAC):
[image]

Charts works fine on tablet and phone (IoS)

New PV7 chart
This is headed as 'Solar Forecast' but is displays the last historical 7 days which is not useful (for me). The vertical cut off screen is more obvious here:
[image]

**Expected behavior**
Charts fit screen!

**Predbat version**
v8.4.8

**Environment details**
GE AC3.0 AC coupled
Standard HAOS install on pi

**Screenshots**
See above

springfall2008, 2024-09-20:
Fix on 'main' for testing

SwiftRR, 2024-09-20:
I have tested the 'main' webUI charts and they all look good now. y axis fits and everything scales nicely.

Do we need the 'double inner scroll bar' that I see on computer (iMac)? The top little scroll bar just moves the title box up and down by a tiny amount. In the title box, could the bat be moved up a little and the vertical space reduced. It takes up a fair bit of space.

A nice correction of the y axis scale. Thanks

Rob

springfall2008, 2024-09-21:
I should probably get rid of the little top scroll bar, not sure how to....

closed 2024-09-22 by springfall2008