springfall2008/batpred #1460: Latest Octopus Energy integration causing issues with Predbat

deanjesper, 2024-09-16:
**Describe the bug**

Upgrading the Octopus Energy integration above v12.0.1 throws errors in the Predbat log:

  File "/config/predbat.py", line 9400, in fetch_inverter_data
    inverter.update_status(self.minutes_now)
  File "/config/inverter.py", line 913, in update_status
    discharge_start = datetime.strptime(self.base.get_arg("discharge_start_time", index=self.id), "%H:%M:%S")
  File "/usr/lib/python3.10/_strptime.py", line 568, in _strptime_datetime
    tt, fraction, gmtoff_fraction = _strptime(data_string, format)
  File "/usr/lib/python3.10/_strptime.py", line 349, in _strptime
    raise ValueError("time data %r does not match format %r" %
ValueError: time data 'unknown' does not match format '%H:%M:%S'

2024-09-16 22:02:09.111114: Info: record_status Error: Exception raised time data 'unknown' does not match format '%H:%M:%S'
2024-09-16 22:02:09.111155: Error: time data 'unknown' does not match format '%H:%M:%S'
Traceback (most recent call last):
  File "/config/hass.py", line 184, in timer_tick
    item["callback"](None)
  File "/config/predbat.py", line 11071, in update_time_loop
    raise e
  File "/config/predbat.py", line 11065, in update_time_loop
    self.update_pred(scheduled=False)
  File "/config/predbat.py", line 9889, in update_pred
    self.fetch_inverter_data()
  File "/config/predbat.py", line 9400, in fetch_inverter_data
    inverter.update_status(self.minutes_now)
  File "/config/inverter.py", line 913, in update_status
    discharge_start = datetime.strptime(self.base.get_arg("discharge_start_time", index=self.id), "%H:%M:%S")
  File "/usr/lib/python3.10/_strptime.py", line 568, in _strptime_datetime
    tt, fraction, gmtoff_fraction = _strptime(data_string, format)
  File "/usr/lib/python3.10/_strptime.py", line 349, in _strptime
    raise ValueError("time data %r does not match format %r" %
ValueError: time data 'unknown' does not match format '%H:%M:%S'

**Expected behavior**

There shouldn't be any errors

**Predbat version**

8.4.8

**Environment details**

----------------------------------------------------------
 Add-on version: 1.2.0
 You are running the latest version of this add-on.
 System: Home Assistant OS 13.1  (amd64 / generic-x86-64)
 Home Assistant Core: 2024.9.2
 Home Assistant Supervisor: 2024.09.1

**Screenshots**

N/A

**Log file**

See above

gcoan, 2024-09-16:
@deanjesper I'm running Octopus Integration v12.2.1 and Predbat v8.4.8 and there are no such errors in my logfile

Maybe try restarting the Predbat add-on and checking that the Octopus integration is setup correctly and that the appropriate events are all enabled per the predbat installation instructions

yellowduck, 2024-09-17:
I have the same error. I upgraded everything last night, and pulled in GivTCP v3 without realising there are big changes there as well. Running in docker mode. If I check the log, the first error occurs earlier than the one reported.

`2024-09-16 20:00:02.203698 INFO pred_bat: Inverter 0: Type GE GivEnergy
2024-09-16 20:00:02.206853 INFO pred_bat: Inverter 0 using Rest API http://192.168.68.100:6345
2024-09-16 20:00:02.209956 INFO pred_bat: Error: Exception raised ('Connection aborted.', ConnectionResetError(104, 'Connection reset by peer'))
2024-09-16 20:00:02.210473 INFO pred_bat: Warn: Inverter 0 unable to read REST data from http://192.168.68.100:6345/readData - REST will be disabled
2024-09-16 20:00:02.216369 INFO pred_bat: Info: record_status Inverter 0 unable to read REST data from http://192.168.68.100:6345/readData - REST will be disabled
2024-09-16 20:00:02.217683 INFO pred_bat: Warn: Inverter control auto restart trigger: REST read failure
`

yellowduck, 2024-09-17:
I have managed to roll back GivTCP, and the error has disappeared.

WillCalderwood, 2024-09-19:
I'm getting a similar error. There are various warning in the log file.
```
2024-09-19 08:36:14.832197: Warn: Inverter 0 read bad REST data from http://homeassistant.local:6345/runAll - REST will be disabled
2024-09-19 08:36:14.845699: Info: record_status Inverter 0 read bad REST data from http://homeassistant.local:6345/runAll - REST will be disabled
2024-09-19 08:36:14.845967: Warn: Inverter control auto restart trigger: REST read failure
2024-09-19 08:36:14.846015: Info: auto_restart not defined in apps.yaml, Predbat can't auto-restart inverter control
2024-09-19 08:36:14.846055: Inverter 0 does not have timed pause support enabled
2024-09-19 08:36:14.846090: Warn: Return bad float value unknown from soc_max using default 10.0
2024-09-19 08:36:14.859772: Info: record_status Warn: Return bad float value unknown from soc_max
2024-09-19 08:36:14.860643: Warn: Unable to read inverter time string unknown using formats ['%Y-%m-%dT%H:%M:%S%z', '%Y-%m-%d %H:%M:%S%z', '%H:%M:%S']
2024-09-19 08:36:14.860743: Warn: Inverter control auto restart trigger: Unable to read inverter time
2024-09-19 08:36:14.860783: Info: auto_restart not defined in apps.yaml, Predbat can't auto-restart inverter control
2024-09-19 08:36:14.860931: Warn: Return bad float value unknown from reserve using default 0.0
```

The is the error. I've added an extra logging line to `inverter.py` to try and work out what's going on, so the error is on 913.
```
  File "/config/predbat.py", line 11065, in update_time_loop
    self.update_pred(scheduled=False)
  File "/config/predbat.py", line 9889, in update_pred
    self.fetch_inverter_data()
  File "/config/predbat.py", line 9400, in fetch_inverter_data
    inverter.update_status(self.minutes_now)
  File "/config/inverter.py", line 914, in update_status
    discharge_start = datetime.strptime(self.base.get_arg("discharge_start_time", index=self.id), "%H:%M:%S")
  File "/usr/lib/python3.10/_strptime.py", line 568, in _strptime_datetime
    tt, fraction, gmtoff_fraction = _strptime(data_string, format)
  File "/usr/lib/python3.10/_strptime.py", line 349, in _strptime
    raise ValueError("time data %r does not match format %r" %
ValueError: time data 'unknown' does not match format '%H:%M:%S'
```
It looks like the inverter isn't returning what's expected to me.

gcoan, 2024-09-19:
@WillCalderwood those errors point to a communications error with your inverter.  Is there anything in the GivTCP logs?

Try deleting all the PKL files out of /config/GivTCP and restarting GivTCP

yellowduck, 2024-09-19:
What version of GivTCP are you using? This is the error I received when upgrading to latest (v3).

WillCalderwood, 2024-09-19:
My bad... I got a new wi-fi setup the other day and my reserved IP for the inverter appears to no longer be working.

gcoan, 2024-09-19:
> I have the same error. I upgraded everything last night, and pulled in GivTCP v3 without realising there are big changes there as well. Running in docker mode

have you seen this other issue with docker setup advice for v3 https://github.com/britkat1980/giv_tcp/issues/236 ?

WillCalderwood, 2024-09-19:
Just to confirm - sorting my IP resolved the issue. I didn't realise my mesh system had it's own DHCP server.

gcoan, 2024-09-19:
> Just to confirm - sorting my IP resolved the issue. I didn't realise my mesh system had it's own DHCP server.

When I installed my mesh wifi I ended up turning off the mesh DHCP and leave the attached broadband router to do the IP allocation for everything, both on its wifi or on the mesh wifi.  Made things a lot simpler

closed 2024-09-22 by springfall2008