springfall2008/batpred #1461: Energy rate from Nordpool spot price

johanzander, 2024-09-17:
I'd like batpred to derive my import / export energy rates from Nordpool spot prices.
I believe this is a common scenario for users in Scandinavia with variable energy tariffs. 
Personally I use Tibber who only add a markup on the hourly Nordpool spot price.

One could use the Nordpool API directly, or use the HACS Nordpool integration which gives a neat sensor that could be used directly in batpred. 
In either case one would need to select area (e.g. 'SE4') and currency (e.g. SEK).

My actual rates would be :
Import rate: ((Nordpool spot price for SE4 in SEK) + Tibber 0.08 SEK markup) * 25% VAT + 1.03 SEK (fixed cost to grid owner + tax).
Export rate: (Nordpool spot price for SE4 in SEK) + 0.6518  SEK (tax subsidies) 

Can this be achieved today in batpred?
Otherwise, consider this a feature request :-).

gcoan, 2024-09-17:
Probably the closest solution out of the box with predbat is the octopus rate url https://springfall2008.github.io/batpred/energy-rates/#octopus-rates-url where predbat will invoke api's to get the import and export rates. 

If you can (eg via a template) reformat the norddata api to add the appropriate adders and present the data as octopus do then this would work out of the box.

Something like that would be more expandable than a bespoke solution specific for norddata.

springfall2008, 2024-09-22:
There is already a Nordpool HA integration which Predbat can use, its even in the template:

metric_octopus_import: 'sensor.nordpool_kwh_oslo_eur_3_10_025'

https://github.com/custom-components/nordpool

closed 2024-10-13 by springfall2008