springfall2008/batpred #1462: **Possible problem with apps.yaml restart and using givTCP v3.x rather than v2.x**

SwiftRR, 2024-09-18:
**Possible problem with apps.yaml restart and using givTCP v3.x rather than v2.x**

I have picked up this potential problem from one of the Facebook forums:

> For any Predbat users, you must ensure that the auto restart code (in apps.yaml) is commented out or pointed to the new container ID.
> Consequence is that otherwise Predbat will restart V2 and 2 instances will then be running together. This will cause issues
```
>  # When enabled automatic restart will restart the add-on if communication fails

  # Example below is auto-restart for GivTCP add-on itself
  auto_restart:
    - shell: 'rm -rf /homeassistant/GivTCP/*.pkl'
    - service: hassio/addon_restart
      addon: a6a2857d_givtcp
```
I have temporarily commented out these lines in my apps.yaml.

This needs assigning to @"geoffreycoan"#p63978 as a documentation fix

Rob

gcoan, 2024-09-18:
The slug id for GivTCP v3 is 533ea71a_givtcp so the apps.yaml line needs to be changed to:

```
auto_restart:
    - shell: 'rm -rf /homeassistant/GivTCP/*.pkl'
    - service: hassio/addon_restart
      addon: 533ea71a_givtcp
```

I'll put this in the documentation. Already on my list is to properly document the GivTCP and MQTT installation process as its not explicit, so I'll wrap this change in with that

gcoan, 2024-09-21:
changes made to documentation to explain different slug id for auto restart for givtcp v2 and v3

closed 2024-09-21 by gcoan