springfall2008/batpred #1469: Internet connection failure not trapped, predbat freezes

gcoan, 2024-09-20:
**Describe the bug**
We had an extended period of the internet being out whilst an open reach engineer tried to fix an error on the line.  Looking at Predbat part way through I could see it had frozen, the plan wasn't being updated, but predbat status was still Idle and error=false.

Looking at the logfile, predbat attempts to retrieve the norddata future prices, fails, but this error isn't trapped and predbat just crashes, making no execution of the current plan. It then reruns 5 minutes later with the same happening again.

When I commented the norddata lines out of apps.yaml, predbat resumed normal execution, updating the plan, etc.

**Expected behavior**
Either:
a) Failure to connect to norddata results in Predbat trapping the error, status and the error attribute being set appropriately

or

b) Since norddata is not an essential part of predbat's current plan, its just to give a more accurate plan for tomorrow, the error is trapped and predbat just continues executing as normal just without the norddata input.

b) is probably slightly better behaviour as long as the error is clearly logged in the logfile

**Predbat version**
8.4.8

**Environment details**
HAOS 2024.9.2 with predbat running in Appdaemon

**Screenshots**
Frozen predbat plan, current time is 10:35 but plan stuck at 9:25:
[image]

Predbat status unchanged from Idle earlier in the morning, error attribute not set:
[image]

**Log file**
To be added

gcoan, 2024-09-20:
Worth highlighting the similarity to #1348 thread starvation issue, also not trapped by predbat

springfall2008, 2024-09-21:
Fix on main for testing.

I'm not sure really how to fix the thread starvation one..

gcoan, 2024-09-21:
Thanks, I'll give it a test tomorrow

gcoan, 2024-09-22:
I gave it a test, disconnecting the internet, and didn't get any errors, but looking at the code and the failing logfile, it says it doesn't start trying to retrieve the norddata prices until 11CET, and in my log the failures start at 9:30 BST

In my test today I think I disconnected the internet too late and Predbat had already got the cached rates for tomorrow.

I'll test again tomorrow and disconnect the internet earlier on

gcoan, 2024-09-23:
Thanks Trefor, disconnected the internet earlier and at around 9:30 Predbat started trying to download the norddata prices, traps the error and continues:

[image]

So that's good, thank you

However I do notice that the status oscillates between a 'Warn: unable to download' and 'Warn: Error downloading' - it would be preferable if it just stuck on a single warning status as the regularly changing status makes it harder to trap in an automation:

[image]