springfall2008/batpred #1478: Solax (SX4) inverter config.py appears to be incorrect for service call method

jefft4, 2024-09-23:
**Describe the bug**
I configured my SX4 inverter to use the service calls, as direct modbus control from predbat wasn't working.
The service calls are also not working - no actions from predbat in the HA history at all.

This seems to be because config.py for the SX4 type says 'has_service_api' is false.
If I read the code correctly, that means that these service calls will never be used, it'll be trying to use charge_control_immediate, which calls enable_charge_discharge_with_time_current and tries to set time periods.

**Expected behavior**
Looks like two variants of the SX4 might be needed, unless predbat can handle having two ways to control the inverter and selecting whichever is defined - (1) has_service_api = true and (2) has_service_api = false.

**Predbat version**
8.4.8

**Environment details**
SolaX X1 Hybrid G4 with battery
Home Assistant container on Ubuntu 22.04
Predbat in docker

springfall2008, 2024-09-29:
Please try 'main' which will now call the service if its configured in apps.yaml, I removed has_service_api from apps.yaml totally.

jefft4, 2024-09-30:
I'm getting this in the log - "Warn: Inverter 0 unable to find service template for charge_start_service" - only for charge_start_service, not the others - but it does appear to have started charging.
Definitions (lifted from the UI):
`charge_start_service
service	: input_select.select_option
entity_id	: input_select.solar_control_mode
option	: Charge

charge_stop_service
service	: input_select.select_option
entity_id	: input_select.solar_control_mode
option	: Stop Charge

charge_start_service
service	: input_select.select_option
entity_id	: input_select.solar_control_mode
option	: Self-use
`

jefft4, 2024-09-30:
The log in full:
```
0413	2024-09-30 15:05:03.385730: Warn: Inverter 0 unable to find service template for charge_start_service
10412	2024-09-30 15:05:03.382258: Inverter 0 Call service input_select/select_option with data {'entity_id': 'input_select.solar_control_mode', 'option': 'Charge'}
10411	2024-09-30 15:05:03.382236: Inverter 0 Call service template charge_start_service = {'service': 'input_select.select_option', 'entity_id': 'input_select.solar_control_mode', 'option': 'Charge'}
```

I may have spoken too soon about control working, too - the plan shows hold charge from 15:00 and we're still in charge mode as of 15:11... will update when we get to 16:00 and it should switch to discharging (self-use).

chalfontchubby, 2026-08-20:
Believed fixed by #1493. Please open a new issue if there is still a problem.

closed 2026-08-20 by chalfontchubby