springfall2008/batpred #1481: v4.8.9 fails on appdaemon in hacs

yellowduck, 2024-09-24:
2024-09-24 08:10:02.096547 WARNING pred_bat: ------------------------------------------------------------
2024-09-24 08:10:17.094962 WARNING pred_bat: ------------------------------------------------------------
2024-09-24 08:10:17.095265 WARNING pred_bat: Unexpected error in worker for App pred_bat:
2024-09-24 08:10:17.095503 WARNING pred_bat: Worker Ags: {'id': '42abffd7ff82448593cbed73330dbb6a', 'name': 'pred_bat', 'objectid': '044dae44d239491e94ddf175717f8709', 'type': 'scheduler', 'function': <bound method PredBat.update_time_loop of <predbat.PredBat object at 0x7ffefc3043a0>>, 'pin_app': True, 'pin_thread': 0, 'kwargs': {'interval': 15, 'random_start': 0, 'random_end': 0, '__thread_id': 'thread-0'}}
2024-09-24 08:10:17.095722 WARNING pred_bat: ------------------------------------------------------------
2024-09-24 08:10:17.096095 WARNING pred_bat: Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/appdaemon/threading.py", line 1022, in worker
    funcref(self.AD.sched.sanitize_timer_kwargs(app, args["kwargs"]))
  File "/apps/apps/batpred/predbat.py", line 11089, in update_time_loop
    self.check_entity_refresh()
  File "/apps/apps/batpred/predbat.py", line 11132, in check_entity_refresh
    self.ha_interface.db_tick()
AttributeError: 'HAInterface' object has no attribute 'db_tick'

matttheotter, 2024-09-24:
@yellowduck going to assume you have done a server reboot?

springfall2008, 2024-09-29:
It appears like ha.py hasn't been updated correctly?

closed 2025-01-01 by springfall2008