springfall2008/batpred #1482: v2.4.9 has detrimental effect on Plan with Agile

SwiftRR, 2024-09-24:
**Describe the bug**
Recent versions of Predbat have produced good working plans when running Octopus Agile import.
On upgrading from v2.4.8 to v2.4.9, the effect has been for the battery SOC to reduce drastically with most action becoming freeze discharging. My battery SOC has reduced to settle on 30-40% SOC rather than heading for a full state.
I downloaded and ran v2.4.9 last night and have now reverted back to v2.4.8. The screenshots tell the story well.
**See screenshots below, run one after the other from downgrading.**

The description of the planning changes in v2.4.9 seems to be in response to issues with freeze charging and discharging and to improve plans for IOG. Whatever has changed within Predbat has had a detrimental effect on plans serving Agile. The v2.4.9 is producing a plan that is unlike the pattern generated over many Predbat versions.  

**Expected behavior**
Charging during lower Agile rate slots, ideally reaching 100% overnight. Then daytime generation serving house load with excess solar exporting to. Similar plans to those produced in the past.

**Predbat version**
v2.4.9

**Environment details**
- Inverter and battery setup:   GE AC3.0 inverter feeding two 8.2 kWh batteries
- Standard HAOS installer:      HAOS 2024.9.2 running on raspberry pi 4 on SSD 
- givTCP v3.0.0:                        Running smoothly

**Screenshots**
[image]

[image]

**Log file**
No real additional evidence

gcoan, 2024-09-24:
Hi @SwiftRR not disagreeing with your concern over the 8.4.9 plan as I also am a bit concerned about the increased weighting to Freeze Discharge, but just digging into the ‘is one better or worse than the other’ question and (out loud) walking through the process I follow:

I find the best way is to look at the plan at a certain point in time, so for example, the 8.4.9 plan is -£2.05 at midnight (27% SoC) whereas the 8.4.8 is -£0.27 (97% SoC).

The extra SoC (97-27 = 70%) has a value. You have 2x 8.2kWh batteries = 16.4kWh, 70% of that is 11.48kWh

If you exported that extra 11.48kWh at 13.97p (after losses) you would earn £1.60 extra export income.

Adjust the 8.4.8 plan with that extra income -0.27 - 1.60 = -£1.87 which is a worse financial result than the 8.4.9 plan, so the 8.4.9 plan is actually better at this point in time.

It can be a bit more complicated than that because in reality it depends on what the import prices are after the (midnight) point in time and whether you would be importing from grid or not and at what future import price, but it gives a way to compare the “betterness” of different plans.

Of course we all want slightly different things - keep plenty in the battery in reserve in case of extra load/price spikes, not use the battery to minimise the charge/discharge losses, etc so we'll probably never get to a plan that pleases everyone, but if we have appropriate levers of control then we can tweak to what we want.

@springfall2008 welcome your thoughts, what levers @SwiftRR can pull to get back more to the 8.4.8 plan,  whether you agree with my plan comparison?

PianSom, 2024-09-27:
I am an IOG user and have been trying out 2.4.9 for the last few days. As has been noted, I am experiencing quite an increase in Freeze Discharge settings, which is generally good as it saves pointless charging and consequent conversion losses.

My feedback is:
1. I am still getting a combined Discharge and Charge slot in the 23:30 slot. I have been addressing this by manually entering a Force Charge for that slot, which solves the issue. In addition this sometimes (but not always) addresses issue 2. More info and Trefor's previous suggestions [here](https://github.com/springfall2008/batpred/issues/1429)  

2. I am often getting a period of Discharge at around 18:30-19:30. This is suboptimal, since occasionally as the evenings grow colder I am increasing my evening load (A2A heat pump to warm a room) and after an unneeded Discharge find that I  empty my battery before the cheap rate starts at 23:30, causing peak rate import. A better plan would - all things being equal - delay discharging until later in the day. There seems to be no reason for this early Discharge.

gcoan, 2024-09-28:
Changing best_soc_keep has improved the Soc level retained in my battery and reduced the amount of freeze discharge- see https://github.com/springfall2008/batpred/issues/1489#issuecomment-2380944055

PianSom, 2024-09-29:
Thanks for the suggestion @gcoan 

Only a n=1 result, but changing best_soc_keep to 1.5 this morning does not address either issue for me - still have a discharge at 19:30 and at 23:30. 

Setting the 23:30 to a manual charge slot solved both issues. Go figure. 🤷

SwiftRR, 2024-09-29:
I started this issue concerned about v8.4.9 and Agile. (I have the version number right this time!)
After a week, I don't think there is an issue but there does seem to be an ongoing problem with IOG.

I waited until the 4 pm rates appeared and this is Predbat's plan. It looks exactly as I would expect times discharge to pick up the low overnight Agile slots.

[image]

I don't have any suggestion for the IOG issue, I am afraid.

Rob

gcoan, 2024-09-29:
I too have got used to the new freeze discharge plans produced by 8.4.9.  Today with low prices predicted tonight and overnight Predbat has alternated between freeze discharge and idle all day (mainly freeze discharge so the solar gets exported), and the batteries are currently 54% full to get me through the evening with just a single part-block forced discharge planned to empty the batteries before the cheap fill periods.

Quite happy with how this is operating on Agile.  Increasing best_soc_keep to 1.5 (from 0.5) has given me a bit more energy buffer in case of unexpected demand, but Predbat is making sure I use it all up before recharging at the cheap overnight rates.

SwiftRR, 2024-09-29:
I am still on 0.5 for best_soc_keep and it seems to be working well for my Agile set up and usage.

I am seeing a lot of 'freeze discharging 99' during the day with the SOC often about 50. This seems to result in a slow trickle charge from excess solar, but with most going to grid (if there are low rates predicted at night from Nordpool).
I think this status means 'discharge from 99', which it can't do. Have I learnt what 'freeze discharging' means?

Rob

gcoan, 2024-09-29:
> I am seeing a lot of 'freeze discharging 99' during the day with the SOC often about 50. 
> I think this status means 'discharge from 99', which it can't do. 

I think of it as freeze the SoC at a maximum of 99% but allow discharging below that level.

> Have I learnt what 'freeze discharging' means?

I don't think we ever will ....

SwiftRR, 2024-09-30:
> I think of it as freeze the SoC at a maximum of 99% but allow discharging below that level.

I didn't quite get my 'understanding' right before!

I meant _"freeze discharging means 'discharge **to** 99"._

If SOC is 50, discharging is impossible, so Predbat shouldn't even try.
I think that discharging is only possible if SOC is 100 or 99 (when it should keep SOC at 99).

But if SOC is 50, I do see charging in my plan. This is proposed plan for tomorrow (Tue 1st Oct).

[image]

This contradicts documentation:
https://springfall2008.github.io/batpred/what-does-predbat-do/

_So my understanding is 'One step forwards, two steps back'._

Rob

gcoan, 2024-09-30:
The documentation doesn't really cover what the 'limit' does when in Freeze Discharge mode.  Maybe the limit has no bearing on Predbat's behaviour (for freeze discharge) and the documentation should be explicit about this 🤷‍♂️?

"Freeze discharging - The battery is in Discharge mode, the same as Idle (Eco) mode, but with charging disabled. The battery or solar covers the house load. As charging is disabled, if there is excess solar generated, the current SoC level will be held and the excess solar will be exported. If there is a shortfall of solar power to meet house load, the battery will discharge."

Key points I take from this is that battery is in discharge mode and it shouldn't be charging.  The plan you show shows export happening (good) but also shows SoC increasing which as you say contradicts the documentation.  So looks like a Freeze Discharge bug to me @springfall2008 

My own plan for tomorrow shows no such SoC increasing behaviour.  In my plan the freeze discharge periods show SoC either decreasing (excess load over solar) or SoC remaining constant:
[image]

Maybe it's because I have hybrid inverters and you have AC coupled that Predbat's behaving differently for the two of us?

PianSom, 2024-09-30:
Just to confirm, when my AIO is in Freeze Discharge mode it DOES charge, At approx 500W, for some reason.

Last Sat morning was sunny - here is a graph showing the constant battery charge in the morning, with other solar being exported. (I got bored of Freeze Discharge at 11.30am, and decided to change up the battery for an expected high demand that evening. Predbat status message below.)

[image]

[image]

Off topic - I wish I could figure why Predbat inserted that spike at 2.45am while discharging. Happens most days.

SwiftRR, 2024-09-30:
> The documentation doesn't really cover what the 'limit' does when in Freeze Discharge mode. Maybe the limit has no bearing on Predbat's behaviour (for freeze discharge) and the documentation should be explicit about this 🤷‍♂️?

I see 'Limit' influencing Predbat's behaviour in putting together the plan. It isn't mentioned in 'What does Predbat do?' but does feature in https://springfall2008.github.io/batpred/predbat-plan-card/#understanding-the-predbat-plan:
_Limit % - Alongside any battery activity (charging, discharging, etc) there will be a SoC limit. This limit is what the SoC is planned to be at the end of the 30 minute time slot. e.g. 'Charge↗ 70%' is charge to 70% SoC, and 'Discharge↘ 4%' is discharge the battery to the 4% reserve level._

Since upgrading to v8.4.9., 'Freeze discharging 99' shows up regularly for me during daylight hours if my SOC is in the middle, say 30-70%.
And 'normal planned behaviour' for me is a **trickle charge** as I have mentioned and shown before.

> Maybe it's because I have hybrid inverters and you have AC coupled that Predbat's behaving differently for the two of us?

An interesting idea. Predbat is certainly attempting to be 'Jack of all trades', formulating plans from many different setups.

**Here's a related curved ball .....**
When weighing up losses, AC coupled is treated differently from hybrid, based on the extra DC to AC to DC conversion from PV generation to storage in battery.
But my PV generation mostly goes to my house load (including ASHP) or is exported to the grid. My battery is charged mainly from the grid and Agile makes good use of this during low rate slots. 
.... so in practice, should AC coupled and hybrid inverters be treated differently?
The defaults shown in the Web UI are 

- battery loss charge: 0.03
- battery loss discharge : 0.03
- inverter loss for AC coupled: 0.04 (nominally 0.00 for hybrid)

I think that contribution by inverter loss from AC coupled may be negligible or certainly much less than default.
Documentation has higher defaults of 0.05 for charge and discharge losses: https://springfall2008.github.io/batpred/customisation/#battery-loss-options

Of course, I can tweak the inverter loss value myself and we are encouraged anyway to experiment with loss values.
I need to often remind myself that Predbat is **predicting** what is needed in the future and this is not necessarily what **will** happen. Predbat certainly changes its mind and all our plans are dynamic, not set in stone.

Rob

gcoan, 2024-09-30:
The problem with the limit definition is that it only covers charging and discharging, not freeze charging and freeze discharging, and by definition, the freeze variants are intending to freeze the SoC, not charge/discharge to a specific value.

A thought on the trickle charge that @SwiftRR and @PianSom both say you see on your AIO's, I wonder if this is because PauseCharge mode isn't being set?   The original (Gen 1) firmware would pause almost all discharging or charging simply by setting the discharge/charge rate to zero.  With newer firmware I believe this doesn't happen, you get a trickle charge happening unless battery pause mode is set, when I think you still get a trickle but it's much reduced.

When it's next freeze discharging, take a look at what the battery pause mode is set to?

None of which explains your plan @SwiftRR where Predbat **predicts** the SoC to rise on a freeze discharge. Doesn't make sense and doesn't match what my plan predicts.  I can't explain that behaviour.

As you say, inverter AC losses do play a part but not that significant a %

SwiftRR, 2024-09-30:
> The problem with the limit definition is that it only covers charging and discharging, not freeze charging and freeze discharging, and by definition, the freeze variants are intending to freeze the SoC, not charge/discharge to a specific value.

So what does Freeze discharging 99 actually mean? 

> The original (Gen 1) firmware would pause almost all discharging or charging simply by setting the discharge/charge rate to zero. With newer firmware I believe this doesn't happen, you get a trickle [...]

My AC coupled inverter received an upgrade earlier this year. Following this, I was able to use the more advanced giTCP settings and pause works. This is from my givTCP log:

[image]

Perhaps @gcoan's answer to my earlier question sums up the freeze statuses well:

> @SwiftRR "Have I learnt what 'freeze discharging' means?"
> @gcoan "I don't think we ever will ...."

Time for me to '_let freezing discharges lie_' ..........

Rob

PianSom, 2024-09-30:
@gcoan - here are my GivTCP logs for the period above (with apologies for the abysmal formatting!). 

Note, in particular, 
```
2024-09-28 05:30:23,517 - Inv1 - write       -  [INFO    ] - Setting Battery Pause Mode to PauseCharge was a success
```

<details><summary>GivTCP logs</summary>
<p>

28/09/2024
00:00:06
2024-09-28 00:00:06,456 - Inv1 - read        -  [INFO    ] - Midnight, so resetting Day/Night stats...
28/09/2024
00:00:16
2024-09-28 00:00:16,023 - Inv1 - write       -  [INFO    ] - Setting Charge Target to: 100
28/09/2024
00:00:17
2024-09-28 00:00:17,875 - Inv1 - write       -  [INFO    ] - Setting Charge Target 100 was a success
28/09/2024
00:00:28
2024-09-28 00:00:28,238 - Inv1 - read        -  [INFO    ] - Midnight, so resetting Day/Night stats...
28/09/2024
00:00:28
2024-09-28 00:00:28,239 - Inv1 - read        -  [ERROR   ] - inverter Update failed so using last known good data from cache: (<class 'TypeError'>, TypeError("unsupported operand type(s) for /: 'float' and 'list'"), <traceback object at 0x7f459120d980>)
28/09/2024
00:00:48
2024-09-28 00:00:48,740 - Inv1 - read        -  [INFO    ] - Midnight, so resetting Day/Night stats...
28/09/2024
00:00:48
2024-09-28 00:00:48,740 - Inv1 - read        -  [ERROR   ] - inverter Update failed so using last known good data from cache: (<class 'TypeError'>, TypeError("unsupported operand type(s) for /: 'float' and 'list'"), <traceback object at 0x7f45911b1c00>)
28/09/2024
02:20:09
2024-09-28 02:20:09,876 - Inv1 - write       -  [INFO    ] - Setting Discharge Slot 1 to: 02:20 - 04:01
28/09/2024
02:20:12
2024-09-28 02:20:12,465 - Inv1 - write       -  [INFO    ] - Setting Discharge Slot 1 was a success
28/09/2024
02:20:12
2024-09-28 02:20:12,474 - Inv1 - write       -  [INFO    ] - Setting Battery Mode to: Timed Export
28/09/2024
02:20:14
2024-09-28 02:20:14,569 - Inv1 - write       -  [INFO    ] - Setting export mode was a success
28/09/2024
02:20:15
2024-09-28 02:20:15,917 - Inv1 - write       -  [INFO    ] - Enabling Discharge Schedule was a success
28/09/2024
02:20:15
2024-09-28 02:20:15,946 - Inv1 - write       -  [INFO    ] - Setting battery reserve target to: 28
28/09/2024
02:20:17
2024-09-28 02:20:17,304 - Inv1 - write       -  [INFO    ] - Setting shallow charge 28 was a success
28/09/2024
02:20:35
2024-09-28 02:20:35,285 - Inv1 - write       -  [INFO    ] - Setting Discharge Slot 1 to: 00:00 - 04:01
28/09/2024
02:20:38
2024-09-28 02:20:38,098 - Inv1 - write       -  [INFO    ] - Setting Discharge Slot 1 was a success
28/09/2024
02:30:18
2024-09-28 02:30:18,390 - Inv1 - write       -  [INFO    ] - Setting battery reserve target to: 32
28/09/2024
02:30:19
2024-09-28 02:30:19,728 - Inv1 - write       -  [INFO    ] - Setting shallow charge 32 was a success
28/09/2024
02:40:18
2024-09-28 02:40:18,912 - Inv1 - write       -  [INFO    ] - Setting Charge Slot 1 to: 00:00 - 02:45
28/09/2024
02:40:21
2024-09-28 02:40:21,714 - Inv1 - write       -  [INFO    ] - Setting Charge Slot 1 was a success
[... 221 lines cut ...]
               ^^^^^^^^^^^^^^^^^^^^^^
28/09/2024
10:15:09
  File "/usr/local/lib/python3.11/site-packages/flask/app.py", line 2138, in make_response
28/09/2024
10:15:09
    raise TypeError(
28/09/2024
10:15:09
TypeError: The view function for 'rdData' did not return a valid response. The function either returned None or ended without a return statement.
28/09/2024
11:31:49
2024-09-28 11:31:49,114 - Inv1 - write       -  [INFO    ] - Setting battery charge rate to: 6000 (50)
28/09/2024
11:31:50
2024-09-28 11:31:50,476 - Inv1 - write       -  [INFO    ] - Setting battery charge rate 50 was a success
28/09/2024
11:31:50
2024-09-28 11:31:50,506 - Inv1 - write       -  [INFO    ] - Setting Battery Pause Mode to: Disabled
28/09/2024
11:31:51
2024-09-28 11:31:51,636 - Inv1 - write       -  [INFO    ] - Setting Battery Pause Mode to Disabled was a success

</p>
</details>

SwiftRR, 2024-09-30:
You have quite a few errors in your givTCP log.

My givTCP log from 01:00 and now (20:37) has 3 time out errors but nothing like this.
I am on givTCP v3.0.0 and I have certainly had ups and downs upgrading from givTCP v2.4.9.
But it seems to be running smoothly now.

You do have some errors for your dodgy 23:30 slot.
See if you can capture the log tonight between say 23:00 and 00:30 to see if anything is revealed.

Rob

SwiftRR, 2024-10-01:
Last call on _Freeze discharging_

This is the transformation of a block of **dark** grey _Freeze discharge 99_ into today plan with some in **light** grey:

[image]

Compare this plan with the copied plan several comments above.
This is just as I would expect. _Freeze discharge_ is now **light** grey and has a limit of 73, same as SOC. This fits with documentation. 

It is almost as if **dark** grey _Freeze discharging 99_ is Predbat's holding pattern – almost a signal that it hasn't made its mind up yet!

Rob

PianSom, 2024-10-02:
@gcoan 
I am currently in Freeze Discharge mode. I can see the GivTCP log entry where Pause mode was set. I can confirm by looking at the portal that it was indeed successfully set. 

That notwithstanding, the battery is charging. (I believe at 500W, but there’s not enough sun to confirm that. )

SwiftRR, 2024-10-02:
Do you get 500 W charge with a status of Freeze Discharge 99 (_FD99_) in dark grey (the 99 is important)?
Currently my SOC is 46 and status is _FD99_ until 12 pm). This is impossible and the 'normal' pattern is then a trickle charge. My plan has SOC increasing by 1 every 30 min.

I wonder whether _FD99_ in dark grey is Predbat's 'holding pattern'. The only option is to charge or stay still.
In documentation, dark grey isn't mentioned as a colour. There is 'silver grey' for Freeze charge.
https://springfall2008.github.io/batpred/predbat-plan-card/#understanding-the-predbat-plan

_I think Freeze discharge is also silver grey when current SOC > limit, but not in documentation._

A real grey area .......

gcoan, 2024-10-02:
> I am currently in Freeze Discharge mode. I can see the GivTCP log entry where Pause mode was set. I can confirm by looking at the portal that it was indeed successfully set.

@PianSom if battery pause mode is set (and assuming that the pause start/end time slot are also set correctly to 00:00:00 and 23:59:00 respectively), and (double check) battery charge rate is set to zero, then I don't know how we stop the trickle charging on your inverter as these are the two control "levers" that we have AFAIK.

I will take a look at the code for the dark grey Freeze Discharge, if Trefor has added a new bit of code then its possible this isn't reflected in the documentation that I wrote a while ago by reverse engineering the code.

Still doesn't explain why the plan predicts SoC to increase on a freeze discharge. No idea on that one

PianSom, 2024-10-02:
>  and (double check) battery charge rate is set to zero

@gcoan - The register Battery Charge Power is indeed set to zero.

And, yes, I am in Freeze Discharge at 99%. I'd note that in your screenshots above @SwiftRR it seems that all your Freeze Discharges are dark grey (and all are 99%), while your Freeze Charges are light grey (at varying percentages) - no?

SwiftRR, 2024-10-02:
Here's a snapshot of my recent givTCP3.00 log when Predbat plan status was FD99. 
[image]
The log includes Pausedischarge and Pausecharge which may equate to GE's translation for Freeze discharge and Freeze charge.

The sky has lightened here and Predbat has tweaked its plan to:
[image]

Hopefully 🌥️ will make a proper appearance soon. Last 2 days have been 2.3 and 2.2 kWh. 2 days ago, I have nearly 3 inches of rain with another 1.5 inches yesterday. Sept solar was 75% of 2023. 

Rob

gcoan, 2024-10-02:
> I'd note that in your screenshots above @SwiftRR it seems that all your Freeze Discharges are dark grey (and all are 99%), while your Freeze Charges are light grey (at varying percentages) - no?

No, the freeze discharges in my plan were all at 99% and in a darker grey.

Looking at the code for where predbat outputs the html plan:
[image]

It appears that discharging with limit=99% is used as a 'special value' to indicate freeze discharges, and then when the plan is output the colour of the cell is then overwritten with dark grey.  I'd need to check further in the code of where the freeze discharges is actually instructed to the inverter, but my suspicion is that the limit 99 isn't actually a real value that's sent to the inverter at all in which case its misleading to include the limit in the plan.  
I suggest it would be better to just say 'freeze discharge' and not have a limit.  Will need to look at the code further though

gcoan, 2024-10-02:
> The log includes Pausedischarge and Pausecharge which may equate to GE's translation for Freeze discharge and Freeze charge.

Ignoring the 11:24 where I think we only see the end of a series of actions,

11:30 shows what I'd expect for freeze discharge (which is to stop charging but allow discharge), the charge rate is set to zero and the battery pause mode is set to PauseCharge

11:45 then shows the battery swapping to what looks like to freeze charge (stop discharge), the charge rate is set to 3000 and battery pause mode is set to PauseDischarge

Bit of a curious swap, but that's what the inverter instructions are.

Another wild thought on the trickle charge, maybe the AIO doesn't like the charge rate being set to zero% in the same way it doesn't like the battery reserve being set to 100.
As a test, when the inverter is next in freeze discharge but it is slowly charging, try setting the inverter charge rate to 1% and see if this changes the charge rate.   Predbat will overwrite the setting at its next 5 minute run so it's safe to do this.

SwiftRR, 2024-10-02:
Remember that I have an AC coupled inverter. Does hybrid produce a similar plan?

Here's my pattern 1:00pm - 1:35pm
Stats tell the story.
[image]

SwiftRR, 2024-10-02:
I have submitted an issue to givTCP, summarising all go the givTCP 3.0.0 add-on log glitches that I have encountered:
https://github.com/britkat1980/giv_tcp/issues/260

Rob

gcoan, 2024-10-02:
> I didn't quite get my 'understanding' right before!
> 
> I meant _"freeze discharging means 'discharge **to** 99"._
> 
> If SOC is 50, discharging is impossible, so Predbat shouldn't even try. I think that discharging is only possible if SOC is 100 or 99 (when it should keep SOC at 99).
> 
> But if SOC is 50, I do see charging in my plan. This is proposed plan for tomorrow (Tue 1st Oct).
> 
> <img alt="image" width="872" src="https://private-user-images.githubusercontent.com/68816364/372147098-caa89831-43e4-4c49-b33b-353977f9d82e.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjc5MDA2ODksIm5iZiI6MTcyNzkwMDM4OSwicGF0aCI6Ii82ODgxNjM2NC8zNzIxNDcwOTgtY2FhODk4MzEtNDNlNC00YzQ5LWIzM2ItMzUzOTc3ZjlkODJlLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMDIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDAyVDIwMTk0OVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWMzYjIxNGI4Y2YwNTNmZTA3M2Y4OGQ3YTY1OTNkYzk0ZDA0YzZhYzAwNjIyZjU2Nzk1ZjBlNWQxNzk2YzVlM2MmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.d7xUSjjTEEVOZLtCfOKTLdO8QTVFgVZDVFUZ0J4f8SY">
> This contradicts documentation: https://springfall2008.github.io/batpred/what-does-predbat-do/

I've been to look at the code and it is pretty much what I expected it to do (from line 8681 of predbat.py):

```
      if self.set_discharge_freeze and (self.discharge_limits_best[0] == 99):
                            # In discharge freeze mode we disable charging during discharge slots
                            inverter.adjust_charge_rate(0)
                            if inverter.inv_charge_discharge_with_rate:
                                inverter.adjust_discharge_rate(0)
                            inverter.adjust_pause_mode(pause_charge=True)
                            self.log("Discharge Freeze as discharge is now at/below target - current SOC {} and target {}".format(self.soc_kw, discharge_soc))
                            status = "Freeze discharging"
                            status_extra = " target {}%-{}%".format(inverter.soc_percent, self.discharge_limits_best[0])
                            isDischarging = True
                        else:
                            status = "Hold discharging"
                            status_extra = " target {}%-{}%".format(inverter.soc_percent, self.discharge_limits_best[0])
                            self.log(
                                "Discharge Hold (ECO mode) as discharge is now at/below target or freeze only is set - current SOC {} and target {}".format(
                                    self.soc_kw, discharge_soc
                                )
                            )
                            inverter.adjust_pause_mode()
```

A discharge limit of 99 is taken as a special value to set the inverter into freeze discharge mode.  The charge rate is set to zero, battery pause mode is set to pause charging, and the logfile messages say that its freeze discharging as soc is at or below the target of 99%.

All of which leads me to conclude that the limit of 99% we see on the plan is meaningless. It doesn't even get sent to the inverter, its an artefact of the way the predbat code internally "knows" it has got to set freeze discharge mode.

I have added the missing 'freeze discharge is dark grey' to the 'understanding the predbat plan' and I will change the code so that this limit of 99 isn't output for freeze discharging.

I also now think I understand probably why your inverter shows SoC rising in the slots where Predbat is freeze charging @SwiftRR . I believe it is because Predbat actually plans at 5 minute increments but the plan only shows 30 minute activity.  So during the 30 minute slot I surmise that predbat is going to both freeze discharge AND set the inverter to Idle/Eco mode.  That's why the plan doesn't look as if it makes sense, its not showing all the detail of the 5 minute slot.  This has caught me out before.

That covers off I think the Predbat plan parts of this github issue I believe, so maybe we can close it now?
@PianSom your issue of the AIO not freeze discharging properly is I think a different *physicality* issue with the AIO itself.  Its being sent the commands by Predbat to freeze and stop charging, but it doesn't.  As I suggested above it may be that it doesn't like the charge rate being set to zero and there's a different value needed, I don't know, or it may be that the AIO's can't physically do a freeze charging - but this sounds unlikely.  Either way, I think its a different issue from what we started on

SwiftRR, 2024-10-02:
I only became aware of freeze discharge 99 when I upgrade from v8.4.8 to v8.4.9 and noticed that blocks during the day were just shown with freeze discharge 99 status. The '99' was the issue and, as you have suggested, the 99 is meaningless.
You have added a comment about dark grey in documentation and 99 is going.

**So yes, please close this issue now!** 🙂

Rob

closed 2024-10-02 by gcoan