springfall2008/batpred #1484: 'Stepped' charging plan unnecessarily stops charging

gcoan, 2024-09-24:
**Describe the bug**
Today we had a 2 hour free electricity power up event (on a lowish solar day) so I wanted to take maximum benefit of this.
Prior to the event predbat had done freeze discharges so my 9.5kWh battery was just over half full and the 5.2 was empty - in 2 hours of charging both should therefore be filled.

[image]

Predbat's plan was to charge for the two hours, BUT it had steps of 62, 67, 100 and 100% SoC.

Both batteries started charging at 15:00

At 15:20 the 9.5 battery was at 65%, the 5.2 at 17%.  Since the 9.5 was above the 'step target' for that slot, Predbat put the 9.5 inverter into pause charge mode turned off the planned charge:
[image]
[image]
[image]

I tried setting the battery to force charge for the 2 hour period but this made no difference, the stepped plan stopped the 9.5 battery from charging when it should have been in the free electricity.

**Expected behavior**
Preferable solution:

Predbat takes account of battery capacity in setting the SoC target for each slot rather than setting a single SoC target that is just the average of the multiple battery SoC's.  This would give much better control of multi-inverter setups with differing battery capacities, something predbat doesn't handle very well at present (all batteries are assumed to be the same size).

Alternative solution:

When charging for a period of time at the same rate (in this case 2 hours at 100%), simply set the target rate to the end SoC target for the period rather than devising intermediate (and false) targets for the charging period.

**Predbat version**
8.4.9

**Environment details**
2024.9.2
2x GE Hybrid inverters, one with a 9.5, one with a 5.2 battery.

**Log file**

springfall2008, 2024-09-28:
Interesting issue, let me think about how to fix

springfall2008, 2024-09-29:
Would something like this work, where instead of setting flag SOC% you set the % based on the charge you plan to add for each battery?

The only thing is some might end up full so really you would want to account for that when setting the SOC target on the others....

    def adjust_battery_target_multi(self, inverter, soc, is_charging):
        """
        Adjust target SOC based on the current SOC of all the inverters accounting for their
        charge rates and battery capacities
        """
        required_kwh = self.soc_max * (soc / 100.0)
        add_this = required_kwh * (inverter.battery_rate_max_charge / self.battery_rate_max_charge)
        new_soc_kwh = min(inverter.soc_kwh + add_this, inverter.soc_max)
        new_soc_percent = calc_percent_limit(new_soc_kwh, inverter.soc_max)
        self.log("Inverter {} adjust target soc to {}% based on requested all inverter soc {}%".format(inverter.inverter_id, new_soc_percent, soc))
        inverter.adjust_battery_target(new_soc_percent, is_charging)

gcoan, 2024-09-29:
yes that sounds like a good approach @springfall2008 

Even as a first step to just add the SoC % increment to the charge-to target would stop the 'stopping prematurely' problem from occurring which is my problem at the moment.

Getting more sophisticated with working out those adjustments based on battery limits would be a nice further enhancement but not as critical to fixing the problem.  And if one battery does get full before the other then the excess will just get exported or (with the fast response gen 1 firmware or gen 2/3 behaviour) the other inverter will soak it up rather than let it get exported.

springfall2008, 2024-09-29:
A test version is on 'main', can you please check?

gcoan, 2024-09-29:
> A test version is on 'main', can you please check?

Will do. I'll install it now, the plan has lots of charging tonight

gcoan, 2024-09-29:
The main version was working fine but ;ooks like it crashes when trying to set the charge up:

```
2024-09-29 22:17:03.473007 INFO pred_bat: Adjust idle time, charge 22:30:00-23:00:00 discharge 00:00:00-00:00:00
2024-09-29 22:17:03.478644 INFO pred_bat: Adjust idle time computed idle is 22:10:00-22:30:00
2024-09-29 22:17:03.480345 INFO pred_bat: Error: Exception raised 'Inverter' object has no attribute 'inverter_id'
2024-09-29 22:17:03.484360 INFO pred_bat: Error: Traceback (most recent call last):
  File "/homeassistant/appdaemon/apps/batpred/predbat.py", line 11119, in update_time_loop
    self.update_pred(scheduled=False)
  File "/homeassistant/appdaemon/apps/batpred/predbat.py", line 9963, in update_pred
    status, status_extra = self.execute_plan()
                           ^^^^^^^^^^^^^^^^^^^
  File "/homeassistant/appdaemon/apps/batpred/predbat.py", line 8754, in execute_plan
    self.adjust_battery_target_multi(inverter, self.charge_limit_percent_best[0], isCharging)
  File "/homeassistant/appdaemon/apps/batpred/predbat.py", line 8346, in adjust_battery_target_multi
    inverter.inverter_id, soc, soc_percent, new_soc_percent, self.dp2(add_kwh), self.dp2(add_this), self.dp2(new_soc_kwh)
    ^^^^^^^^^^^^^^^^^^^^
AttributeError: 'Inverter' object has no attribute 'inverter_id'

2024-09-29 22:17:03.535631 INFO pred_bat: Info: record_status Error: Exception raised 'Inverter' object has no attribute 'inverter_id'
```

reverted back to 8.4.9, restarted appdaemon, clean run through, re-installed main, error:

```
2024-09-29 22:39:32.468258 INFO pred_bat: Error: Exception raised 'Inverter' object has no attribute 'inverter_id'
2024-09-29 22:39:32.474312 INFO pred_bat: Error: Traceback (most recent call last):
  File "/homeassistant/appdaemon/apps/batpred/predbat.py", line 11184, in update_time_loop
    self.update_pred(scheduled=False)
  File "/homeassistant/appdaemon/apps/batpred/predbat.py", line 10009, in update_pred
    status, status_extra = self.execute_plan()
                           ^^^^^^^^^^^^^^^^^^^
  File "/homeassistant/appdaemon/apps/batpred/predbat.py", line 8800, in execute_plan
    self.adjust_battery_target_multi(inverter, self.charge_limit_percent_best[0], isCharging)
  File "/homeassistant/appdaemon/apps/batpred/predbat.py", line 8384, in adjust_battery_target_multi
    self.log("Inverter {} adjust target soc for hold to {}% based on requested all inverter soc {}%".format(inverter.inverter_id, new_soc_percent, soc))
                                                                                                            ^^^^^^^^^^^^^^^^^^^^
AttributeError: 'Inverter' object has no attribute 'inverter_id'

2024-09-29 22:39:32.517622 INFO pred_bat: Info: record_status Error: Exception raised 'Inverter' object has no attribute 'inverter_id'
2024-09-29 22:39:32.530460 WARNING AppDaemon: Excessive time spent in callback 'update_time_loop() in pred_bat', Thread 'thread.thread-1' - now complete after 184.529682 seconds (limit=120.0)
```

And I noticed that despite the crash, the error attribute in predbat.status is still set to false, it should be set to true...

[image]

reverted back to 8.4.9, restarted appdaemon, all running OK now

[appdaemon.log](https://github.com/user-attachments/files/17181462/appdaemon.log)

springfall2008, 2024-10-01:
This crash is now fixed.

I released a beta version 8.4.10 which is the same as 'main' for now.

gcoan, 2024-10-08:
I did do some screen shots of 8.4.10, and there is a bit of feedback on potentially needing to look at the charge rate (so I'll share those separately), but we're in a power up event right now and I can see the different limits being applied to each inverter:

Plan says to charge to 79% then 85 then 100:
[image]

But the two inverters are on different SoC's which normally cause a problem with the stepped charge rates, however Predbat is now setting the target SoC for each differently:
[image]

gcoan, 2024-10-08:
Bit later on and I am seeing effects of the stepped charging still not quite being right.

overall target for the half hour slot is 85%

[image]

target for inverter G 87% (SoC 84%), H 81% (SoC 74%)

but as G is quite close to the target SoC the inverter has slowed the charge rate down to 1.2kWh rather than the 2.4kWh it should be going at
[image]

the target SoC needs to be set higher to stop this happening

Can see that the target SoC is being progressively refined by Predbat
[image]

chalfontchubby, 2026-08-20:
@gcoan since you raised this one, wanted to check in - are you happy to close this if it's resolved for you now, or would it be worth raising a fresh issue if there's still something specific to chase? We're trying to focus the backlog on live issues.

gcoan, 2026-11-10:
> [@gcoan](https://github.com/gcoan) since you raised this one, wanted to check in - are you happy to close this if it's resolved for you now, or would it be worth raising a fresh issue if there's still something specific to chase? We're trying to focus the backlog on live issues.

Looking at my target soc history for the last week, it's always been at 100% (a function of rates and house demand I guess).  So no evidence this is still occurring.  Closing

closed 2026-11-10 by gcoan