springfall2008/batpred #1485: Total Predbat savings is negative

Snid0, 2024-09-25:
**Describe the bug**
The total predbat savings entity is negative.
Both predbat.savings_total_predbat and  predbat.savings_yesterday_predbat are negative.

**Expected behavior**
I would be expecting predbat to be saving my money not costing me more? Or am I just reading this wrong?

**Predbat version**

v8.4.9

**Environment details**

- SolarEdge
- Docker

**Screenshots**
[image]
[image]

springfall2008, 2024-09-30:
Without more context its hard to comment, there two things here that can make it negative:

1. You have settings that are more pessimistic than reality, so Predbat did something more expensive to prepare for less solar or more load than reality. Look at things like best_soc_keep, pv10 weight and load10 scaling settings, plus anything like carbon cost.
2. Your ending battery level is higher than the non-predbat case which means although you spent a little more the value is also higher. That said your final SOC was 2.81 so unless your unit rate is quite high it wouldn't explain £1.77.

gcoan, 2025-12-29:
> What other info would you need from me to help figure it out? The plans predbat is coming up with seem correct to me and its definitely saving me cash!

your logfile, debug file and copies of the plan for today and yesterday would be useful to understand.  Trefor suggests some of the reasons for a negative savings figure above

gcoan, 2026-01-02:
I'd rather we keep everything self contained in the predbat documentation rather than have too much cross linking to other articles, that way we keep it easier for new predbat users.
I've looked at that link and its using the same modbus integration that is recommended for predbat, but there's quite a lot of stuff to wade through.

If we can boil down to what we need for predbat and then add it to the predbat docs.  I can do that, or you can, I don't mind.

Add the template sensor.solar_house_consumption_kwh and change apps.yaml ?  

And if there are sensor name differences we should get those right.

closed 2026-08-28 by chalfontchubby