springfall2008/batpred #1486: Charge rate limiting

MichaelA1a1, 2024-09-27:
**Describe the bug**

First of all i am not sure if this is a Predbat issue or possibly an issue with my set up at home, but i have only been having the issue since i installed Predbat within Home Assistant so just trying to understand if it could be causing this and its a setting that i have missed. 

But overnight when i am charging the car the rate at which it charges at changes, so it will go straight up to the 7kw that should be supported with no issue, but it will only stay there for around 15-20 minutes before dropping down for a period of time to say 1-2kw and then back up to 7kw. I have tested it out across different cars and different makes and the issue still remains. I'll add a screen shot from the GivEnergy Web interface to illustrate whats going on.

**Expected behavior**

Car charge rate should stay up at 7kw whilst charging is active.

**Predbat version**

v8.4.9

**Environment details**

Giv Energy AIO & Anderson A3 Charger
Standard HAOS installer or Docker

[image]

gcoan, 2024-09-27:
I think it would be useful if you posted a graph of the car charge rate sensor in home assistant so we can confirm that its Predbat incorrectly instructing the charger to charge at a reduced rate.

And a predbat logfile as well please will help diagnose the issue

springfall2008, 2024-09-28:
I doubt its predbat as it doesn't control the car charging rate

MichaelA1a1, 2024-10-02:
Hi both, 

Thank you for that. Just trying to go round the houses to try and rule out whats causing the issue. 
Am hoping that its related to the charger install as they tried to connect it up to a Solar CT Clamp which i think may be throwing things out. 

Coming next week to re check everything so hopefully be back to being able to charge at 7kw consistently

chalfontchubby, 2026-08-20:
Thanks for the report. This issue has been open for over a year with no recent activity - we're volunteers working through a large backlog and need to filter what's still relevant. If this is still a problem for you, please let us know and we'll keep it open. Otherwise, we'll close it in a week (around 2026-08-28).

MichaelA1a1, 2026-08-21:
Yes all sorted, happy for it to be closed. 

Thank you

closed 2026-08-21 by gcoan