springfall2008/batpred #1487: Sofar discharging has ceased to work.

dbw1966, 2024-09-27:
**Describe the bug**
When discharge set in the plan, or manual force discharge, the Sofar HYD-3600-ES does not enter force discharge, and predbat is sending the "auto" command.

This was earlier tested on issue #1454 and tested as "main", but is now not functional.

**Expected behavior**
Discharge per plan or manual.

**Predbat version**
v.8.4.9

**Environment details**
RPI, HA Core, Predbat Addon.  Sofar HYD-3600-ES, 2 x GTX5000 batteries.

- Inverter and battery setup
- Standard HAOS installer or Docker
- Anything else?

**Screenshots**
If applicable, add screenshots to help explain your problem. The most useful ones can be your battery chart, the Predbat HTML plan and your current settings in HA.
[image]
[image]
[image]

**Log file**
[predbat.log](https://github.com/user-attachments/files/17162079/predbat.log)
[home-assistant.log](https://github.com/user-attachments/files/17162080/home-assistant.log)

Can you capture a log file from the time of the issue, debug mode is not normally required.
If you are not keeping the full logs then please enable this in appdaemon.yaml (see the installation instruction in the Predbat docs area for details on how to do this)

dbw1966, 2024-09-27:
Tried with 8.4.7 and 8.4.8 and none of those seem to work.  The only one that did was the "main" prior to the 8.4.8 release.  Please advise when there's a new "main" I can try.

dbw1966, 2024-09-27:
Its started working on a scheduled discharge.  Predbat was displaying 8.4.9 as the running version, but settings was also showing an update available, which I'd ignored as it was set to auto update.  I have clicked for the update, and also updated HA, and its now working.....

springfall2008, 2024-09-28:
That's good, so you think this ticket is closed?

dbw1966, 2024-09-29:
I'd like to see behaviour on a discharge session again.

While checking things out this morning, I noted the car is charging from
battery.  This is obviously going to cost me 20% more than from the grid.

I have tried with allow car to charge from battery set to true or false.

Sofar inverter.  Is this because 1406 implement charge freeze for sofar has
not been implemented.  FYI here is the code that should do it on the
safar2mqtt.

void batterySave()
{
static unsigned long lastRun = 0;

if(checkTimer(&lastRun, BATTERYSAVE_INTERVAL) && BATTERYSAVE)
{
modbusResponse  rs;

//Get grid power
unsigned int p = 0;

if(!readSingleReg(SOFAR_SLAVE_ID, SOFAR_REG_GRIDW, &rs))
p = ((rs.data[0] << 8) | rs.data[1]);
else
Serial.println("modbus error");

Serial.print("Grid power: ");
Serial.println(p);
Serial.print("Battery save mode: ");

// Switch to auto when any power flows to the grid.
// We leave a little wriggle room because once you start charging the
battery,
// gridPower should be floating just above or below zero.
if((p < 65535/2 || p > 65525) && (INVERTER_RUNNINGSTATE != discharging))
{
//exporting to the grid
if(!sendPassiveCmd(SOFAR_SLAVE_ID, SOFAR_FN_AUTO, 0, "bsave_auto"))
Serial.println("auto");
}
else
{
//importing from the grid
if(!sendPassiveCmd(SOFAR_SLAVE_ID, SOFAR_FN_STANDBY, SOFAR_PARAM_STANDBY,
"bsave_standby"))
Serial.println("standby");
}
}
}

dbw1966, 2024-09-30:
Hi @springfall2008  I am looking at the system now, and you seem to be sending set/auto/true with the set/discharge:2800 message and approx 30 seconds later sending it again.  The first on is ignored, but the second one puts the inverter back to idle (normal) state.

[image]
[image]

[predbat.log](https://github.com/user-attachments/files/17194680/predbat.log)
[home-assistant.log](https://github.com/user-attachments/files/17194684/home-assistant.log)

Target SOC is being set.  

I just watch a discharge event start.  At 19:52:39 Sofar2mqtt/set/discharge:2800 and also Set/auto:true.  However, the inverter did start discharging to grid, then at 19:55:07 I got another Set/auto/true and the inverter stopped discharging to grid and returned to its normal running state.

I have reconfirmed this behavior with a force discharge as follows.

You can see grid current ramp up at 20:14:34, it tails off by 20:15:24.
[image]

The command set/discharge:2800 was at 20:14:34 and the next set/auto:true was at 20:15:07.

springfall2008, 2024-10-06:
Hmm, so set auto is being called when discharging when it shouldnt be?

springfall2008, 2024-10-06:
Please test the version on 'main' that has a fix

dbw1966, 2024-10-07:
That seems to fix that issue.  Huge thanks,  I believe Predbat should now achieve its potential.  
 I note reserve is set at 4%, but I don't think that does anything at the sofar2mqtt as I cannot see it as a set option.

The only remaining thing that may detract is issue Issue #1406 because ChargeFreeze seems to be a good tool.

closed 2025-01-01 by springfall2008