springfall2008/batpred #1488: Predbat needs reboot when car unplugged with Ohme integration

murkle, 2024-09-27:
When the car is unplugged PredBat doesn't update its plan unless Home Assistant is restarted, see https://github.com/dan-r/HomeAssistant-Ohme/issues/91

Not sure if it's the Ohme integration or PredBat at fault - is there anything I can do to help diagnose this? 🙏

springfall2008, 2024-09-28:
Have you tried the switch 'switch.predbat_octopus_intelligent_ignore_unplugged' setting this on should fix it?

murkle, 2024-09-28:
Thanks, found that under "Expert mode". Tested that, seems to be the same. Here's what I get after unplugging the car (was unplugged about 8.30pm)
[image]
[image]
[image]

murkle, 2024-09-28:
Then after rebooting Home Assistant it looks OK 
[image]

springfall2008, 2024-09-30:
There's something wrong here there is no 'Car' common on your plan, are you sure the car is actually enabled?

murkle, 2024-09-30:
The plan changes when I plug the car in - so something is working. What should it look like?

springfall2008, 2024-10-05:
There should be a car column on the plan:

[image]

murkle, 2024-10-06:
Thanks, solved that (num_cars :1 not 0 🤦‍♂️)

Still the same I'm afraid - car unplugged at 21:18 but the plan didn't update to account for that at 21:20 or 21:25

[image]

murkle, 2024-10-06:
... and after rebooting Home Assistant it's OK
[image]

gcoan, 2024-10-06:
> Thanks, solved that (num_cars :1 not 0 🤦‍♂️)

did you write that the wrong way round?  num_cars needs to be set to 1 if you want Predbat to manage or be aware of car charging via IOG,  and the car column is omitted if num_cars is zero

murkle, 2024-10-06:
Yes it was 0 now I've changed it to 1

springfall2008, 2024-10-07:
Resolving the ticket

murkle, 2024-10-07:
@springfall2008 to be clear: the problem is still there after changing num_cars to 1

springfall2008, 2024-10-13:
Did you set 'switch.predbat_octopus_intelligent_ignore_unplugged' to Enabled?

murkle, 2024-10-13:
@springfall2008 Yes, here's what I currently have:

[image]
[image]

springfall2008, 2024-10-14:
What is the car plugged in sensor set to? Can you show your 'config' page on the Predbat Web interface?

murkle, 2024-10-14:
Do you mean car_charging_now? These are the sensors pointing to the Ohme plugin:
`
  car_charging_now:
    - 'binary_sensor.ohme_car_charging'
  octopus_intelligent_slot: 'binary_sensor.ohme_slot_active'
  octopus_ready_time: 'time.ohme_target_time'
  octopus_charge_limit: 'number.ohme_target_percent'
  car_charging_energy: 'sensor.ohme_session_energy'
`

I've also got this (but that should be ignored with Intelligent I think)

`
  car_charging_planned:
    - 'False'
`

Sorry, I can't get the Predbat Web interface working (just updated from AppDaemon to try, separate post on Facebook)

murkle, 2024-10-14:
[image]
[image]
[image]
[image]
[image]

deanjesper, 2024-12-03:
I've found that when 'switch.predbat_octopus_intelligent_ignore_unplugged' is enabled, the 'Car' kWh isn't shown and it doesn't find the slots for future charging. If I disable it, the 'Car' kWh is shown and the slots are included in the plan.

Bizarelly 21:00-21:30 is showing as a peak slot...

[image]

Planned dispatches
- start: '2024-12-03T20:45:00+00:00'
end: '2024-12-03T21:15:00+00:00'
charge_in_kwh: -0.98
source: smart-charge
location: null
- start: '2024-12-03T21:45:00+00:00'
end: '2024-12-03T22:00:00+00:00'
charge_in_kwh: -1.629
source: smart-charge
location: null
- start: '2024-12-03T22:00:00+00:00'
end: '2024-12-03T22:15:00+00:00'
charge_in_kwh: -0.783
source: smart-charge
location: null
- start: '2024-12-03T22:30:00+00:00'
end: '2024-12-03T22:45:00+00:00'
charge_in_kwh: -0.772
source: smart-charge
location: null
- start: '2024-12-03T23:00:00+00:00'
end: '2024-12-03T23:15:00+00:00'
charge_in_kwh: -0.773
source: smart-charge
location: null
- start: '2024-12-03T23:30:00+00:00'
end: '2024-12-04T05:53:31+00:00'
charge_in_kwh: -42.151
source: smart-charge
location: null

closed 2026-08-21 by murkle