springfall2008/batpred #1489: Suboptimal IOG plan

nickgee31, 2024-09-28:
**Describe the bug**
My GivEnergy inverter is today freeze discharging and discharging mid-afternoon, leaving no battery available for the evening.

**Expected behavior**
All discharges should happen as late as possible in the day.

**Predbat version**

V8.4.9

**Environment details**

- GivEnergy Gen 2 5KW + 9.5 battery
- Standard HAOS installer

**Screenshots**
[image]
[image]

**Log file**
[predbat.log](https://github.com/user-attachments/files/17174236/predbat.log)

gcoan, 2024-09-28:
I'd suggest that whilst the plan may be discharging earlier than you'd like, it's a perfectly reasonable plan that retains enough battery charge to last you through to the IOG cheap rate charge period with no grid imports 

What I would question though is your load sensor, it's predicting very very low house load every half hour slot, so low that any load scaling is in the margins of error. I think you need to check the sensor is giving an accurate load history as it seems too low. 

If it is correct then arguably the plan is fine as is. If you want to keep more charge for later then increasing best soc keep might do what you want. 

There are other issues on the recent freeze discharge changes. 

See also #1462

nickgee31, 2024-09-28:
Hey,

I am generally not here on weekends so it is quite reasonable that my load is that low on a weekend. I guess what I am getting at is that on the off chance that load is increased then it should leave enough before the cheap period at 23:30 - not potentially leave with nothing left.

gcoan, 2024-09-28:
It would be worth you trying to configure a higher level for best_soc_keep to advise Predbat of a soc level to keep.  This is not a hard soc level that will always be kept (that's best_soc_min) but advises Predbat of a soc level that you wish to keep for 'just in case's'

I do agree with your sentiment though, delaying forced discharge as long as possible is a good strategy in case house load is higher than expected

gcoan, 2024-09-28:
Best_soc_keep definitely makes a positive impact in how much soc is retained in case of unexpected load. 

My plan tomorrow with best_soc_keep set to my default 0.5, most of the day the battery is in freeze discharge mode and whilst Soc drops the battery never empties (assuming the load forecast is correct):

[image]
[image]
[image]

I increased best_soc_keep to 1.5 (kWh) and there's less freeze discharging through the day and the Soc is about 10% higher:

[image]
[image]
[image]

nickgee31, 2024-09-29:
Thank you. The issue with changing the best_soc_keep is that it now leaves the battery with 20% at the end of the day rather than down to 5% like I would like. It used to be better when you could have best_soc_keep at 2.0 and best_soc_min at 0.5 - that used to work rather well but that functionality was removed.

gcoan, 2024-09-29:
> It used to be better when you could have best_soc_keep at 2.0 and best_soc_min at 0.5 - that used to work rather well but that functionality was removed

That's pretty much what I have set, bsk is now 1.5 and bsm 0.5. Not sure what you mean by the functionality was removed. I'm not on iog so can't test exactly but give it a go and see what it does for your Soc. 

If this retains too much Soc you could dynamically change the bsk with an automation to different levels in the evening to further reduce the retained soc if you wanted to

nickgee31, 2024-09-29:
What I mean is that setting best_soc_min to 0.5 would actually mean it would discharge down to that level, whereas it doesn't do that anymore.

To be honest I don't want to introduce automations to change levels, it should just be a set and forget kind of thing like it used to be.

phiebs, 2024-09-29:
I have been tweaking various settings to try and achieve the same result. The way I’ve done this is to set the `rates_export_override` to prevent discharging in the peak evening slot and also to encourage discharging just before the cheap rate kicks in. I do this with `rate_increment` and set it to -10 for peak period and +10 from 22:00-23:30. You may need to tweak the times and increment rates for your particular setup but it works well now for me.

phiebs, 2024-09-29:
Just to follow up after giving this some more thought, it would be super cool to have a switch that did a “export all battery just before lower rate starts” or similar. We know the battery kWh value and the max rate of discharge so should be easy to calculate when to start this. E.g. 9kWh in battery with 6kW discharge rate, would equal 1.5 hours of discharge to end exactly at the start of the lower rate window. This would change all the time but once the capacity remaining is greater than time until cheap rate x max export power we can safely trigger an export without fear of running out.

nickgee31, 2024-09-29:
I think that's kind of what we need @phiebs, the tweaking of export rates is okay but just thinking if there was some early IOG slots then the battery wouldn't discharge before them if the export rate was too low.

phiebs, 2024-09-29:
Yep, that’s why I think the toggle to discharge before next cheap slot would solve the issue. Could probably write an automation to do just this but would be way more convenient within predbat. P.s. as we come into winter the increment adjustment on export rates helps with avoiding any use during peak hours to maximise gains with DFS sessions

nickgee31, 2024-09-29:
@springfall2008 Is this something you think may be possible at all? To delay discharges until the next cheap slot to prevent running out of battery before the end of day.

gcoan, 2024-09-29:
> as we come into winter the increment adjustment on export rates helps with avoiding any use during peak hours to maximise gains with DFS sessions

Yet to see what shape DFS takes this winter but this is exactly what I did last year to discourage predbat from peak rate discharging

springfall2008, 2024-09-29:
The original post issue was zero load prediction, fixing that would have solved the problem.

The secondly issue is that discharge freeze (exporting solar directly) causes less losses than charging the battery and exporting it later. If that feature were disabled (currently its hard wired on) then you would probably get what you wanted.

You might also try including load_scaling10 to make your worst case load predictions a bit higher

nickgee31, 2024-09-30:
The load prediction was just really low for a weekend, I will adjust the load_scaling10 to compensate for that.

Can I edit something in the code locally to disable the feature to see how it behaves?

nickgee31, 2024-09-30:
Managed to find it, in config.py on line 1031. Changing support_discharge_freeze to False has resulted in a much better plan being created and not discharging until last thing in the day.

closed 2024-10-03 by nickgee31