springfall2008/batpred #1497: Status	Error: Exception raised 'NoneType' object has no attribute 'lower'

markbowden, 2024-09-29:
**Describe the bug**
Following the install videos when I try and run PredBat the status shows as "Status | Error: Exception raised 'NoneType' object has no attribute 'lower'". I make it to the point of looking through the apps.yaml and when I try and access the Prebat Dash via the shortcut on the sidepanel, I see the following error. I've only changed the invertor max rate, and then retried with a clean install, but I see the same result regardless. 

**Expected behavior**
This shouldn't happen?

**Predbat version**
v8.4.9

**Environment details**

- Inverter and battery setup - Solar Edge
- Standard HAOS installer or Docker - Standard HAOS
- Anything else? - Running an RPi5

**Screenshots**
If applicable, add screenshots to help explain your problem. The most useful ones can be your battery chart, the Predbat HTML plan and your current settings in HA.

**Log file**
2024-09-29 23:30:01.077302: Info: record_status Error: Exception raised 'NoneType' object has no attribute 'lower'
2024-09-29 23:30:01.077385: Error: 'NoneType' object has no attribute 'lower'
Traceback (most recent call last):
  File "/config/hass.py", line 184, in timer_tick
    item["callback"](None)
  File "/config/predbat.py", line 11155, in run_time_loop
    raise e
  File "/config/predbat.py", line 11150, in run_time_loop
    self.update_pred(scheduled=True)
  File "/config/predbat.py", line 9919, in update_pred
    self.fetch_sensor_data()
  File "/config/predbat.py", line 8971, in fetch_sensor_data
    self.load_minutes, self.load_minutes_age = self.minute_data_load(self.now_utc, "load_today", self.max_days_previous, required_unit="kWh")
  File "/config/predbat.py", line 1031, in minute_data_load
    history = self.get_history_wrapper(entity_id=entity_id, days=max_days_previous)
  File "/config/predbat.py", line 971, in get_history_wrapper
    history = self.ha_interface.get_history(entity_id, days=days, now=self.now)
  File "/config/ha.py", line 374, in get_history
    self.db_mirror_list[sensor.lower()] = True
AttributeError: 'NoneType' object has no attribute 'lower'

gcoan, 2024-09-30:
We will need some more information.  Can you post your apps.yaml and full logfile please.

What mode is predbat set to?  Try monitor mode to start off with if it's not already set.

Assume you have checked that the inverter software is already working fine?

springfall2008, 2024-09-30:
It looks like "load_today" hasn't been set in apps.yaml

markbowden, 2024-09-30:
> We will need some more information. Can you post your apps.yaml and full logfile please.
> 
> What mode is predbat set to? Try monitor mode to start off with if it's not already set.
> 
> Assume you have checked that the inverter software is already working fine?

I've attached the files. I'm talking to the invertor / battery using the official SolarEdge integration, and I think that's where the problem lies. I think I need to use the Modbus one linked in the documentation... 
[logfile.txt](https://github.com/user-attachments/files/17186589/logfile.txt)
[apps.yaml.txt](https://github.com/user-attachments/files/17186590/apps.yaml.txt)

markbowden, 2024-09-30:
> It looks like "load_today" hasn't been set in apps.yaml

Thanks for the reply, I think that's one of the issues. 

If I have a SolarEdge setup, do I have to still use GivTCP or should I use the SolarEdge version from here? https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Configuration

closed 2025-01-01 by springfall2008