springfall2008/batpred #1500: Plan does not reflect actual SOC during charging slot

toxicneo, 2024-09-30:
**Describe the bug**
I'm on Octopus Flux and set to charge to 100% overnight 2am to 5am via the inverter settings, the inverter does this each day without issue. Batpred recognises the charge slot & SOC target and adds it to the plan, but the predicted SOC is not reflected in the plan and continues to show the battery discharging through the charging slot.

**Expected behavior**
Plan show the charge slot and SOC correctly

**Predbat version**
8.4.9 standalone in HACS

**Environment details**

- Gen 1 3.6kW inverter with 5.2kWh Gen 1 battery
- HAOS running on HP thin client

**Screenshots**
[image]

**apps.yaml**
`# ------------------------------------------------------------------
# This is an example configuration, please modify it
# ------------------------------------------------------------------
---
pred_bat:
  module: predbat
  class: PredBat

  # Sets the prefix for all created entities in HA - only change if you want to run more than once instance
  prefix: predbat

  # Timezone to work in
  timezone: Europe/London

  # XXX: This is a configuration template, delete this line once you edit your configuration

  # Set to auto-match with a GivEnergy serial number, but you can override the serial or the sensor names
  # if it doesn't work or if you have more than one inverter you will need to list both
  geserial: 're:sensor.givtcp_(.+)_soc_kwh'
  geserial2: 're:sensor.givtcp2_(.+)_soc_kwh'

  # Sets the maximum period of zero load before the gap is filled, default 30 minutes
  # To disable set it to 1440
  load_filter_threshold: 30

  #
  # Sensors, more than one can be specified and they will be summed up automatically
  #
  # For two inverters the load today would normally be the master load sensor only (to cover the entire house)
  # If you have three phase and one inverter per phase then you would need three load sensors
  #
  # For pv_today if you have multiple solar inverter inputs then you should include one entry for each inverter
  #
  load_today:
    - sensor.givtcp_{geserial}_load_energy_today_kwh
  import_today:
    - sensor.givtcp_{geserial}_import_energy_today_kwh
  export_today:
    - sensor.givtcp_{geserial}_export_energy_today_kwh
  pv_today:
    - sensor.givtcp_{geserial}_pv_energy_today_kwh

  # Load forecast can be used to add to the historical load data (heat-pump)
  # To link to Predheat
  # Data must be in the format of 'last_updated' timestamp and 'energy' for incrementing kWh
  #load_forecast:
  #  - predheat.heat_energy$external
  #
  # If you enable ge_cloud_data then the load/import and export data will be fetches from the GE cloud instead of from GivTCP sensors
  # this is usually less efficient and of course prone to internet downtime, but could be useful if you lost your GivTCP data
  # Set the serial to the inverter serial to pull the data from and the key to your API key
  # When this is set load_today, import_today and export_today are not used
[... 316 lines cut ...]
  export_triggers:
     - name: 'large'
       minutes: 60
       energy: 1.0
     - name: 'small'
       minutes: 15
       energy: 0.25

  # If you have a sensor that gives the energy consumed by your solar diverter then add it here
  # this will make the predictions more accurate. It should be an incrementing sensor, it can reset at midnight or not
  # It's assumed to be in Kwh but scaling can be applied if need be
  #iboost_energy_today: 'sensor.xxxxx'
  #iboost_energy_scaling: 1.0
  # Gas rates for comparison
  #metric_octopus_gas: 're:(sensor.(octopus_energy_|)gas_[0-9a-z]+_[0-9a-z]+_current_rate)'

  # Nordpool market energy rates
  #futurerate_url: 'https://www.nordpoolgroup.com/api/marketdata/page/325?currency=GBP'
  #futurerate_adjust_import: False
  #futurerate_adjust_export: True
  #futurerate_peak_start: "16:00:00"
  #futurerate_peak_end: "19:00:00"
  #futurerate_peak_premium_import: 14
  #futurerate_peak_premium_export: 6.5
`

springfall2008, 2024-10-06:
Can you add your logfile from Predbat please as this seems quite odd?

springfall2008, 2024-10-06:
If I were to guess I think you have charge_rate to set 0?

toxicneo, 2024-10-07:
[predbat.log](https://github.com/user-attachments/files/17276457/predbat.log)
Hi Trefor, I think this is the file you need.

My charge rate is set to 2600 both in the GE app and I can see it in MQTT too as the same value.

springfall2008, 2024-10-13:
Okay so I think the issue is Monitor mode has a bug in it where it reports discharge incorrectly. I've pushed a fix to 'main' for testing which will be in the next release. 

Try changing to 'control charge & discharge mode' and set 'read only' to True if you don't want the plan to be changed.

toxicneo, 2024-10-14:
Perfect, changing modes has worked and it shows the correct SOC, thank you Trefor!

closed 2026-08-21 by gcoan