springfall2008/batpred #1501: Charging in peak IOG hours

KWillow65, 2024-09-30:
**Describe the bug**
For the third time this evening, Predbat has decided to charge in my peak hours. A restart of Predbat Addon fixes it but then it comes back again. 

**Expected behavior**
For the charge not to happen.

**Predbat version**

V8.4.9

**Environment details**

- FoxESS Inverter
- Standard HAOS installer 

**Screenshots**
[image]

**Log file**
[predbat.log](https://github.com/user-attachments/files/17195556/predbat.log)
[apps.log](https://github.com/user-attachments/files/17195562/apps.log)

AKlamaznikovs, 2024-10-01:
I don't have a screenshot, but I have experienced similar behaviour in the past few days. There was enough battery to last until the "night" rate, yet Predbat decided to top up at "day" rate. In my setup, I use "control charge only".

KWillow65, 2024-10-01:
I had a bit of a thought it could possibly be to do with the price increase on IOG overnight. Therefore it might have been seeing the previous day rate (yesterday's) as cheaper than today's day rate. I've not seen it reoccur today yet.

closed 2024-10-02 by KWillow65