springfall2008/batpred #1502: Solis inverter charge timing issue

FarmingMartin, 2024-10-01:
**Describe the bug**
After a planned charge window ends predbat assigns another random charge window to my solis inveter.
Example today 1/10/24
Planned charge window 2.00 t0 6.00.
Worked ok, but at 6.00 prebat assigned the following time to my solis inveter 
2.00 to 15.30
This resulting in battety continuing to charge to 100% then all house load drawn from the grid. Prebat status shows idle with no planned charge showing .
Now set prebat to monitor mode as I can't trust it at the moment.
Issue raised two weeks previously and updated by myself regularly but seems to be being ignored.
Solis inverter using solaX modbus

**Expected behavior**
A clear and concise description of what you expected to happen.

**Predbat version**

V8.4.9

**Environment details**

- Inverter and battery setup
- solis inverter with pylontec US5000 batteries
- 
- Anything else?

springfall2008, 2024-10-03:
Can you try this release please? 

https://github.com/springfall2008/batpred/releases/tag/v8.4.11

FarmingMartin, 2024-10-03:
Hi thanks for the reply. Will this update automatically via the predbat automation or do I need to download something ?I'm at work at present will try to update once I get home.Sent from my Galaxy

https://github.com/springfall2008/batpred/releases/tag/v8.4.11

FarmingMartin, 2024-10-04:
Tried the update but still not working. Charge planned last night and this morning random charge charge set on my Solis inverter so battery was charging but predbat status shows idle. Also had error message. See screen shots below and log file.
[predbat.log](https://github.com/user-attachments/files/17253568/predbat.log)
[image]
[image]
[image]
[image]

springfall2008, 2024-10-05:
Can you share your apps.yaml and predbat log file?

FarmingMartin, 2024-10-05:
Please find the requested log file and apps YAML (hopefully)
hope this helps you to find a solution.

[predbat.log](https://github.com/user-attachments/files/17267380/predbat.log)
[apps yaml.txt](https://github.com/user-attachments/files/17267381/apps.yaml.txt)

FarmingMartin, 2024-10-06:
Just an update. Batpred had planned a charge for several time slots which seemed to of worked as requested, but then Batpred has planned a hold charge 
[image]
[image]
at 6.00 and has programmed a charge from 6.00 till 1.00 at 40 A which will cause charge for this time period. See attached photos.

FarmingMartin, 2024-10-06:
Then at 5.55 it removed the hold charge window and reset a charge on my inverter to 1.00 to 1.30. Which it has also set in its charge window with in batpred. This would now work as required.
To me it looks like it could be something to do with the hold charge when there is no any charge windows scheduled after it. I could be wrong but it's worth a look. Attached photos showing latest predbat charge windows and solis inverter programmed charge times. 
[image]
[image]

FarmingMartin, 2024-10-06:
here are the latest logs.
[predbat log 1.txt](https://github.com/user-attachments/files/17268332/predbat.log.1.txt)
[predbat log 2.txt](https://github.com/user-attachments/files/17268333/predbat.log.2.txt)
[predbat log 3.txt](https://github.com/user-attachments/files/17268334/predbat.log.3.txt)

Freddo2, 2024-10-11:
Hi, just to confirm similar behaviour with apparently charge windows being made that are not on the plan. This is with  v8.4.11,  Solax Modbus 2024.09.05 and a Solis RHI-3K-48ES-5G.

springfall2008, 2024-10-13:
looking at the log I can't see anything wrong with the settings that are being made.

My theory now is sometimes the press and poll doesn't work and leaves a time at the previous setting. However Predbat can't actually read back the real inverter settings right now so it thinks they are right and then subsequently doesn't change the setting.

I'm going to try a workaround by making it always update the start and end times...

springfall2008, 2024-10-13:
Please try the fix on 'main' and see if it helps, if not please send me the logfile for the period in question (the above was missing part of the period).

Freddo2, 2024-10-13:
Thank you. 

Here is my log file. This line is an example where there is a charge current despite the slot being clear on the plan:

2024-10-12 00:05:03.233599: Inverter 0 SOC: 4.0kW 50% Current charge rate 3000W Current discharge rate 3000W Current power 2988.0W Current 

Looking at the behaviour today (before your suggested edit)  I think that parts of the time slots are not being cleared and this is leaving artefacts. Sometimes these involve a clash between charge and discharge slots, sometimes they cause new charge periods, and sometimes things change before they would become active. 

(I was guessing that turning on and off the scheduled charge was running faster than the solax connection and solis change.)

I'll see let you know what what your edits do. 

[20241012 predbat.txt](https://github.com/user-attachments/files/17356558/20241012.predbat.txt)

Freddo2, 2024-10-15:
Predbat updated to 8.5.0 last night. It looks like there was a glitch at 05:17 this morning . From looking at the charge 1 sensors, Charge 1 should have updated from 10:30:00-11:00:00 to 00:00:00-00:00:00, but this wasn't a clean change and only the start hour changed. This opened up an unplanned charge window. 

[20241015 predbat.txt](https://github.com/user-attachments/files/17374861/20241015.predbat.txt)

springfall2008, 2024-10-27:
I'm struggling to know how to fix this as it seems like a communication error where the register isn't being written correctly.
Is it possible for Predbat to read back the real value to confirm it was written?

Freddo2, 2024-10-27:
[work around automations .txt](https://github.com/user-attachments/files/17535225/work.around.automations.txt)

I am using some automations to cover this (they are in the text file, which appears to be working OK). There is a control automation that is triggered by a change in the predbat best charge and discharge schedules, then the values from this are written to helpers, and finally the helpers (charge and disacharge) are written to solax with a button press. There are default values for the discharge schedule as it isn't always used, but needs to be written as it may still contain legacy values. 

There is a brief delay in the sequence as an attempt to try to prevent possible timing issues. Running the automations through a control automation that is triggered by the change to the best predbat charge/discharge times doesn't create a loop as only 1 instance of the control automation is allowed in HA. So, it isn't elegant but does appear to work...

The writing and reading via solax appears to be stable (I have another helper which uses automation 4).  When I was looking at the helpers, it did appear that numerical precision was coming into the time values which would need to be accounted for in any checks.

FarmingMartin, 2024-10-27:
Unfortunatly I have stopped using batpred now in favour of PV-OPT which seems to work better with my Solis inverter  so I am unable to give you any feedback. If this issue is resolved in the future then I will try batpred again as apart from the timing issue I really like it.

closed 2025-01-01 by springfall2008