springfall2008/batpred #1506: inverter_limit_charge and _discharge Beaviour

jasonwfrancis1, 2024-10-03:
Currently the following settings inverter_limit_charge and _discharge only accept numbers in the config file.

would it be possible to make these settings accept a home assistant helper entity so that control of the values can be more dynamic?

the scenario is I can charge/discharge at 12kw but when my car is also charging I need to lower the limit for the batteries. To do this now I have to edit the config file, if it was able to put it in home assistant helper entities then an automation could handle this for me.

springfall2008, 2024-10-05:
You should be able to set them to a template sensor/input in HA e.g. sensor.xxxx or input_number.yyy does it not work?

jasonwfrancis1, 2024-10-05:
Just tried it, it doesn’t appear to work. I out the input number entity in single quotes like the other lines and it just defaulted the charge rate back to what’s reported by the inverter. Changing the number didn’t do anything.

gcoan, 2024-10-05:
> You should be able to set them to a template sensor/input in HA e.g. sensor.xxxx or input_number.yyy does it not work?

I had previously tried setting inverter_limit_charge to a HA entity as a potential way to tackle inverter cross charging https://github.com/springfall2008/batpred/issues/1031#issuecomment-2101644298

But per the screenshots etc, it doesn't work

springfall2008, 2024-10-07:
>But per the screenshots etc, it doesn't work

I would re-test it worked just fine for me when I tried, maybe a typo in the YAML file?

closed 2026-08-28 by chalfontchubby