springfall2008/batpred #1507: Variable charging rates due to temperature

Philhews, 2024-10-04:
With winter rapidly approaching and the temperature of my battery now dropping it has started limiting the charge rate via the BMS. 
Previously this was about 3.2kW rather than 3.6 and didn't really cause much of a problem but now that GE have changed the rate it is down to about 2.7kW which means that predbat is planning based on a much higher charge rate than I am getting.

I know I can set inverter_charge_limit in the apps.yaml config but that also then sets the actual charge rate so on the occasion that my battery goes above the temperature threshold it would be limiting the charge rate unnecessarily.

In a perfect world predbat would have a temperature curve for charge rate (sort of like the SOC based one for top end charging), and base it on the battery cell temperature (or a custom sensor that people can configure) and then it can plan based on that temperature without actually changing the charge rate on the inverter. As the temperature changes the plan would re-calculate using a new charge rate.

As a more simple solution a flat configurable 'planning charge rate' variable (not apps.yaml) could be added so that the plan can be based on a different rate to what it sets on the inverter.

I am aware in both of these situations the rate expected in the plan and that actually achieved could be different as just the act of charging could warm the battery enough to change the rate but I think it would be better than currently where I am stuck either planning a charge I can't achieve (and therefore having less battery than I expected), or potentially losing out on charge due to being limited by the inverter charge rate setting.

springfall2008, 2024-10-05:
What inverter do you have and have you tried the charge rate curve feature?

Philhews, 2024-10-05:
It's a GE Gen 2 3.6 Hybrid.

I have the charge curve set for the 90-100% throttling, hadn't thought about using it to limit the modelled rate on the rest of the range, would I have to add a row in config for each percent value? (100 rows) or will it assume the same value across a gap?

closed 2026-08-28 by chalfontchubby