springfall2008/batpred #1508: Starting Predbat in Monitor mode, should not be changing inverter settings

arfond, 2024-10-05:
I have these two lines in my apps.yaml 

 inverter_limit_discharge:
   - 9000
   
  inverter_limit_charge:
   - 10000

At the moment I'm running PredBat in monitor mode exclusively as I test our some of my own automations (PredBat charts still very useful!).

I've noticed that when I restart PredBat it writes the discharge limit to the inverter. I think this shouldn't be happening when PredBat is in monitor mode. I don't see this behaviour with the charge limit.

**Predbat version**

Add-on 1.21
App 8.4.11

**Environment details**

- Dual AIO
- HA in proxmox.
- GivTCPv3 Dev (I've noticed this behaviour in main version and Beta)

**Log file**

GivTCP log (note timestamp)

2024-10-05 08:43:57,129 - Inv3 - write       -  [INFO    ] - Setting battery discharge limit 9000 was a success

PredBat log:
-----------------------------------------------------------
 Add-on: Predbat
 Home Battery Prediction and Control
-----------------------------------------------------------
Bootstrap Predbat
Startup
 Add-on version: 1.2.1
 You are running the latest version of this add-on.
 System: Home Assistant OS 13.1  (amd64 / qemux86-64)
 Home Assistant Core: 2024.9.3
 Home Assistant Supervisor: 2024.09.1
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-timezone: starting
s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
[08:43:39] INFO: Configuring timezone (Europe/London)...
s6-rc: info: service base-addon-log-level successfully started
s6-rc: info: service base-addon-timezone successfully started
s6-rc: info: service legacy-cont-init: starting
Predbat files are installed correctly for version v8.4.11
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
**** Starting Standalone Predbat ****
2024-10-05 08:43:40.223699: Loading apps.yaml
s6-rc: info: service legacy-services successfully started
2024-10-05 08:43:40.274234: Info: Connected to Home Assistant at http://supervisor/core
2024-10-05 08:43:40.276596: Info: Add-on slug is 6adb4f0d_predbat
2024-10-05 08:43:40.277134: Creating task: <coroutine object HAInterface.socketLoop at 0x7f6dd8937ed0>
2024-10-05 08:43:40.277630: Info: Web Socket task started
2024-10-05 08:43:40.277725: Info: Start socket for url http://supervisor/core/api/websocket
2024-10-05 08:43:40.278496: Creating task: <coroutine object WebInterface.start at 0x7f6dd8937e60>
Web interface started
2024-10-05 08:43:40.280140: Warn: unable to find /addon_configs/6adb4f0d_predbat/appdaemon.yaml skipping checks as Predbat maybe running outside of AppDaemon
2024-10-05 08:43:40.294063: Info: Web Socket active
Watching ['./download.py', './unit_test.py', './predbat.py', './config.py', './hass.py', './apps.yaml', './utils.py', './inverter.py', './web.py', './prediction.py', './ha.py'] for changes
2024-10-05 08:44:03.068964: Info: record_status Idle
2024-10-05 08:45:09.413832: Info: record_status Idle
2024-10-05 08:50:09.383667: Info: record_status Idle

springfall2008, 2024-10-06:
I believe that 'read only' toggle is the one you want to use, although I agree having monitor mode change things is a little misleading.

arfond, 2024-10-06:
Thank you @springfall2008

closed 2026-08-28 by chalfontchubby