springfall2008/batpred #1511: Predbat with agile and Ohme

retr083, 2024-10-05:
**Describe the bug**
Every time EV is plugged in and Ohme schedules a charge the agile rates change to the lowest half hourly rate for the entire time the car is scheduled to charge (see image, 5.29p being the lowest price at 22:30-23:00 but duplicating across the entire ev charge schedule). I am not using predbat to schedule charges.

**Expected behavior**
Agile rates should not change but predbat should know what times the car is going to charge.
**Predbat version**

8.4.11

**Environment details**

- Givenergy 5KW hybrid & 9.5kwh battery
- Appdaemon install
- Octopus addon
- Ohme addon

**Screenshots**
If applicable, add screenshots to help explain your problem. The most useful ones can be your battery chart, the Predbat HTML plan and your current settings in HA.

[image]

**Log file**
Can you capture a log file from the time of the issue, debug mode is not normally required.
If you are not keeping the full logs then please enable this in appdaemon.yaml (see the installation instruction in the Predbat docs area for details on how to do this)

[2024-10-05 192042.835978 WARNING Ap.txt](https://github.com/user-attachments/files/17267099/2024-10-05.192042.835978.WARNING.Ap.txt)

springfall2008, 2024-10-05:
Hmm, its probably as its designed for intelligent GO, I didn't know Ohme could schedule without GO. I think I can make that an option for you...

retr083, 2024-10-05:
that would be great, this is the entity in the ohme integration that gives the list of charge slots, sensor.ohme_charge_slots

springfall2008, 2024-10-05:
Please try updating to 'main' and setting

octopus_slot_low_rate: False

In apps.yaml - this will tell Predbat to not change the rates based on the charge slots

retr083, 2024-10-05:
Working perfectly, thanks so much

closed 2024-10-13 by springfall2008