springfall2008/batpred #1520: Switching from Givenergy to Foxess - comments please

TheSargey, 2024-10-10:
Evening all, 

Apologies, I appreciate this isn't a feature request. I've had Givenergy and predbat running for a good year now and happy with it, however I'm looking to upgrade the Givenergy and have had a couple of installers recommending Foxess over both Giv and others (tesla etc) 

I'm really keen on keeping batpred (because it's a work of magic, thanks Trefor) just wanted opinions on the current state of play around how well batpred works with foxess inverters? I note there is "basic" integration although clearly not to the level that Givenergy is at. Seems also that additional kit is required to link the inverter to HA (I. E. WiFi dongle of some kind) 

Any comments appreciated and again, sorry for posting in this section. Is there a better forum to ask such questions? 

Cheers, 

Sarge

icornish72, 2024-10-20:
I'm a foxess user, never had GIv. Using Predbat now for a month or so in Charge Only mode mainly.
Fox is reasonably reliable (so far - only had it 6 months), as long as you ensure your installer also sets up the modbus connection and you use the Modbus foxess connector in HACS (using the cloud app means you have a 5 minute wait for action).

TheSargey, 2024-10-20:
> I'm a foxess user, never had GIv. Using Predbat now for a month or so in Charge Only mode mainly. Fox is reasonably reliable (so far - only had it 6 months), as long as you ensure your installer [...]

Thanks Cornish, which modbus "dongle" are you using? Is it a USB or WiFi one? Did your installer include the dongle or did you get that yourself?

Cheers,

Sarge

icornish72, 2024-10-21:
I have the ew11, about £50 from Amazon. It's a Wifi unit.
I asked if my installer would do it and they refused, but since I already had 90% of the install completed, and it seemed fairly straightforward to do (I'm ok on basic electronics), I did the connection myself - just took my time and checked 5 times before i turned the inverter etc on again!

TheSargey, 2024-10-21:
> I have the ew11, about £50 from Amazon. It's a Wifi unit.

Ah excellent, that's the modbus I've been considering. Hopefully I can get my installer to do some of the wiring on my behalf etc but if needs be it doesn't look too complicated.

Does the fox inverter work with all of the batpred plan options? Like freeze charge and so forth? And did you just use the fox.yaml that is published here or did you do your own?

Sorry for all the questions, I just want to be clear before taking a rather expensive plunge into the unknown!

Cheers,

Sarge

StealthChesnut, 2024-10-30:
@TheSargey connecting the wiring isn't too difficult, I used a USB adapter in the back of a computer directly underneath to get going. Diagrams and instructions on my repo (https://github.com/StealthChesnut/HA-FoxESS-Modbus/wiki/USR-W610-Setup-Guide#connecting-to-your-inverter-h1-or-ac1), proceed with caution at your own risk, of course!

I am in the process of getting predbat working - it does connect and control charging, but I have a few cases where it hasn't been able to control charging. Just trying to troubleshoot at the moment.

icornish72, 2024-10-30:
I have a fox system with modbus and batpred works well for me - only use it to control my charge as my goal is not to maximise income (high energy user) but to optimise cost and ensure we avoid using peak prices from the grid.
Did tinker with managing discharge and charge, and it worked, but left me feeling like I was flying by the seat of my pants.

TheSargey, 2024-10-30:
> I have a fox system with modbus and batpred works well for me - only use it to control my charge as my goal is not to maximise income (high energy user) but to optimise cost and ensure we avoid [...]

Thanks Cornish - ultimately I don't think I'd be looking to use batpred for export either (I don't do that now with Givenergy). Just out of curiosity, does the FoxESS system not have Octopus integration for such things already implemented? I thought I read somewhere recently you were able to set the inverter via their cloud to do this? Prior to using batpred I used Octopus R&D to control my inverter and pick time periods etc. But it wasn't anywhere near as "elegant" as batpred...

icornish72, 2024-10-30:
Yes the foxess system does support agile via the cloud system, but it's a
fairly 'crude' system where you set a max price for charging and it will
charge if below that price.
I found it worked fairly well but wasn't very useful in terms of managing
load in cases where the price wasnt going below the normal cut off point
(like this week for example!). BatPred at least looks at our load patterns
and plans around that.

TheSargey, 2024-10-30:
> Yes the foxess system does support agile via the cloud system, but it's a fairly 'crude' system where you set a max price for charging and it will charge if below that price. I found it worked [...]

Thanks again Cornish, sounds similar to the Givenergy/octopus setup. I'm erring now on the side of just upgrading my current setup rather than switching out to Fox. 

Thanks for all the input gents, much appreciated.

closed 2024-10-30 by TheSargey